libretro/Lakka-LibreELEC #1831: Does not work on H96PRO+

Marcusus57, 2023-02-14:
Hello, does not work with:
Lakka-S912.arm-2.3.1.img ;
Lakka-AMLGX.aarch64-4.3-box.img
On H96PRO+ TV BOX 3GB RAM 32GB, Android 7.1.2
The image file was burned to a microsd with win32diskimager1.0.0
After doing the RESET button procedure + AC adapter connection, the boot remains stuck on the H96PRO+ logo.

ToKe79, 2023-02-14:
Hi, when you are using the `box` image, you have to edit `uEnv.ini` and replace `@@DTB_NAME@@` with proper file name - see folder `dtb` and try to find one matching your device. In your case probably `gxm_q201_3g_1gbit.dtb`.

Marcusus57, 2023-02-18:
Thank you but gxm_q201.dtb does not work

Marcusus57, 2023-02-18:
> Hi, when you are using the `box` image, you have to edit `uEnv.ini` and replace `@@DTB_NAME@@` with proper file name - see folder `dtb` and try to find one matching your device. In your case probably `gxm_q201_3g_1gbit.dtb`.

Thank you but "gxm_q201.dtb" does not work

Marcusus57, 2023-02-19:
No gxm.dtb files work, the tv box must be rooted?

Marcusus57, 2023-02-19:
The version gxm_q201_3g_1gbit.dtb is on the EmuELEC package but not in LAKKA, i will test it

ToKe79, 2023-02-20:
I don't think the dtb files are interchangeable. Depends on what kernel version EmuELEC uses.
If I may suggest, try using [devel image from here](https://nightly.builds.lakka.tv/devbuild_latest/AMLGX.aarch64/).
As we use LibreELEC as base, you can also test with their [latest beta release](
https://libreelec.tv/downloads/amlogic/).

Marcusus57, 2023-02-21:
For the moment the only thing that works is EmuELEC 3.9 with the dtb file that you mention.  Lakka creators should understand what makes EmuELEC 3.9 work to patch their version.

ToKe79, 2023-02-21:
@Marcusus57 EmuELEC 3.9 might be using old (3.14?) patched kernel from Amlogic, it also uses separate package for the device trees (I might be wrong, just a quick view at the 3.9 tree). Lakka uses mainline kernel. You are not comparing apples to apples, therefore I suggested recent LibreELEC for testing, not a 2 year old release of EmuELEC.

Also Lakka is direct fork of LibreELEC, EmuELEC is fork of CoreELEC (aimed only for Amlogic devices), which is forked from LibreELEC. Both make their own changes to suit specific (i.e. Amlogic) devices.

Lakka follows LibreELEC for devices supported by LibreELEC (and adds support for additional devices, but keeping the build system as close as possible to LibreELEC), so if your device does not work with LibreELEC, it will not work with Lakka.

Also it is not possible for such a small group as Lakka to find a solution for each of the Amlogic devices, and there is plenty of them. Therefore we leave the maintenance of such devices to bigger groups, as is LibreELEC, and use this strong base for delivering RetroArch/libretro emulation. We are not able to debug each device included in the build system, but we provide these images with the hope they will be useful for some folks and get more usable with the future development in upstream (Kernel, LibreELEC).

Marcusus57, 2023-02-21:
Okay, thank you for this clarification, I will test the latest version of LIBREELEC and also "Lakka-AMLGX.aarch64-devbuild-v5.x-20230214-35e3e48-box.img.gz" and I will let you know!

Marcusus57, 2023-02-22:
H96PRO+ is the ugly duckling of emuelec and libreelec... :/
Libreelec 10.0.4 not working

Marcusus57, 2023-02-24:
> @Marcusus57 EmuELEC 3.9 might be using old (3.14?) patched kernel from Amlogic, it also uses separate package for the device trees (I might be wrong, just a quick view at the 3.9 tree). Lakka uses [...]

I'm disgusted:
The start system in Coreelec is not compatible with LibreELECE  Armbian  etc. If you run coreelec on your TV box, you will no longer be able to run LE and Armbian normally until the full recovery of the standard firmware via the USB Burn Tool and the new activation of the universal multi-boot, which is used in all new systems.