libretro/Lakka-LibreELEC #1834: Composite is not working properly.

thericeking, 2023-03-01:
If you are sure that it's a bug in Lakka, please answer the following questions:
- Which version of Lakka are you using?
- 
- What system hardware are you using? 
Raspbery Pi 3 B+

- What did you do?

Attempted composite out with instructions in the following link:

http://www.lakka.tv/doc/Raspberry-Pi/

Add video=Composite-1:720x480@60e to cmdline.txt right after quiet (separate these statements with a space).

Staring from Lakka 4.0 add following line to config.txt:

dtoverlay=vc4-kms-v3d,composite=1

- What did you expect to happen? Proper NTSC 640x480 (480i) to consumer CRT

- What happened instead? 

Works but I get black bars on the side when playing games because resolution output in the official documentation says to Add video=Composite-1:720x480@60e to cmdline.txt right after quiet

Even after changing the line to video=Composite-1:640x480@60e to cmdline.txt right after quiet

I don't get past the black screen and Lakka logo loading screen

gouchi, 2024-03-17:
Can you try to make a test with [Lakka 5.x nightly](https://nightly.builds.lakka.tv/devbuild_latest/RPi3.aarch64/) ?

Might be related to this [issue](https://github.com/libretro/Lakka-LibreELEC/issues/1932) also.

Thank you.

gouchi, 2024-05-06:
Hi,

Is it fixed with https://lakka.tv/articles/2024/05/02/rpi-composite/ ?

Thank you.

gouchi, 2024-05-11:
The build should fix your issue.

Please open a new issue if it is not case.

Thank you.

closed 2024-05-11 by gouchi