libretro/Lakka-LibreELEC #1840: when

amhooman, 2023-03-26:
I am using a raspberry pi 4, newest version

love768952693, 2023-04-03:
I also have this issue when using the Xbox One S controller in the X64 environment. I am using Lakka version 4.3, and the UI interface will get stuck after connecting. After restarting the system again, there will be no Bluetooth device d. It feels like the Bluetooth driver is not driving on Lakka, and when I enter the terminal to check, there is also no device status. I installed Batocera again and found that Bluetooth works.

I have another idea. I want to put the native Bluetooth driver of Batocera in Lakka. Does anyone know how to operate it?

amhooman, 2023-04-03:
Turns out it crashes on any bluetooth device .-.

amhooman, 2023-04-04:
I can't access command line easily. Is there a way to fix it to e menu?make it work on th

gouchi, 2024-07-15:
Hi @amhooman , 

When you say crash it means the menu is unresponsive ?

Anyway, can you try to make a test with [latest nightly](https://nightly.builds.lakka.tv/latest/RPi4.aarch64/) ?

If you still reproduce the issue, please provide some [log](https://lakka.tv/doc/Troubleshooting-Lakka/#automated-collection-of-relevant-logs) using `createlog`.

Thank you.