libretro/Lakka-LibreELEC #1846: 8BitDo Ultimate Wired Controller for Xbox not working

ciphermenial, 2023-04-21:
Version: Lakka 4.3
Hardware: Zotac Mini PC

When running 'lsusb' it shows the controller as 'Bus 002 Device 005: ID 2dc8:2002 8BitDo 8BitDo Ultimate Wired Controller for Xbox' but it does not work.

slowpokefarm, 2023-04-22:
I couldn't make it work on linux systems without installing https://github.com/medusalix/xone

ToKe79, 2024-04-11:
do you get a message in RetroArch about gamepad connected, but no autoconfig found? do you see the device as `/dev/input/jsX` (where X is a digit)? could you test with latest [development build for PC](https://nightly.builds.lakka.tv/devbuild_latest/Generic.x86_64/)?

Ntemis, 2024-07-10:
Can you retest on 5.0 stable? I give you a few days and closing this down.

gouchi, 2024-07-15:
@ciphermenial @slowpokefarm Please reply.

Thank you.

slowpokefarm, 2024-07-15:
@Ntemis @gouchi I possibly could as I have said controller at hand, could you point me to the image I could test it on? I have an x86 system and a few orange Pi systems at hand, can test it on any of those.

gouchi, 2024-07-15:
@slowpokefarm Thank you. 

You can get the stable release for Lakka 5
[https://lakka.tv/get/windows/](https://lakka.tv/get/windows/)

If still not working, you may try [latest nightlies](https://nightly.builds.lakka.tv/latest/).

Thank you

slowpokefarm, 2024-07-16:
@gouchi @Ntemis I have checked it on nightly build with orange Pi 3 LTS. Controller works without issues. The only unexpected behaviour is that the LED on controller is blinking as if it was trying to connect, but it's already connected and working.

Ntemis, 2024-07-16:
Excellent  closing  down

closed 2024-07-16 by Ntemis