libretro/Lakka-LibreELEC #1847: PCSX2, achievements can't be activated with this core

simonboy77, 2023-04-30:
**Which version of Lakka are you using?**
Generic.x86_64-4.3 (Build date Jan 16 2023)
and also tried
Generic.x86_64-devel-20230411030216-948d92b (Nightly downloaded on Apr 30 2023)

**What system hardware are you using?**
https://www.intel.com/content/www/us/en/products/sku/205603/intel-nuc-11-pro-kit-nuc11tnki5/specifications.html

**What did you do?**
I started TimeSplitters 2 using the PCSX2 core.

**What did you expect to happen?**
Being able to earn achievements.

**What happened instead?**
A popup message in the bottom informed me that "_achievements can't be activated with this core_". But the game ran great regardless, so I finished the first level knowing that that would give an achievement, but the achievement didn't unlock.

**What else did you try?**
Then I tried running the game on the nightly build on the same hardware with the same core (now named LRPS2), which gave the same popup message "_achievements can't be activated with this core_" and the same log messages regarding no exposed memory (logs below). I didn't bother finishing the first level to test.

Then I ran the linux PCSX2 emulator on my laptop with the same TimeSplitters 2 ISO, finished the first level and successfully unlocked the achievement for it.
<br/><br/>

<details>
<summary>Logs (From nightly build, but 4.3 reported the same)</summary>
[INFO] [Core]: Using content: "/storage/roms/PS2/TimeSplitters 2 (USA).iso".
[INFO] [Core]: Content ran for a total of: 00 hours, 00 minutes, 00 seconds.
[INFO] [Core]: Unloading core..
[INFO] [Core]: Unloading core symbols..
[INFO] Threaded video stats: Frames pushed: 518, Frames dropped: 0.
[INFO] RetroArch 1.14.0 (Git ad89b0c)
[INFO] === Build =======================================
[INFO] CPU Model Name: 11th Gen Intel(R) Core(TM) i5-1135G7 @ 2.40GHz
[INFO] Capabilities: MMX MMXEXT SSE SSE2 SSE3 SSE4 SSE4.2 AES AVX AVX2
[INFO] Built: Jan 29 2023
[INFO] Version: 1.14.0
[INFO] Git: ad89b0c
[INFO] =================================================
[INFO] [Input]: Found input driver: "udev".
[INFO] [Core]: Loading dynamic libretro core from: "/tmp/cores/pcsx2_libretro.so"
[INFO] [Environ]: SET_SUPPORT_NO_GAME: yes.
[INFO] [Overrides]: Redirecting save file to "/storage/savefiles/TimeSplitters 2 (USA).srm".
[INFO] [Overrides]: Redirecting save state to "/storage/savestates/TimeSplitters 2 (USA).state".
[INFO] [Environ]: SET_PIXEL_FORMAT: XRGB8888.
[INFO] [Environ]: GET_LOG_INTERFACE.
[INFO] [Environ]: GET_MESSAGE_INTERFACE_VERSION.
[INFO] [Environ]: SYSTEM_DIRECTORY: "/tmp/system".
[INFO] [Environ]: GET_SAVE_DIRECTORY.
[libretro INFO] Bios Found: USA     v01.90(23/06/2003)  Console
[INFO] [Environ]: GET_CORE_OPTIONS_VERSION.
[INFO] [Environ]: GET_LANGUAGE: "0".
[INFO] [Environ]: RETRO_ENVIRONMENT_SET_CORE_OPTIONS_V2_INTL.
[libretro INFO] Applying preset n: 1
[INFO] [Environ]: SET_DISK_CONTROL_EXT_INTERFACE.
[INFO] [Content]: Content loading skipped. Implementation will load it on its own.
[libretro INFO] Loading selected BIOS:  /tmp/system/pcsx2/bios/SCPH-50001.rom0
[INFO] [Environ]: SYSTEM_DIRECTORY: "/tmp/system".
[libretro INFO] Bios Found: USA     v01.90(23/06/2003)  Console
[libretro INFO] Detected BIOS: USAv01.90(23/06/2003)Console 
[libretro INFO] No injection data found for Bios SCPH-50001.rom0 - language: English 
[libretro WARN] Patches: No CRC found, using 00000000 instead.
[INFO] [Environ]: GET_RUMBLE_INTERFACE.
[INFO] [Environ]: SET_CONTROLLER_INFO.
[libretro INFO] Renderer option set to: Auto
[INFO] [Environ]: GET_PREFERRED_HW_RENDER, video driver name: glcore.
[INFO] [Environ]: GET_PREFERRED_HW_RENDER - Context callback set to RETRO_HW_CONTEXT_OPENGL_CORE.
[INFO] [Environ]: SET_HW_RENDER, context type: glcore.
[INFO] Requesting core OpenGL context (3.3).
[INFO] Reached end of SET_HW_RENDER.
[INFO] [RCHEEVOS]: Load started, hardcore active
[... 111 lines cut ...]
[INFO] [RCHEEVOS]: Fetched user unlocks 2624
[INFO] [RCHEEVOS]: Fetched game data 2624
[INFO] [RCHEEVOS]: Fetched hardcore user unlocks 2624
[INFO] [RCHEEVOS]: Starting periodic rich presence update task for game 2624
[INFO] [RCHEEVOS]: Could not map region starting at $000000
[INFO] [RCHEEVOS]: Registered 0x100000 bytes of SYSTEM RAM at $000000 (null filler)
[INFO] [RCHEEVOS]: Could not map region starting at $100000
[INFO] [RCHEEVOS]: Registered 0x1F00000 bytes of SYSTEM RAM at $100000 (null filler)
[INFO] [RCHEEVOS]: Could not map region starting at $2000000
[INFO] [RCHEEVOS]: Registered 0x4000 bytes of SYSTEM RAM at $2000000 (null filler)
[INFO] [RCHEEVOS]: Could not map region starting at $000000
[INFO] [RCHEEVOS]: Registered 0x100000 bytes of SYSTEM RAM at $000000 (null filler)
[INFO] [RCHEEVOS]: Could not map region starting at $100000
[INFO] [RCHEEVOS]: Registered 0x1F00000 bytes of SYSTEM RAM at $100000 (null filler)
[INFO] [RCHEEVOS]: Could not map region starting at $2000000
[INFO] [RCHEEVOS]: Registered 0x4000 bytes of SYSTEM RAM at $2000000 (null filler)
[ERROR] [RCHEEVOS]: No memory exposed by core
[INFO] [RCHEEVOS]: Asked the load tasks to terminate
[INFO] [RCHEEVOS]: You have 1 of 55 achievements unlocked.
[INFO] [RCHEEVOS]: Error fetching badge: Load aborted
[INFO] [RCHEEVOS]: Error fetching badge 245438: Load aborted
[INFO] [RCHEEVOS]: Error starting session for game 2624: Load aborted
[INFO] [KMS]: New FB: 1920x1080 (stride: 7680).
[INFO] [KMS]: New FB: 1920x1080 (stride: 7680).
</details>

Ntemis, 2023-04-30:
report here
https://github.com/libretro/pcsx2

closed 2023-04-30 by Ntemis

simonboy77, 2023-04-30:
Thank you, will do!

Ntemis, 2023-04-30:
actually i linked to the wrong repo
this is the correct one 
https://github.com/libretro/LRPS2