libretro/Lakka-LibreELEC #1849: Lakka USB does not boot on motherboard ga-g41mt-s2pt cpu pentium dual core e5500

MarcosDolce23, 2023-05-03:
I'm trying to install Lakka [Lakka-Generic.x86_64-nightly-v4.x-20230411-948d92b.img.gz](https://nightly.builds.lakka.tv/latest/Generic.x86_64/Lakka-Generic.x86_64-nightly-v4.x-20230411-948d92b.img.gz) on a PC with motherboard ga-g41mt-s2pt, a CPU pentium dual core e5500 with 2gb of RAM.

I burned the image on a kingston dtse9 8gb with Etcher but when I select the USB to boot it gives me a "disk error" and after that the PC boot the current OS installed.

gouchi, 2023-07-05:
Did you try with another image if you get the same issue ? Even with [nightly](https://nightly.builds.lakka.tv/devbuild_latest/Generic.x86_64/) ?

ToKe79, 2024-04-04:
no response.

closed 2024-04-04 by ToKe79