libretro/Lakka-LibreELEC #1852: Cannot Access Manage Playlists

Graham15, 2023-06-02:
Lakka 4.3

i5, 8GB ram and 1TB SSD. Using integrated graphics.

Creating playlists via Import Content, Manual Scan and selecting appropriate content directory, system name and default core. Once playlists are created, using Manage Playlists (settings, playlists) to up date (refresh) new roms.

Expecting the Manage Playlists to open and allow access to features within.

Displays Lakka logo briefly and returns back to main menu.

Graham15, 2023-06-09:
Issue also caused when using "Scan Directory" and seems to occur randomly.

:)

Vexnik, 2023-06-15:
This also happened on the Nintendo Switch build, after importing roms Lakka just crashed and restart the os preventing me from opening the Manage Playlist option. I thought I was the only one dealing with this issue.

Graham15, 2023-06-19:
Unfortunately, I have been unable to test the latest development build (v5, 11th Apr) to see if this issue has been fixed or not. Once installed, Lakka would not boot from my pc. The installation went through ok but after the reboot, would not recognize my hard disk as was trying to boot off other media (i.e pxe boot). 

I can also confirm, that having installed 4.2 prior to this, the issues didn't occur at all. Currently back on 4.3 again.

:)

amak79, 2023-07-17:
When I enable the `Settings->Logging->Log to File` option (`Logging Verbosity` is off), the problem goes away and I'm able to access `Manage Playlists`. The problem returns if I disable the option.

I'm using Lakka 4.3 x86_64.

Graham15, 2023-07-19:
I have since re-installed 4.3 multiple times with development builds inbetween and currently the "Manage Playlists" issue seems to have resolved itself. Thanks for the tip about logging to file. This will comes in handy should this re-appear again. Thanks.

:)

gouchi, 2023-09-19:
Closing this issue as it has been resolved.

closed 2023-09-19 by gouchi