libretro/Lakka-LibreELEC #1856: [enhancement] Partition Setup

Graham15, 2023-06-19:
First off, appologies if this is the wrong place to post this but here goes... :)

During installation, would it be possible for Lakka to instead of wiping the whole disk, look for the presence of an existing partition (like LAKKA_DISK in live version) and leave that partition alone. I'm sure some users will have huge collections of roms which would mean that if they needed to re-install Lakka, they would have to move these roms off first and then copy back later (this can be quite time consuming). For me, I only have a few but when I have had to re-install Lakka, this has taken longer than the installation. I understand that once Lakka is installed, it can be updated within itself but in some cases, a re-installation is required.

:)

Graham15, 2023-06-19:
I guess this could also apply to remappings, savefiles, savestates, thumbnails, etc. Anything that is added by the user after installation could be stored in this partition.

Regarding the config, maybe have a tool which could export/import these settings. Just an idea.

:)

ToKe79, 2024-04-02:
sorry, installation alongside other partitions is currently not being considered. same for backup/export/import of settings/saves/etc.

closed 2024-04-02 by ToKe79