libretro/Lakka-LibreELEC #1861: [4.3][PC] Hangs on Lakka screen indefinitely

Codes301, 2023-06-29:
Lakka 4.3 Stable / Generic PC / 64 bit

Hardware:
Beelink Mini S12
CPU: Intel 12th Gen Intel Processor Alder Lake-N95
RAM: single channel 8GB DDR4 RAM(Max 3200MHz)
SSD: M.2 PCIe 2280  - 256 GB 
SSD 2: Kingston 240 GB

hardware: [link](https://www.bee-link.com/catalog/product/index?id=450)

I installed Lakka OS on this device and it has not gone past the black Lakka OS screen. I've tested out one of the nightly images and that did not work either. Nightly tested (Lakka-Generic.x86_64-devbuild-v5.x-20230415-8199f1e). I've tested out installation on a 2.5 inch SSD drive and that drive did not go past the black Lakka screen either.

Another thing to note is that very quickly when booting into Lakka from bios is that a very quick flash of messages show in white before disappearing. 

I made out one of them with the following words: 
Spectre V2: WARNING: unprivileged ...
ACPI Error: AE_NOT_FOUND, During name lookup/catalog (20200 ...
ACPI bios error (bug) Could not resolve symbol ...

There's more lines of it but with my phone's camera messed up it's hard to view it.
Does this mean there are bios settings for these Beelink Mini PC's that are conflicting with it?

I kept only 2 logs so far.
[20230629014301.log](https://github.com/libretro/Lakka-LibreELEC/files/11901454/20230629014301.log)
[20230629015151.log](https://github.com/libretro/Lakka-LibreELEC/files/11901455/20230629015151.log)

gouchi, 2023-07-05:
May you try to boot in live mode using for the boot prompt `live ssh textmode` ?

From there, you should be able to get some log with [createlog](https://lakka.tv/doc/Troubleshooting-Lakka/#automated-collection-of-relevant-logs).

Thank you.

RobokingsCosmos, 2023-07-22:
Mine is a Ryzen 5 and currently having the same issue here, no idea on how to retrieve logs.

RobokingsCosmos, 2023-07-22:
16 GB of ram and Nvidia 3060 ti with 16x pcie

gouchi, 2023-07-22:
You can try with [Lakka 5.x](https://nightly.builds.lakka.tv/devbuild_latest/Generic.x86_64/), if you get better result but I doubt.

Your GPU is recent for the open source nouveau driver  so don't expect any good result for performance.

It would be better to have AMD or Intel GPU to get best result.

Thank you.

RobokingsCosmos, 2023-07-22:
> You can try with [Lakka 5.x](https://nightly.builds.lakka.tv/devbuild_latest/Generic.x86_64/), if you get better result but I doubt.
>
> Your GPU is recent for the open source nouveau driver  so don't [...]

Tried the installer but it shows syslinux then reboots loop. Not to mention it requires CSM to be turned on.

RobokingsCosmos, 2023-07-22:
I did however try to compile the latest build myself due to how outdated they are but can't due to failure.

gouchi, 2023-07-24:
Thank you for your tests.

> I did however try to compile the latest build myself due to how outdated they are but can't due to failure.

You might try [this build](https://nightly.builds.lakka.tv/members/dmrlawson/Lakka-Generic.x86_64-5.0-devel-20230719172834-7b936e4.img.gz).

Anyways, I am afraid for now we can't support your PC because of the GPU.

RobokingsCosmos, 2023-07-24:
> Thank you for your tests.
> 
> > I did however try to compile the latest build myself due to how outdated they are but can't due to failure.
> 
> You might try [this build](https://nightly.builds.lakka.tv/members/dmrlawson/Lakka-Generic.x86_64-5.0-devel-20230719172834-7b936e4.img.gz).
> 
> Anyways, I am afraid for now we can't support your PC because of the GPU using the nouveau open source driver.
speaking about the nouveau it seems to be the only driver to support sound on my ubuntu linux distro, while the others don't.

RobokingsCosmos, 2023-07-24:
here's the link to the video of me installing lakka (no sound)
https://youtu.be/jHuDUTYXskE

RobokingsCosmos, 2023-07-24:
It failed to copy the grub files from the drive.

gouchi, 2023-12-19:
Closing this issue as it should work with [Lakka 5.x X11 build](https://nightly.builds.lakka.tv/devbuild_latest/x11.x86_64/).

closed 2023-12-19 by gouchi