libretro/Lakka-LibreELEC #1868: Build linux:host fails

RobokingsCosmos, 2023-07-23:
This issue tracker is NOT for help or support. Please use one of the following channels for help and support:
- the Lakka forum (https://forums.libretro.com/c/libretro/lakka-tv-general)
- via IRC @ #lakkatv on freenode.net
- #lakkatv on libretro discord https://discord.gg/retroarch

Before you report a bug make sure you have tried the latest nightly version of Lakka at https://nightly.builds.lakka.tv/latest/. Your bug might be already fixed.

If you are sure that it's a bug in Lakka, please answer the following questions:
- 5.xx
- B550M-C
- Compile the source code
- Failure from something like this "https://codeload.github.com/intel/Intel-Linux-Processor-Microcode-Data-Files/tar.gz/refs/tags/microcode-20230613" ""
- Error patching makefile even manually doing it is a fail
`[1;36mGET[0m      linux (archive)
--2023-07-22 22:17:26--  https://www.kernel.org/pub/linux/kernel/v6.x/linux-6.1.38.tar.xz
Resolving www.kernel.org (www.kernel.org)... 2604:1380:4641:c500::1, 139.178.84.217
Connecting to www.kernel.org (www.kernel.org)|2604:1380:4641:c500::1|:443... connected.
HTTP request sent, awaiting response... 301 Moved Permanently
Location: https://mirrors.edge.kernel.org/pub/linux/kernel/v6.x/linux-6.1.38.tar.xz [following]
--2023-07-22 22:17:26--  https://mirrors.edge.kernel.org/pub/linux/kernel/v6.x/linux-6.1.38.tar.xz
Resolving mirrors.edge.kernel.org (mirrors.edge.kernel.org)... 2604:1380:45d1:ec00::1, 147.75.199.223
Connecting to mirrors.edge.kernel.org (mirrors.edge.kernel.org)|2604:1380:45d1:ec00::1|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 134926628 (129M) [application/x-xz]
Saving to: '/home/dmdomnic/Downloads/Lakka-LibreELEC/sources/linux/linux-6.1.38.tar.xz'

          /home/dmd   0%[                    ]       0  --.-KB/s               
         /home/dmdo   0%[                    ] 343.61K  1.48MB/s               
        /home/dmdom   1%[                    ]   2.32M  5.44MB/s               
       /home/dmdomn   3%[                    ]   4.77M  7.60MB/s               
      /home/dmdomni   5%[>                   ]   6.63M  7.97MB/s               
     /home/dmdomnic   7%[>                   ]   9.07M  8.79MB/s               
    /home/dmdomnic/   9%[>                   ]  11.82M  9.56MB/s               
   /home/dmdomnic/D  11%[=>                  ]  14.44M  10.0MB/s               
  /home/dmdomnic/Do  13%[=>                  ]  17.24M  10.5MB/s               
 /home/dmdomnic/Dow  15%[==>                 ]  19.85M  10.8MB/s               
/home/dmdomnic/Down  17%[==>                 ]  22.46M  11.0MB/s               
home/dmdomnic/Downl  19%[==>                 ]  24.93M  11.1MB/s               
ome/dmdomnic/Downlo  21%[===>                ]  27.30M  11.1MB/s               
me/dmdomnic/Downloa  23%[===>                ]  29.71M  11.2MB/s               
e/dmdomnic/Download  24%[===>                ]  31.96M  11.2MB/s               
/dmdomnic/Downloads  26%[====>               ]  34.27M  11.2MB/s    eta 8s     
dmdomnic/Downloads/  28%[====>               ]  36.54M  11.8MB/s    eta 8s     
mdomnic/Downloads/L  30%[=====>              ]  38.81M  12.0MB/s    eta 8s     
domnic/Downloads/La  31%[=====>              ]  40.96M  11.9MB/s    eta 8s     
omnic/Downloads/Lak  33%[=====>              ]  43.24M  12.0MB/s    eta 8s     
mnic/Downloads/Lakk  35%[======>             ]  45.82M  12.2MB/s    eta 7s     
nic/Downloads/Lakka  37%[======>             ]  48.08M  12.1MB/s    eta 7s     
ic/Downloads/Lakka-  39%[======>             ]  50.46M  11.9MB/s    eta 7s     
c/Downloads/Lakka-L  41%[=======>            ]  52.83M  11.9MB/s    eta 7s     
/Downloads/Lakka-Li  42%[=======>            ]  55.29M  11.8MB/s    eta 7s     
Downloads/Lakka-Lib  44%[=======>            ]  57.76M  11.7MB/s    eta 6s     
ownloads/Lakka-Libr  46%[========>           ]  59.68M  11.6MB/s    eta 6s     
wnloads/Lakka-Libre  47%[========>           ]  61.73M  11.4MB/s    eta 6s     
nloads/Lakka-LibreE  49%[========>           ]  63.60M  11.2MB/s    eta 6s     
loads/Lakka-LibreEL  51%[=========>          ]  65.66M  11.2MB/s    eta 6s     
oads/Lakka-LibreELE  52%[=========>          ]  67.73M  11.1MB/s    eta 5s     
ads/Lakka-LibreELEC  54%[=========>          ]  69.73M  11.0MB/s    eta 5s     
ds/Lakka-LibreELEC/  55%[==========>         ]  71.63M  10.9MB/s    eta 5s     
s/Lakka-LibreELEC/s  57%[==========>         ]  73.87M  10.9MB/s    eta 5s     
/Lakka-LibreELEC/so  59%[==========>         ]  76.27M  11.0MB/s    eta 5s     
Lakka-LibreELEC/sou  61%[===========>        ]  78.52M  10.9MB/s    eta 5s     
akka-LibreELEC/sour  62%[===========>        ]  80.68M  10.8MB/s    eta 5s     
kka-LibreELEC/sourc  64%[===========>        ]  82.94M  10.8MB/s    eta 5s     
ka-LibreELEC/source  66%[============>       ]  85.02M  10.7MB/s    eta 5s     
a-LibreELEC/sources  67%[============>       ]  87.29M  10.6MB/s    eta 5s     
-LibreELEC/sources/  69%[============>       ]  89.21M  10.4MB/s    eta 4s     
LibreELEC/sources/l  70%[=============>      ]  91.29M  10.4MB/s    eta 4s     
ibreELEC/sources/li  72%[=============>      ]  93.46M  10.5MB/s    eta 4s     
breELEC/sources/lin  74%[=============>      ]  95.46M  10.5MB/s    eta 4s     
[... 54 lines cut ...]
patching file drivers/hid/Kconfig
Hunk #1 succeeded at 1290 with fuzz 2 (offset -2 lines).
patching file drivers/hid/Makefile
Hunk #1 FAILED at 166.
1 out of 1 hunk FAILED -- saving rejects to file drivers/hid/Makefile.rej
patching file drivers/hid/ithc/Kbuild
patching file drivers/hid/ithc/Kconfig
patching file drivers/hid/ithc/ithc-debug.c
patching file drivers/hid/ithc/ithc-dma.c
patching file drivers/hid/ithc/ithc-dma.h
patching file drivers/hid/ithc/ithc-main.c
patching file drivers/hid/ithc/ithc-regs.c
patching file drivers/hid/ithc/ithc-regs.h
patching file drivers/hid/ithc/ithc.h
*********** FAILED COMMAND ***********
patch -d "${PKG_BUILD}" -p1 < ${i} >&${VERBOSE_OUT}
**************************************
*********** FAILED COMMAND ***********
${SCRIPTS}/unpack "${PKG_NAME}" "${PARENT_PKG}"
**************************************
[1;31mFAILURE: scripts/build linux:host has failed![0m`

It is important to provide logs for debugging as part of your report. Instructions for generating logs can be found in the Troubleshooting Lakka documentation at https://www.lakka.tv/doc/Troubleshooting-Lakka/

Please delete this text before posting.

Ntemis, 2023-07-23:
Thanks for the report , the patch was recently added for trial purposes as supposedly should be compatible with our 6.1 kernel. I will remove it soon

RobokingsCosmos, 2023-07-23:
Will there be the package nvidia-driver-535 implementation on Lakka 6.1? Or can I include it myself?

Ntemis, 2023-07-24:
issue should be fixed, check and close

RobokingsCosmos, 2023-07-25:
I'll close and say that it's still not fixed on ecwolf after reinstallation and recompilation.

closed 2023-07-25 by RobokingsCosmos