libretro/Lakka-LibreELEC #1869: Build fails for RPi4 on nss building

Eskuero, 2023-08-04:
Running this building command under the 5.x branch:

`DISTRO=Lakka PROJECT=RPi DEVICE=RPi4 ARCH=aarch64 make image`

I get the following error. It also happens when building for arm arch.

```
make[4]: Leaving directory '/buildir/Lakka-LibreELEC/build.Lakka-RPi4.aarch64-5.0-devel/build/nss-3.89/nss/cmd/bltest'
make[4]: Entering directory '/buildir/Lakka-LibreELEC/build.Lakka-RPi4.aarch64-5.0-devel/build/nss-3.89/nss/cmd/ecperf'
/buildir/Lakka-LibreELEC/build.Lakka-RPi4.aarch64-5.0-devel/toolchain/bin/host-gcc -o Linux6.4_x86_64_gcc_glibc_PTH_64_OPT.OBJ/ecperf.o -c -std=c99 -O2 -fPIC  -m64 -pipe -ffunction-sections -fdata-sections -DHAVE_STRERROR -DLINUX -Dlinux -Wall -Wshadow -Werror -DXP_UNIX -DXP_UNIX -UDEBUG -DNDEBUG -D_DEFAULT_SOURCE -D_BSD_SOURCE -D_POSIX_SOURCE -DSDB_MEASURE_USE_TEMP_DIR -D_REENTRANT -UDEBUG -DNDEBUG -D_DEFAULT_SOURCE -D_BSD_SOURCE -D_POSIX_SOURCE -DSDB_MEASURE_USE_TEMP_DIR -D_REENTRANT -DNSS_NO_INIT_SUPPORT -DUSE_UTIL_DIRECTLY -DNO_NSPR_10_SUPPORT -DSSL_DISABLE_DEPRECATED_CIPHER_SUITE_NAMES -DNSS_USE_STATIC_LIBS -I/buildir/Lakka-LibreELEC/build.Lakka-RPi4.aarch64-5.0-devel/toolchain/include -I/buildir/Lakka-LibreELEC/build.Lakka-RPi4.aarch64-5.0-devel/toolchain/include/nspr -I./../dist/Linux6.4_x86_64_gcc_glibc_PTH_64_OPT.OBJ/include -I./../dist/public/ -I./../dist/private/ -I/buildir/Lakka-LibreELEC/build.Lakka-RPi4.aarch64-5.0-devel/toolchain/include/nspr -I./../dist/Linux6.4_x86_64_gcc_glibc_PTH_64_OPT.OBJ/include -I./../dist/public/ -I./../dist/private/ -I/buildir/Lakka-LibreELEC/build.Lakka-RPi4.aarch64-5.0-devel/toolchain/include/nspr -I../../dist/Linux6.4_x86_64_gcc_glibc_PTH_64_OPT.OBJ/include -I../../dist/public/ -I../../dist/private/ -I../../dist/public/nss -I../../dist/public/nspr -I../../dist/public/libdbm -I../../dist/Linux6.4_x86_64_gcc_glibc_PTH_64_OPT.OBJ/../public/security -I./include  -I../../nss/lib/softoken -I/buildir/Lakka-LibreELEC/build.Lakka-RPi4.aarch64-5.0-devel/toolchain/include/nspr -I../../../dist/Linux6.4_x86_64_gcc_glibc_PTH_64_OPT.OBJ/include -I../../../dist/public/nss -I../../../dist/private/nss  ecperf.c
ecperf.c: In function 'genericThread':
ecperf.c:99:24: error: storing the address of local variable 'sig' in '*threadData.p2' [-Werror=dangling-pointer=]
   99 |         threadData->p2 = (void *)&sig;
      |         ~~~~~~~~~~~~~~~^~~~~~~~~~~~~~
ecperf.c:91:13: note: 'sig' declared here
   91 |     SECItem sig;
      |             ^~~
ecperf.c:86:21: note: 'data' declared here
   86 | genericThread(void *data)
      |               ~~~~~~^~~~
ecperf.c: In function 'PKCS11Thread':
ecperf.c:71:24: error: storing the address of local variable 'sig' in '*threadData.p2' [-Werror=dangling-pointer=]
   71 |         threadData->p2 = (void *)&sig;
      |         ~~~~~~~~~~~~~~~^~~~~~~~~~~~~~
ecperf.c:53:13: note: 'sig' declared here
   53 |     SECItem sig;
      |             ^~~
ecperf.c:47:20: note: 'data' declared here
   47 | PKCS11Thread(void *data)
      |              ~~~~~~^~~~
cc1: all warnings being treated as errors
make[4]: *** [../../coreconf/rules.mk:292: Linux6.4_x86_64_gcc_glibc_PTH_64_OPT.OBJ/ecperf.o] Error 1
make[4]: Leaving directory '/buildir/Lakka-LibreELEC/build.Lakka-RPi4.aarch64-5.0-devel/build/nss-3.89/nss/cmd/ecperf'
make[3]: *** [../coreconf/rules.mk:44: ecperf] Error 2
make[3]: Leaving directory '/buildir/Lakka-LibreELEC/build.Lakka-RPi4.aarch64-5.0-devel/build/nss-3.89/nss/cmd'
make[2]: *** [coreconf/rules.mk:44: cmd] Error 2
make[2]: Leaving directory '/buildir/Lakka-LibreELEC/build.Lakka-RPi4.aarch64-5.0-devel/build/nss-3.89/nss'
make[1]: *** [manifest.mn:21: all] Error 2
make[1]: Leaving directory '/buildir/Lakka-LibreELEC/build.Lakka-RPi4.aarch64-5.0-devel/build/nss-3.89/nss'
[1;31mFAILURE: scripts/build nss:host during make_host (package.mk)[0m
*********** FAILED COMMAND ***********
INCLUDES="-I${TOOLCHAIN}/include" make BUILD_OPT=1 USE_64=1 PREFIX=${TOOLCHAIN} NSPR_INCLUDE_DIR=${TOOLCHAIN}/include/nspr USE_SYSTEM_ZLIB=1 ZLIB_LIBS="-lz -L${TOOLCHAIN}/lib" SKIP_SHLIBSIGN=1 NSS_TESTS="dummy" CC=${CC} LDFLAGS="${LDFLAGS} -L${TOOLCHAIN}/lib" V=1
**************************************
[1;31mFAILURE: scripts/build nss:host has failed![0m

```

Ntemis, 2023-08-04:
what operating system?

Eskuero, 2023-08-04:
Arch Linux x64

Ntemis, 2023-08-05:
Not supported try debian/ubuntu

closed 2023-08-05 by Ntemis

Eskuero, 2023-08-05:
Remember to update the docs as well then, since they point otherwise.
https://www.lakka.tv/doc/Compiling-Lakka/

Ntemis, 2023-08-05:
No one never does, feel free to pr the changes