libretro/Lakka-LibreELEC #1873: [Request] Update cores

zzedmore, 2023-09-03:
Hi, i'm using the latest dev build for Raspberry Pi 4 64bit which is `Lakka-RPi4.aarch64-devbuild-v5.x-20230813-afdaa47.img.gz`. Only thing is that the PUAE core is v4.9.1 b2c4acb which is from January ! I'm only interested in the Amiga (PUAE) core and the C64 (vice) core. Both of these have had a TON of updates since January.

Ntemis, 2023-09-05:
all cores were updated recently but i will take a look, thanks for the ping

closed 2023-09-05 by Ntemis

zzedmore, 2023-09-05:
@Ntemis Any idea when a new Lakka RPi 4 64bit image will be built ? Updates to the cores don't really mean much to me unless I have an image to write to an SD card 😂

Ntemis, 2023-09-05:
Build server is down so no ETA

zzedmore, 2023-09-06:
As these Lakka images only get compiled about once a year, could you also update the PUAE2021 and VICE cores please ? They've had some useful updates for when I finally get round to using them! Although, all the cores will probably be out of date by the time the images are made 😢