libretro/Lakka-LibreELEC #1874: Compiling error related to retroarch_joypad_autoconfig

GabrielAnacletoBaumgartner, 2023-09-03:
I was trying to compile lakka with X11/nvidia-drivers following the **[#640 pull from the old repositories](https://github.com/libretro/Lakka/pull/640)** but with any branch i tried (4.x or 5.x) i got an error in the 10th step.

It seems related to the libretro optional package "retroarch_joypad_autoconfig". It stops compiling right after trying to remove a cfg file named "Nintendo - Switch Pro Controller (bare)", I checked, and in fact there is only a file named "Nintendo - Switch Pro Controller (bare)"

This is the log.10 file:

[1;36mUNPACK[0m      retroarch_joypad_autoconfig
[1;33mBUILD[0m      retroarch_joypad_autoconfig [1;37m(target)[0m
    [1;35mTOOLCHAIN[0m      manual
make[1]: Entering directory '/home/gabriellaptop/Lakka-LibreELEC/build.Lakka-Generic.x86_64-5.0-devel/build/retroarch_joypad_autoconfig-5666e46bb89caf4e9af358fdb97a2b384cb62f36'
for driver in android dinput hid linuxraw mfi parport qnx sdl2 udev x xinput; do \
	install -Dm644 -t /home/gabriellaptop/Lakka-LibreELEC/build.Lakka-Generic.x86_64-5.0-devel/install_pkg/retroarch_joypad_autoconfig-5666e46bb89caf4e9af358fdb97a2b384cb62f36/etc/retroarch-joypad-autoconfig/$driver $driver/*.cfg; \
done
install -Dm644 -t /home/gabriellaptop/Lakka-LibreELEC/build.Lakka-Generic.x86_64-5.0-devel/install_pkg/retroarch_joypad_autoconfig-5666e46bb89caf4e9af358fdb97a2b384cb62f36/etc/doc/. COPYING README.md retropad_layout.png
make[1]: Leaving directory '/home/gabriellaptop/Lakka-LibreELEC/build.Lakka-Generic.x86_64-5.0-devel/build/retroarch_joypad_autoconfig-5666e46bb89caf4e9af358fdb97a2b384cb62f36'
rm: cannot remove '/home/gabriellaptop/Lakka-LibreELEC/build.Lakka-Generic.x86_64-5.0-devel/install_pkg/retroarch_joypad_autoconfig-5666e46bb89caf4e9af358fdb97a2b384cb62f36/etc/retroarch-joypad-autoconfig/udev/Nintendo - Switch Pro Controller (bare).cfg': No such file or directory
[1;31mFAILURE: scripts/build retroarch_joypad_autoconfig:target during makeinstall_target (package.mk)[0m
*********** FAILED COMMAND ***********
rm "${INSTALL}"/etc/retroarch-joypad-autoconfig/udev/Nintendo\ -\ Switch\ Pro\ Controller\ \(bare\).cfg
**************************************
*********** FAILED COMMAND ***********
${SCRIPTS}/build "${1}" "${PARENT_PKG}"
**************************************
[1;31mFAILURE: scripts/install retroarch_joypad_autoconfig:target has failed![0m

Note: I'm totally new at Linux, but i'm using Ubuntu as recommended and followed every step of [Compiling Lakka page](https://www.lakka.tv/doc/Compiling-Lakka/).

ShigeakiAsai, 2023-09-03:
Hello, All

I have same problem in lakka build.

I think this compile error concerns to "Update all lakka packages Retroach latest + cores"(5b413c99f6452db02c889b1018c91660876a41f8)"'s commit.

I was checking following package commit.
PKG_NAME="retroarch_joypad_autoconfig"
PKG_SITE="https://github.com/libretro/retroarch-joypad-autoconfig"

Target version is latest commit(5666e46bb89caf4e9af358fdb97a2b384cb62f36).
But, previous commit(ccd5bd3820969e1236f6eb32f90ed2f7d95ca06f) conteins filename changing.
bare -> old

Therefore, I thnk it needs to change to replaced filename in package.mk
packages/lakka/retroarch_base/retroarch_joypad_autoconfig/package.mk

Thanks,
ASAI, Shigeaki

GabrielAnacletoBaumgartner, 2023-09-04:
> Hello, All
> 
> I have same problem in lakka build.
> 
> I think this compile error concerns to "Update all lakka packages Retroach latest + cores"([5b413c9](https://github.com/libretro/Lakka-LibreELEC/commit/5b413c99f6452db02c889b1018c91660876a41f8))"'s commit.
> 
> I was checking following package commit. PKG_NAME="retroarch_joypad_autoconfig" PKG_SITE="https://github.com/libretro/retroarch-joypad-autoconfig"
> 
> Target version is latest commit(5666e46bb89caf4e9af358fdb97a2b384cb62f36). But, previous commit(ccd5bd3820969e1236f6eb32f90ed2f7d95ca06f) conteins filename changing. bare -> old
> 
> Therefore, I thnk it needs to change to replaced filename in package.mk packages/lakka/retroarch_base/retroarch_joypad_autoconfig/package.mk
> 
> Thanks, ASAI, Shigeaki

Thanks!!! It worked! But sadly i got a new error ;-;
step 15

Ntemis, 2023-09-05:
https://github.com/libretro/Lakka-LibreELEC/commit/d47fc019cbdf3ba16af905852b51c4cbd3c3b815

closed 2023-09-05 by Ntemis