libretro/Lakka-LibreELEC #1875: Compiling issue related to libtool object

GabrielAnacletoBaumgartner, 2023-09-04:
I'm trying to compile an X11 Lakka v5.x, i'm running Ubuntu as recommended i'm getting an error at step 15.

As the log is very big, this is the last part:

mv unistd.h-t unistd.h
sed -e 1h -e '1s,.*,/* DO NOT EDIT! GENERATED AUTOMATICALLY! */,' -e 1G \
      -e 's|@''GUARD_PREFIX''@|GL|g' \
      -e 's/@''HAVE_WCTYPE_H''@/1/g' \
      -e 's|@''INCLUDE_NEXT''@|include_next|g' \
      -e 's|@''PRAGMA_SYSTEM_HEADER''@|#pragma GCC system_header|g' \
      -e 's|@''PRAGMA_COLUMNS''@||g' \
      -e 's|@''NEXT_WCTYPE_H''@|<wctype.h>|g' \
      -e 's/@''HAVE_CRTDEFS_H''@/0/g' \
      -e 's/@''GNULIBHEADERS_OVERRIDE_WINT_T''@/0/g' \
      -e 's/@''GNULIB_ISWBLANK''@/1/g' \
      -e 's/@''GNULIB_ISWDIGIT''@/1/g' \
      -e 's/@''GNULIB_ISWXDIGIT''@/1/g' \
      -e 's/@''GNULIB_WCTYPE''@/0/g' \
      -e 's/@''GNULIB_ISWCTYPE''@/0/g' \
      -e 's/@''GNULIB_WCTRANS''@/0/g' \
      -e 's/@''GNULIB_TOWCTRANS''@/0/g' \
      -e 's/@''HAVE_ISWBLANK''@/1/g' \
      -e 's/@''HAVE_ISWCNTRL''@/1/g' \
      -e 's/@''HAVE_WCTYPE_T''@/1/g' \
      -e 's/@''HAVE_WCTRANS_T''@/1/g' \
      -e 's/@''HAVE_WINT_T''@/1/g' \
      -e 's/@''REPLACE_ISWBLANK''@/0/g' \
      -e 's/@''REPLACE_ISWDIGIT''@/0/g' \
      -e 's/@''REPLACE_ISWXDIGIT''@/0/g' \
      -e 's/@''REPLACE_ISWCNTRL''@/0/g' \
      -e 's/@''REPLACE_TOWLOWER''@/0/g' \
      -e '/definitions of _GL_FUNCDECL_RPL/r /home/gabriellaptop/Lakka-LibreELEC/build.Lakka-Generic.x86_64-5.0-devel/build/gettext-0.21.1/gettext-tools/gnulib-lib/c++defs.h' \
      -e '/definition of _GL_WARN_ON_USE/r /home/gabriellaptop/Lakka-LibreELEC/build.Lakka-Generic.x86_64-5.0-devel/build/gettext-0.21.1/gettext-tools/gnulib-lib/warn-on-use.h' \
      /home/gabriellaptop/Lakka-LibreELEC/build.Lakka-Generic.x86_64-5.0-devel/build/gettext-0.21.1/gettext-tools/gnulib-lib/wctype.in.h > wctype.h-t
mv wchar.h-t wchar.h
mv wctype.h-t wctype.h
make  all-am
make[6]: Entering directory '/home/gabriellaptop/Lakka-LibreELEC/build.Lakka-Generic.x86_64-5.0-devel/build/gettext-0.21.1/.x86_64-linux-gnu/gettext-tools/gnulib-lib'
/bin/bash ../libtool  --tag=CC   --mode=link /home/gabriellaptop/Lakka-LibreELEC/build.Lakka-Generic.x86_64-5.0-devel/build/gettext-0.21.1/.x86_64-linux-gnu/libreelec-local-cc  -march=native -O2 -Wall -pipe -I/home/gabriellaptop/Lakka-LibreELEC/build.Lakka-Generic.x86_64-5.0-devel/toolchain/include -Wno-format-security  -Wl,-rpath,/home/gabriellaptop/Lakka-LibreELEC/build.Lakka-Generic.x86_64-5.0-devel/toolchain/lib -L/home/gabriellaptop/Lakka-LibreELEC/build.Lakka-Generic.x86_64-5.0-devel/toolchain/lib -o libxml_rpl.la  libxml/rpl_la-DOCBparser.lo libxml/rpl_la-HTMLparser.lo libxml/rpl_la-HTMLtree.lo libxml/rpl_la-SAX.lo libxml/rpl_la-SAX2.lo libxml/rpl_la-buf.lo libxml/rpl_la-c14n.lo libxml/rpl_la-catalog.lo libxml/rpl_la-chvalid.lo libxml/rpl_la-debugXML.lo libxml/rpl_la-dict.lo libxml/rpl_la-encoding.lo libxml/rpl_la-entities.lo libxml/rpl_la-error.lo libxml/rpl_la-globals.lo libxml/rpl_la-hash.lo libxml/rpl_la-legacy.lo libxml/rpl_la-list.lo libxml/rpl_la-nanoftp.lo libxml/rpl_la-nanohttp.lo libxml/rpl_la-parser.lo libxml/rpl_la-parserInternals.lo libxml/rpl_la-pattern.lo libxml/rpl_la-relaxng.lo libxml/rpl_la-schematron.lo libxml/rpl_la-threads.lo libxml/rpl_la-tree.lo libxml/rpl_la-trionan.lo libxml/rpl_la-uri.lo libxml/rpl_la-valid.lo libxml/rpl_la-xinclude.lo libxml/rpl_la-xlink.lo libxml/rpl_la-xmlIO.lo libxml/rpl_la-xmlmemory.lo libxml/rpl_la-xmlmodule.lo libxml/rpl_la-xmlreader.lo libxml/rpl_la-xmlregexp.lo libxml/rpl_la-xmlsave.lo libxml/rpl_la-xmlschemas.lo libxml/rpl_la-xmlschemastypes.lo libxml/rpl_la-xmlstring.lo libxml/rpl_la-xmlunicode.lo libxml/rpl_la-xmlwriter.lo libxml/rpl_la-xpath.lo libxml/rpl_la-xpointer.lo  
libtool:   error: 'libxml/rpl_la-DOCBparser.lo' is not a valid libtool object
make[6]: *** [Makefile:3896: libxml_rpl.la] Error 1
make[6]: Leaving directory '/home/gabriellaptop/Lakka-LibreELEC/build.Lakka-Generic.x86_64-5.0-devel/build/gettext-0.21.1/.x86_64-linux-gnu/gettext-tools/gnulib-lib'
make[5]: *** [Makefile:3661: all] Error 2
make[5]: Leaving directory '/home/gabriellaptop/Lakka-LibreELEC/build.Lakka-Generic.x86_64-5.0-devel/build/gettext-0.21.1/.x86_64-linux-gnu/gettext-tools/gnulib-lib'
make[4]: *** [Makefile:3029: all-recursive] Error 1
make[4]: Leaving directory '/home/gabriellaptop/Lakka-LibreELEC/build.Lakka-Generic.x86_64-5.0-devel/build/gettext-0.21.1/.x86_64-linux-gnu/gettext-tools'
make[3]: *** [Makefile:2897: all] Error 2
make[3]: Leaving directory '/home/gabriellaptop/Lakka-LibreELEC/build.Lakka-Generic.x86_64-5.0-devel/build/gettext-0.21.1/.x86_64-linux-gnu/gettext-tools'
make[2]: *** [Makefile:406: all-recursive] Error 1
make[2]: Leaving directory '/home/gabriellaptop/Lakka-LibreELEC/build.Lakka-Generic.x86_64-5.0-devel/build/gettext-0.21.1/.x86_64-linux-gnu'
make[1]: *** [Makefile:362: all] Error 2
make[1]: Leaving directory '/home/gabriellaptop/Lakka-LibreELEC/build.Lakka-Generic.x86_64-5.0-devel/build/gettext-0.21.1/.x86_64-linux-gnu'
[1;31mFAILURE: scripts/build gettext:host during make_host (default)[0m
*********** FAILED COMMAND ***********
make ${PKG_MAKE_OPTS_HOST}
**************************************
[1;31mFAILURE: scripts/build gettext:host has failed![0m

I didn't modified anything, just followed [Compiling Lakka](https://www.lakka.tv/doc/Compiling-Lakka/) Page and [#640 pull from the old repositories](https://github.com/libretro/Lakka/pull/640)

ShigeakiAsai, 2023-09-05:
Hello, @GabrielAnacletoBaumgartner

I could not reproduce this issue.
It may be happened on environment.

Please confirm the version of libtool.
Or, is libtool installed your environment?

My build environment:
hoge@PCname:~$ libtool --version
Command 'libtool' not found, but can be installed with:
sudo apt install libtool-bin

hoge@PCname:~$ .//Lakka-LibreELEC/build.Lakka-Generic.x86_64-5.0-devel/toolchain/bin/libtool --version
libtool (GNU libtool) 2.4.7
Written by Gordon Matzigkeit, 1996

Copyright (C) 2014 Free Software Foundation, Inc.

Note.
I changed username and hostname.

Thanks
ASAI, Shigeaki

ToKe79, 2024-04-02:
is this still an issue? we do not face this issue on the devel builds. if so, feel free to reopen with comment on current status.