libretro/Lakka-LibreELEC #1877: Compiling errors (AGAIN!)

GabrielAnacletoBaumgartner, 2023-09-07:
Ok, i've been trying to compile a custom X11-nividia Lakka build for 5 DAYS! And i'm still getting errors.

I already [had issues compiling the system](https://github.com/libretro/Lakka-LibreELEC/issues/1875) but now i made a few changes.

I edited the name, logo, splash and options file in the Lakka-LibreELEC/distributions/Lakka, v4 and v5. But even before editing i was getting errors. If this is my environment's fault, you should provide the "install all dependencies script" script, after all the Readme.MD is COMPLETELY OUTDATED.

Note the compiling errors i'm getting are consistent even with no modifications (simply just cloned and compiled) to the previous mentioned files.

Basically what i added to the options was this:

`
  EXCLUDE_LIBRETRO_CORES+=" stella2014 mednafen_lynx atari800"
  #Atari
  EXCLUDE_LIBRETRO_CORES+=" hbmame mame2000 mame2003 mame2003_midway mame2003_plus mame2009 mame2010 mame2015 mame2016"
  #Arcade
  EXCLUDE_LIBRETRO_CORES+=" mednafen_pce_fast mednafen_supergrafx quasi88 np2kai nekop2"
  #NECpcEngine
  EXCLUDE_LIBRETRO_CORES+=" desmume2015"
  #DS
  EXCLUDE_LIBRETRO_CORES+=" quicknes bnes emux_chip8 emux_gb emux_nes emux_sms fceumm fixnes"
  #NES
  EXCLUDE_LIBRETRO_CORES+=" snes9x2002 snes9x2005 snes9x2005_plus snes9x2010 bsnes2014_accuracy bsnes2014_balanced bsnes2014_performance bsnes_cplusplus98 bsnes_hd_beta bsnes bsnes_mercury_accuracy bsnes_mercury_balanced bsnes_mercury_performance mednafen_snes mednafen_supafaust higan_sfc_balanced higan_sfc mesen-s"
  #SNES
  EXCLUDE_LIBRETRO_CORES+=" sameboy tgbdual gearboy meteor fixgb gambatte tempgba vba_next vbam mednafen_gba"
  #GBA
  EXCLUDE_LIBRETRO_CORES+=" dosbox_core dosbox dosbox_pure dosbox_svn_ce dosbox_svn minivmac gpsp mu px68k x1 81 fuse theodore cap32 crocods puae2021 puae uae4arm vice_x128 vice_x64 vice_x64sc vice_xcbm2 vice_xcbm5x0 vice_xpet vice_xplus4 vice_xscpu64 vice_xvic"
  #RetroOS
  EXCLUDE_LIBRETRO_CORES+=" genesis_plus_gx_wide picodrive gearsystem"
  #Genesis
  EXCLUDE_LIBRETRO_CORES+=" race"
  #Neogeo
  EXCLUDE_LIBRETRO_CORES+=" cannonball craft prboom nxengine dinothawr mrboom xrick"
  #Singlecores
  EXCLUDE_LIBRETRO_CORES+=" tyrquake vitaquake2-rogue vitaquake2-xatrix vitaquake2-zaero vitaquake2 vitaquake3 boom3 boom3_xp gong opentyrian pascal_pong"
  #GameClone
  EXCLUDE_LIBRETRO_CORES+=" openlara chailove ecwolf ecwolf puzzlescript jumpnbump lowresnx lutro reminiscence scummvm superbroswar tic80 vaporspec vitavoyager wasm4 easyrpg cruzes"
  #GameEngine
  EXCLUDE_LIBRETRO_CORES+=" mpv ffmpeg pocketcdg gme moonlight"
  #Multimedia
  `
This way it would remove the cores i won't use. And yes, i inserted these lines before _# exclude some cores at build time_
The changes to the name and to the icons/splash is purely due to aesthetics (and i don't know how it could affect compiling)

With Lakka v5.x i get errors at step 258:

`
[1;31mCLEAN[0m      mame
    [1;31m*[0m      [0;31mRemoving /home/gabriellaptop/Lakka-LibreELEC/build.DreamOS-Generic.x86_64-5.0-devel/build/mame-1808a1cc4bf9148505bc97a3f392b55e6121993c ...[0m
    [1;31m*[0m      [0;31mRemoving /home/gabriellaptop/Lakka-LibreELEC/build.DreamOS-Generic.x86_64-5.0-devel/qa_checks/mame-* ...[0m
[1;36mUNPACK[0m      mame
    [1;32mAPPLY PATCH [1;37m(common)[0m[0m      packages/lakka/libretro_cores/mame/patches/mame-crosscompile.patch
patching file 3rdparty/genie/build/gmake.linux/genie.make
patching file 3rdparty/genie/src/host/scripts.c
Hunk #1 succeeded at 147 with fuzz 1.
patching file scripts/toolchain.lua
Hunk #1 succeeded at 335 (offset -363 lines).
    [1;32mAPPLY PATCH [1;37m(common)[0m[0m      packages/lakka/libretro_cores/mame/patches/mame-remove-bgfx.patch
patching file Makefile.libretro
Hunk #1 succeeded at 79 with fuzz 2 (offset 24 lines).
patching file scripts/genie.lua
Hunk #1 succeeded at 437 with fuzz 1 (offset 50 lines).
patching file scripts/src/3rdparty.lua
Hunk #1 FAILED at 1249.
Hunk #2 FAILED at 1433.
2 out of 2 hunks FAILED -- saving rejects to file scripts/src/3rdparty.lua.rej
patching file scripts/src/main.lua
Hunk #1 FAILED at 380.
1 out of 1 hunk FAILED -- saving rejects to file scripts/src/main.lua.rej
*********** FAILED COMMAND ***********
patch -d "${PKG_BUILD}" -p1 < ${i} 1>&${VERBOSE_OUT}
[... 589 lines cut ...]
In file included from /usr/include/signal.h:328,
                 from ./signal.h:52,
                 from /home/gabriellaptop/LakkaV4/build.DreamOS-Generic.x86_64/build/m4-1.4.18/lib/c-stack.c:49:
/home/gabriellaptop/LakkaV4/build.DreamOS-Generic.x86_64/build/m4-1.4.18/lib/c-stack.c:55:26: error: missing binary operator before token "("
   55 | #elif HAVE_LIBSIGSEGV && SIGSTKSZ < 16384
      |                          ^~~~~~~~
/home/gabriellaptop/LakkaV4/build.DreamOS-Generic.x86_64/build/m4-1.4.18/lib/c-stack.c:107:1: warning: 'die' defined but not used [-Wunused-function]
  107 | die (int signo)
      | ^~~
make[4]: *** [Makefile:1910: c-stack.o] Error 1
make[4]: Leaving directory '/home/gabriellaptop/LakkaV4/build.DreamOS-Generic.x86_64/build/m4-1.4.18/.x86_64-linux-gnu/lib'
make[3]: *** [Makefile:1674: all] Error 2
make[3]: Leaving directory '/home/gabriellaptop/LakkaV4/build.DreamOS-Generic.x86_64/build/m4-1.4.18/.x86_64-linux-gnu/lib'
make[2]: *** [Makefile:1572: all-recursive] Error 1
make[2]: Leaving directory '/home/gabriellaptop/LakkaV4/build.DreamOS-Generic.x86_64/build/m4-1.4.18/.x86_64-linux-gnu'
make[1]: *** [Makefile:1528: all] Error 2
make[1]: Leaving directory '/home/gabriellaptop/LakkaV4/build.DreamOS-Generic.x86_64/build/m4-1.4.18/.x86_64-linux-gnu'
[1;31mFAILURE: scripts/build m4:host during make_host (default)[0m
*********** FAILED COMMAND ***********
make ${PKG_MAKE_OPTS_HOST}
**************************************
[1;31mFAILURE: scripts/build m4:host has failed![0m
`

I'm trying to build it with Ubuntu 22.04, running KDE. It's a laptop, 4GB RAM ddr3 1066MHz, 240GB SSD, Core 2 Duo T9900. I really would like to help with Lakka's development, but i can't help if i can't even compile it out of the box.

metchebe, 2023-09-09:
For building Lakka-4.x use Ubuntu Focal (20.04).

Also I suggest doing a test build with only one core with `CUSTOM_LIBRETRO_CORES=2048` to check that the base system is OK first.

GabrielAnacletoBaumgartner, 2023-09-09:
> For building Lakka-4.x use Ubuntu Focal (20.04).
> 
> Also I suggest doing a test build with only one core with `CUSTOM_LIBRETRO_CORES=2048` to check that the base system is OK first.

1. Why can't i build Lakka 4.x on latest Ubuntu, and why this is not in the compiling guide?
2. Does the same rule apply to Lakka 5.x?

GabrielAnacletoBaumgartner, 2023-09-09:
@metchebe I tried to Ubuntu (20.04) and i got this when installing the required to compile packages:

[image]

metchebe, 2023-09-09:
Why are you trying to install those packages? The script should take care of that for you.

What instructions are you following? They might be wrong.

GabrielAnacletoBaumgartner, 2023-09-09:
@metchebe What script? Lakka's script??? It simply never installed dependencies for me! I had to search and found [this post](https://forums.libretro.com/t/fixes-compiling-lakka/14052)

And i found a NEW ISSUE!!! AAAAAAAAAAAAAAAAAAAHHHHHHH!!!!!!!!! (Rage) WHY COMPILING THIS SYSTEM IS SO HARD?!

I started to compile the Lakka v4 in a virtual machine. I downgraded libncurses-dev (i had forgotten this one).

[image]

As you can see, ZX Utils repository is not working. I tried to download it manually and IS INDEED OFF

metchebe, 2023-09-09:
Alright. Take a break and let me get to my computer (tomorrow perhaps) to give you some pointers.

I've built 4.x and 5.x very recently and they do work.

GabrielAnacletoBaumgartner, 2023-09-09:
> Alright. Take a break and let me get to my computer (tomorrow perhaps) to give you some pointers.

Error 404 solved: DNS problem... Why? Aliens of course :/

metchebe, 2023-09-10:
OK there's a lot to go through so I will try to respond to everything, but not necessarily in order.

1. My experience is building Lakka for Raspberry Pi 3 on Ubuntu. I have not build Lakka for PC x86_64 so I am not familiar with problems that might be specific to that target.

2. I assume you have a working installation of Ubuntu, 22.04 for Lakka 5.x and 20.04 for Lakka 4.x. The older version is required for 4.x because the version of the `m4` package that ships with 4.x does not build with the newer glibc version in Ubuntu 22.04. Note that instead of a VM you can use a docker container, which I recommend IF you know how to use docker.

3. The base command that I understand you should be using is `DISTRO=Lakka PROJECT=Generic DEVICE=x11 ARCH=x86_64 make image`. This will also check and install any dependencies you may need. I would ignore any indication to install other dependencies. At most, it might be useful to install `build-essential` beforehand but I don't think it's necessary. Also I don't understand why you had issues with apt finding packages, seems like you forgot to update the repository or something.

4. Your issue with 5.x seems to be an outdated patch for `mame`, I would simply exclude it for now. So after `ARCH=x86_64` add `CUSTOM_LIBRETRO_CORES='core1 core2 core3 ...'` where the cores are the ones you want to include (you don't need to edit files). As I said before, I suggest building first with only `CUSTOM_LIBRETRO_CORES=2048` to make a test build.

5. Optionally, before building you can run `./libretro_update.sh` to update retroarch and/or the cores before building. This way you can have up-to-date retroarch and cores on 4.x and 5.x. In your case I would only do this after you have made a couple of successful builds, since updated cores might need fixes.

GabrielAnacletoBaumgartner, 2023-09-12:
> OK there's a lot to go through so I will try to respond to everything, but not necessarily in order.
> 
>     1. My experience is building Lakka for Raspberry Pi 3 on Ubuntu. I have not built Lakka for PC x86_64 so I am not familiar with problems that might be specific to that target.
> 
>     2. I assume you have a working installation of Ubuntu, 22.04 for Lakka 5.x and 20.04 for Lakka 4.x. The older version is required for 4.x because the version of the `m4` package that ships with 4.x does not build with the newer glibc version in Ubuntu 22.04. Note that instead of a VM you can use a docker container, which I recommend IF you know how to use docker.
> 
>     3. The base command that I understand you should be using is `DISTRO=Lakka PROJECT=Generic DEVICE=x11 ARCH=x86_64 make image`. This will also check and install any dependencies you may need. I would ignore any indication to install other dependencies. At most, it might be useful to install `build-essential` beforehand but I don't think it's necessary. Also I don't understand why you had issues with apt finding packages, seems like you forgot to update the repository or something.
> 
>     4. Your issue with 5.x seems to be an outdated patch for `mame`, I would simply exclude it for now. So after `ARCH=x86_64` add `CUSTOM_LIBRETRO_CORES='core1 core2 core3 ...'` where the cores are the ones you want to include (you don't need to edit files). As I said before, I suggest building first with only `CUSTOM_LIBRETRO_CORES=2048` to make a test build.
> 
>     5. Optionally, before building you can run `./libretro_update.sh` to update retroarch and/or the cores before building. This way you can have up-to-date retroarch and cores on 4.x and 5.x. In your case I would only do this after you have made a couple of successful builds, since updated cores might need fixes.

Hello. It's me again.

I tried doing what you said with Lakka 4, `./libretro_update.sh` to update all the cores, with clean unmodded git clone.
Then i used the command `DISTRO=Lakka PROJECT=Generic ARCH=x86_64 *_DEVICE=x11_* CUSTOM_LIBRETRO_CORES='2048' make image`

[image]
As you can see above It simply tried to compiled with 139 cores even with the command. Not just 2048. After I even tried modding the options file (with "EXCLUDE_LIBRETRO_CORES+=" to reduce it to something around 74) but the result was the same.

But again, i have a faulty repository:

[image]

ERROR 404 means link is down... I will have to wait but i would like to understand why CUSTOM_LIBRETRO_CORES are not working.

metchebe, 2023-09-12:
@GabrielAnacletoBaumgartner your Lakka options file seems to be modified. I can tell because line 377 does not match, see https://github.com/libretro/Lakka-LibreELEC/blob/Lakka-v4.x/distributions/Lakka/options

GabrielAnacletoBaumgartner, 2023-09-13:
> @GabrielAnacletoBaumgartner your Lakka options file seems to be modified. I can tell because line 377 does not match, see https://github.com/libretro/Lakka-LibreELEC/blob/Lakka-v4.x/distributions/Lakka/options

Ok... Original options file now.
This command `DISTRO=Lakka PROJECT=Generic DEVICE=x11 ARCH=x86_64 CUSTOM_LIBRETRO_CORES=2048 make image`

[image]

metchebe, 2023-09-13:
@GabrielAnacletoBaumgartner I checked and `DEVICE=x11` is only available on Lakka-5.x. In fact I tried to use the command on 4.x and it failed immediately with `There are no devices for project: Generic`.

You got a totally different error so it seems that you are using still using modified or mixed Lakka-4.x code.

I will reiterate that I do not have experience building for Generic and x11, but it seems to me now that Lakka-5.x (which also does not use the driver that gave you issues) is probably the easiest way to do it. Please use a clean codebase, since using edited or mixed code makes it difficult to distinguish between issues in Lakka and issues caused by edited code.

GabrielAnacletoBaumgartner, 2023-09-13:
> @GabrielAnacletoBaumgartner I checked and `DEVICE=x11` is only available on Lakka-5.x. In fact I tried to use the command on 4.x and it failed immediately with `There are no devices for project: [...]

I'm sorry i didn't make it clear: the last screenshot i tried to compile Lakka 5. Indeed i got confuse, it's been more than i week trying to compile this "thing". It's been really stressful and i am using two computer to try speed up the compiling process. 

I'm trying a v4 Generic non X11 build only with one core now to see if this "thing" works properly, but as you have more experience than me... Could confirm what are the commands to build Lakka 4 and 5 with X11 (Nvidia Drivers)?

AND why `CUSTOM_LIBRETRO_CORES=2048 does not work properly (not sure if it was clean Lakka v4 or clean Lakka v5)?

metchebe, 2023-09-13:
> Could confirm what are the commands to build Lakka 4 and 5 with X11 (Nvidia Drivers)?

There is no command for Lakka 4, I believe you would have to edit the source code. For Lakka 5 it should be the one I indicated. As far as I can tell there is no official support or test build (yet) for Lakka-5.x on X11, so you might in fact be the first to try! Also I am not certain that this will install the nvidia drivers you require.

> AND why `CUSTOM_LIBRETRO_CORES=2048 does not work properly (not sure if it was clean Lakka v4 or clean Lakka v5)?

As I stated, this happened because the options file had a syntax error because of incorrect editing, so it was not clean.

GabrielAnacletoBaumgartner, 2023-09-13:
> There is no command for Lakka 4, I believe you would have to edit the source code. For Lakka 5 it should be the one I indicated. As far as I can tell there is no official support or test build (yet) [...]

1. What??? Really? So why there's an official release of [2.2 Nvidia Lakka?](https://www.lakka.tv/articles/2019/01/16/lakka-generic-nvidia-experimental-builds/)

I'm clearly [not the first one to try](https://github.com/libretro/Lakka/pull/391) (I know, not the same repository). I need nvidia drivers because i want to build an arcade with 1600x1200 screen and need CRT-Royale to run flawlessly (i bought GT1030 just for it).

2. If i need to compile an Nvidia v4 Lakka should i follow [this guide?](https://forums.libretro.com/t/how-to-compile-lakka-with-x11-and-binary-nvidia-driver-pull-request-391/5087)

3. Tried to compile Lakka v4, 100% clean. This `DISTRO=Lakka PROJECT=Generic ARCH=x86_64 CUSTOM_LIBRETRO_CORES=2048 make image` and it compiled flawlessly. But i cannot see any IMG/ISO file. What should i do?

Note: I've never compiled a system before, just a few repositories, with clear (almost "noob" certified) guides.

metchebe, 2023-09-13:
> What??? Really? So why there's an official release of [2.2 Nvidia Lakka?](https://www.lakka.tv/articles/2019/01/16/lakka-generic-nvidia-experimental-builds/)

It says right there, _experimental build_.

> I'm clearly not the first one to try

Yeah, I specifically meant using `DEVICE=x11`, which comes directly from LibreELEC-11. As they say, it's for _Legacy Generic PC (for older PCs and NVIDIA GPUs)_. Maybe you should try the LibreELEC build, just to see how it works with your hardware [(Link)](https://libreelec.tv/downloads/generic/).

> If i need to compile an Nvidia v4 Lakka should i follow [this guide?](https://forums.libretro.com/t/how-to-compile-lakka-with-x11-and-binary-nvidia-driver-pull-request-391/5087)

Seems way to old to work in 4.x. I wouldn't waste my time with this.

>  But i cannot see any IMG/ISO file. What should i do?

The output is in the `target` folder, you need the `.img.gz` file

metchebe, 2023-09-13:
@GabrielAnacletoBaumgartner I apologize, the syntax error in the options file is from an error in the repository itself.

You can see it here:

https://nightly.builds.lakka.tv/2023-09-12_5.x/0_logs/x11.x86_64-2023-09-12_5.x.html

This seems to be the commit that produced the error: https://github.com/libretro/Lakka-LibreELEC/commit/24811d557cd2212b6021109669059c826c2a84e4

This also shows that the x11 target need more fixing. It seems it would be better to wait.

Ntemis, 2023-09-14:
fixed

GabrielAnacletoBaumgartner, 2023-09-15:
> Legacy Generic PC (for older PCs and NVIDIA GPUs)

Tried, and working flawlessly.

Now just need to build Lakka v5 (DEVICE=x11)?

metchebe, 2023-09-15:
@GabrielAnacletoBaumgartner no, check the log I posted above, ffmpeg fails at the end.

Better to wait for a fix, it's not ready yet.

GabrielAnacletoBaumgartner, 2023-10-05:
@metchebe @Ntemis tried to compile it today (fresh clone, same options). Same result

metchebe, 2023-10-06:
Yeah the [latest log](http://nightly.builds.lakka.tv/2023-10-05_5.x/0_logs/x11.x86_64-2023-10-05_5.x.html) shows that ffmpeg is still failing.

Try disabling it as a dependency in `packages/lakka/retroarch_base/retroarch/package.mk` in the line `PKG_DEPENDS_TARGET="..."` and see what happens.

GabrielAnacletoBaumgartner, 2023-10-07:
I can confirm it compiled and generated an image file, but i also got this message:

> [281/281] [DONE] install image:target
> 
> Total Build Time: 00:57:07.697 (wall clock)
> Accum Build Time: 05:42:46.123 (6 slots)
> 
> Breakdown by status (all slots):
> 
>   Status   Usage         ( Pct )  Count  State
>   ACTIVE   01:22:13.915  (24.0%)  272    busy 
>   FAILED     :  :00.000  (00.0%)  0           
>   GETPKG     :01:51.402  (00.5%)  19     busy 
>   IDLE     04:18:00.467  (75.3%)  281         
>   LOCKED     :  :27.876  (00.1%)  583         
>   MUTEX      :  :00.000  (00.0%)  0      busy 
>   MUTEX/W    :  :00.000  (00.0%)  0      stall
>   STALLED    :  :00.588  (00.0%)  2      stall
>   UNLOCK     :  :11.875  (00.1%)  583         
>   -------------------------------------
>   TOTAL    05:42:46.123  ( 100%)  1740 
> 
> Peak concurrency: 6 out of 6 slots
> 
> 2 job slots were held in a "stall" state for 00:00:00.588
> 
> Slot usage (time in a "busy" state):     | Concurrency breakdown ("busy"):
>                                          |
> #Rank  Slot  Usage        ( Pct )        | # of Slots  Usage        ( Pct )
>  #01    06     :32:04.236 (09.4%)        |     01        :57:04.075 (16.6%)
>  #02    02     :20:17.026 (05.9%)        |     02        :21:00.420 (06.1%)
>  #03    01     :18:21.640 (05.4%)        |     03        :03:48.619 (01.1%)
>  #04    03     :05:57.578 (01.7%)        |     04        :01:03.198 (00.3%)
>  #05    05     :05:43.877 (01.7%)        |     05        :  :39.322 (00.2%)
>  #06    04     :01:40.961 (00.5%)        |     06        :  :29.683 (00.1%)
> -----------------------------------------+---------------------------------
>  TOTALS      01:24:05.317 (24.5%)                      01:24:05.317 (24.5%)
> 
> Successful build, creating image...
> depmod: ERROR: fstatat(6, nvidia.ko): No such file or directory
> depmod: WARNING: /home/vboxuser/Lakka-LibreELEC/build.Lakka-x11.x86_64-5.0-devel/image/system/usr/lib/kernel-overlays/base/lib/modules/6.1.42/nvidia/nvidia-uvm.ko needs unknown symbol nvUvmInterfaceDisableAccessCntr
> depmod: WARNING: /home/vboxuser/Lakka-LibreELEC/build.Lakka-x11.x86_64-5.0-devel/image/system/usr/lib/kernel-overlays/base/lib/modules/6.1.42/nvidia/nvidia-uvm.ko needs unknown symbol nvUvmInterfaceChannelDestroy
> depmod: WARNING: /home/vboxuser/Lakka-LibreELEC/build.Lakka-x11.x86_64-5.0-devel/image/system/usr/lib/kernel-overlays/base/lib/modules/6.1.42/nvidia/nvidia-uvm.ko needs unknown symbol nvUvmInterfaceQueryCaps
> depmod: WARNING: /home/vboxuser/Lakka-LibreELEC/build.Lakka-x11.x86_64-5.0-devel/image/system/usr/lib/kernel-overlays/base/lib/modules/6.1.42/nvidia/nvidia-uvm.ko needs unknown symbol nvUvmInterfaceUnsetPageDirectory
> depmod: WARNING: /home/vboxuser/Lakka-LibreELEC/build.Lakka-x11.x86_64-5.0-devel/image/system/usr/lib/kernel-overlays/base/lib/modules/6.1.42/nvidia/nvidia-uvm.ko needs unknown symbol nvUvmInterfaceInitAccessCntrInfo
> depmod: WARNING: /home/vboxuser/Lakka-LibreELEC/build.Lakka-x11.x86_64-5.0-devel/image/system/usr/lib/kernel-overlays/base/lib/modules/6.1.42/nvidia/nvidia-uvm.ko needs unknown symbol nvUvmInterfaceReleaseChannel
> depmod: WARNING: /home/vboxuser/Lakka-LibreELEC/build.Lakka-x11.x86_64-5.0-devel/image/system/usr/lib/kernel-overlays/base/lib/modules/6.1.42/nvidia/nvidia-uvm.ko needs unknown symbol nvUvmInterfaceMemoryAllocSys
> depmod: WARNING: /home/vboxuser/Lakka-LibreELEC/build.Lakka-x11.x86_64-5.0-devel/image/system/usr/lib/kernel-overlays/base/lib/modules/6.1.42/nvidia/nvidia-uvm.ko needs unknown symbol nvUvmInterfaceMemoryCpuMap
> depmod: WARNING: /home/vboxuser/Lakka-LibreELEC/build.Lakka-x11.x86_64-5.0-devel/image/system/usr/lib/kernel-overlays/base/lib/modules/6.1.42/nvidia/nvidia-uvm.ko needs unknown symbol nvUvmInterfacePmaFreePages
> depmod: WARNING: /home/vboxuser/Lakka-LibreELEC/build.Lakka-x11.x86_64-5.0-devel/image/system/usr/lib/kernel-overlays/base/lib/modules/6.1.42/nvidia/nvidia-uvm.ko needs unknown symbol nvUvmInterfaceSetPageDirectory
> depmod: WARNING: /home/vboxuser/Lakka-LibreELEC/build.Lakka-x11.x86_64-5.0-devel/image/system/usr/lib/kernel-overlays/base/lib/modules/6.1.42/nvidia/nvidia-uvm.ko needs unknown symbol nvUvmInterfaceMemoryCpuUnMap
> depmod: WARNING: /home/vboxuser/Lakka-LibreELEC/build.Lakka-x11.x86_64-5.0-devel/image/system/usr/lib/kernel-overlays/base/lib/modules/6.1.42/nvidia/nvidia-uvm.ko needs unknown symbol nvUvmInterfaceOwnPageFaultIntr
> depmod: WARNING: /home/vboxuser/Lakka-LibreELEC/build.Lakka-x11.x86_64-5.0-devel/image/system/usr/lib/kernel-overlays/base/lib/modules/6.1.42/nvidia/nvidia-uvm.ko needs unknown symbol nvUvmInterfaceDupAddressSpace
> depmod: WARNING: /home/vboxuser/Lakka-LibreELEC/build.Lakka-x11.x86_64-5.0-devel/image/system/usr/lib/kernel-overlays/base/lib/modules/6.1.42/nvidia/nvidia-uvm.ko needs unknown symbol nvUvmInterfaceGetExternalAllocPtes
> depmod: WARNING: /home/vboxuser/Lakka-LibreELEC/build.Lakka-x11.x86_64-5.0-devel/image/system/usr/lib/kernel-overlays/base/lib/modules/6.1.42/nvidia/nvidia-uvm.ko needs unknown symbol nvUvmInterfaceRegisterGpu
> depmod: WARNING: /home/vboxuser/Lakka-LibreELEC/build.Lakka-x11.x86_64-5.0-devel/image/system/usr/lib/kernel-overlays/base/lib/modules/6.1.42/nvidia/nvidia-uvm.ko needs unknown symbol nvUvmInterfaceP2pObjectDestroy
[... cut ...]
> image: creating filesystem on part2...
> image: populating filesystem on part2...
> image: merging part2 into disk image...
> image: creating sparse file for part1...
> image: creating filesystem on part1...
> image: creating bootloader configuration...
> image: installing syslinux to part1...
> image: copying files to part1...
> image: checking filesystem on part1...
> image: merging part1 into disk image...
> image: compressing...
> image: cleanup...
> 

Anyways, i still need to test it. More updates later

GabrielAnacletoBaumgartner, 2023-10-07:
Update:

Disk created successfully, boot and typed live mode as normal, but [this is what happens](https://cdn.discordapp.com/attachments/950481165708562522/1160027097562107934/rn_image_picker_lib_temp_363639cc-19a8-470b-9748-1cd8586f758b.jpg) when Lakka starts.

It looks like the OS boots but RetroArch does not.

metchebe, 2023-10-07:
> Would I run into legal problems if I make a Debian fork with only RetroArch? Like a "Lakkabian"?

I doubt anyone that would know about these things is reading this thread. That said, if I were to publish this project online I would stay away from any naming resembling 'Lakka' or 'Debian'.

For _actually using_ your machine it would probably be easier to just use Windows. Trying to build an unsupported and experimental OS was always going to be difficult.

If you want to keep experimenting then you already know that official LibreELEC works; you could try to build and run retroarch in that, in fact the retroarch binary that you built might work already.

GavinDarkglider, 2023-10-22:
> Update:
> 
> Disk created successfully, boot and typed live mode as normal, but [this is what happens](https://cdn.discordapp.com/attachments/950481165708562522/1160027097562107934/rn_image_picker_lib_temp_363639cc-19a8-470b-9748-1cd8586f758b.jpg) when Lakka starts.
> 
> It looks like the OS boots but RetroArch does not.
> 
> P.S.: I am almost giving up, so... Would I run into legal problems if I make a Debian fork with only RetroArch? Like a "Lakkabian"? Of course it would be open source, free and the only advantage would be better nvidia-drivers support.

Look at the switch build. I had to do some hacky stuff to get X11 working on the switch, and the odin...... first you need to add proper configuration for X for your hardware. Without the logs from X, and retroarch, there isnt much I can say to help you fix that issue though.

Ntemis, 2023-12-22:
x11 images are up now so closing this

closed 2023-12-22 by Ntemis