libretro/Lakka-LibreELEC #1878: My PC is stuck on the flower logo when the Installer restart the PC

edward903, 2023-09-09:
I tried installing Lakka 4.3 to my small PC, which was almost the size of a NUC.
But after it finished resizing the file system and rebooted, it is stuck on flower logo.
I don't know if it finished installing, it only shows a line of information and a flower logo on the screen.
I thought it was installed, so I tried pressing a few keyboard keys, thinking it would give me some feedback. But it didn't, and I couldn't do anything but turn the Numlock lights off or on.
I can only force a shutdown by pressing and holding the power button on my computer. Then I tried to remove the USB storage to start my PC, but I did not see the corresponding message from Lakka, I only saw a GRUB interface, it is clear that it is not installed correctly.
Please tell me what I need to do to find out why it happened?
CPU:Intel(R) Atom(TM) Processor E3940
Memory:DDR3 4G
Storage:32GB mSATA SSD

edward903, 2023-09-09:
My English is poor because my native language is not English.
I had to use a translator to describe my problem, hopefully you know what I'm experiencing and where I need to start debugging.

edward903, 2023-09-09:
I noticed you wrote some stuck solutions in the docs, but I have some issues here that I need to work out.
(1) Although my computer has a wireless network device, I don't have a device that can provide a network link to it. That's where my rental house, so I didn't find a working ethernet connection. Maybe I will take the PC to a place with a network environment and use SSH to test it in a few days, but when I connect my USB storage without a network link, it gets stuck on the Lakka flower logo. If I don’t connect it, it shows GRUB interface.
(2) Because that box is small, I can't find a serial port to initiate serial access.
(3) It got stuck, my issue description mentioned that I used the keyboard enter key but it didn't do anything.

gouchi, 2023-09-19:
May you try to boot using usb stick in live mode ?

`live textmode`

Then launch RA to get some output (maybe screenshot)
`retroarch -v
`

Then graphic card log using
`lspci -nnk | grep -A 3 VGA`

Thank you.

RoxDevvv, 2023-11-30:
i have an old small laptop i have the same behavior 
[image]
[image]

closed 2023-11-30 by Ntemis