libretro/Lakka-LibreELEC #1894: DualSense causes input from any controller to not to work

ilya-fedin, 2023-10-22:
I've tried to connect a DualSense controller via both BT and USB, with both the controller itself doesn't work and others stop working, too. I've an other controller and a keyboard, both of which can't do menu navigation after a DualSense controller. Although when tried to press random keys on the other controller/keyboard, some of them do work. Maybe it's just the arrows that don't work... If I unplug the DualSense controller, other input devices start to work again.
[image]

RobLoach, 2023-10-30:
Mind exporting the config and trying to submit it to the joypad configs? https://github.com/libretro/retroarch-joypad-autoconfig

dmrlawson, 2023-10-30:
We were missing some kernel options that made them behave weirdly. It was fixed in https://github.com/libretro/Lakka-LibreELEC/pull/1837 for Lakka v5 so you could try a v5 nightly version

ilya-fedin, 2023-10-30:
How can I insttall a nightly version with PINE?

rtrzebinski, 2023-11-12:
Same here, is there any older, stable Lakka version without this issue?

Basically when I plug Dual Sense with cable (or BT, doesn't matter) it replaces the keyboard on Input 1, and neither keyboard nor controller works. It should rather register on an empty, unused slot - input 2 in this case.

I see this -> https://nightly.builds.lakka.tv/devbuild_by_devices/RPi3.aarch64/

But I'd rather avoid using unstable version if possible.

gouchi, 2023-12-19:
@ilya-fedin  @rtrzebinski Can you try to make a test with the [Lakka 5.x build ](https://nightly.builds.lakka.tv/devbuild_latest/)as it will be the future stable release.

Thank you.

ilya-fedin, 2023-12-19:
@gouchi can you explain how can I install it with PINN? Thank you as well.

gouchi, 2023-12-19:
@ilya-fedin depending on the device you own but for [Pine 64](https://nightly.builds.lakka.tv/devbuild_latest/A64.aarch64/) or [Pine  H64](https://nightly.builds.lakka.tv/devbuild_latest/H6.aarch64/).

Thank you.

ilya-fedin, 2023-12-19:
@gouchi I own rpi2 and I'm not sure how would I put the image into PINN given that it accepts prepared ones with a manifest. In ideal outcome I would like to be able to update right from Lakka UI or have an option in PINN to download & install (in my case replace already installed) the nightly build. I use multiboot via PINN and so I can't just write the raw image.

n2sml, 2023-12-19:
Hello there, I'm having the same issue, or very similar one.
I'm using Lakka on Nintendo Switch, and by default (after Nyx dump) my JoyCons works well, but when I try to connect one of my DualShock 4's, my JoyCons simply stops working, and my DualShock don't work as well.
I have tried to do key bindings, but the Lakka simply cannot read the inputs i'm pressing on DS4 and on the Joycons.
I'm already tried using 4.x Lakka and nightly 5.x, and in both cases, this problem happens.

gouchi, 2023-12-20:
@ilya-fedin I don't know [PINN](https://github.com/procount/pinn) but can't you try to use our [RPi2 build](https://nightly.builds.lakka.tv/devbuild_latest/RPi2.arm/) to SD card using [Etcher](https://etcher.io/) ?

@n2sml This issue is related to DualSense not DS4 with Switch so you should open a new issue with some [logs](https://www.lakka.tv/doc/Troubleshooting-Lakka/). Thank you.

ilya-fedin, 2023-12-20:
@gouchi no, I don't have a spare SD card

Ntemis, 2024-07-10:
Please test this on 5.0 stable

ilya-fedin, 2024-07-10:
I have switched away from Lakka, sorry :(

Ntemis, 2024-07-10:
Not a problem, closing down

closed 2024-07-10 by Ntemis