libretro/Lakka-LibreELEC #1902: Lakka 5x : No boot on RPi4-GPICase2

ShigeakiAsai, 2023-12-06:
No boot on RPi4-GPICase2.

Lakka-RPi4-GPICase2.aarch64-devbuild-v5.x-20231202-e8e0f37.img.gz

The mis-match bootloader files are used.
- fixup.dat (fixup4x.dat <-> fixup_x.dat)
- start.elf (start4x.elf <-> start_x.elf)

I found the reason in line 20 of Lakka-LibreELEC/packages/tools/bcm2835-bootloader/package.mk
> 
    case "${DEVICE}" in
      RPi4)
        cp -PRv fixup4x.dat ${INSTALL}/usr/share/bootloader/fixup.dat
        cp -PRv start4x.elf ${INSTALL}/usr/share/bootloader/start.elf
        ;;

It needs that RPi4) changes to RPi4*).

This may occure on the other devices such as RPi4-PiBoyDmg and RPi4-RetroDreamer.
But I can't confirm these devices because I don't have them.

Thanks,
ASAI, Shigeaki

dmrlawson, 2023-12-06:
I think you're probably correct, I can test if you make the change

Ntemis, 2023-12-06:
*|RPi4*)
this should be the fix, test it and let me know

ShigeakiAsai, 2023-12-06:
Still happened with your commit (7d751eaf82d540e6588113bb4dd475c2a9616898).

ShigeakiAsai, 2023-12-06:
https://www.raspberrypi.com/documentation/computers/configuration.html#boot-folder-contents
> start4.elf, start4x.elf, start4db.elf and start4cd.elf are equivalent firmware files specific to the Raspberry Pi 4-series (Model 4B, Pi 400, Compute Module 4 and Compute Module 4S).

Your commit uses "start_x.elf",but CM4 needs "start4x.elf".

Ntemis, 2023-12-06:
I thought you meant the other way around

ShigeakiAsai, 2023-12-06:
I am apologize with my wrong and poor English.

closed 2023-12-06 by Ntemis

ShigeakiAsai, 2023-12-06:
Test is "OK", it is booted.

Thank you for your quick fix.

ShigeakiAsai, 2023-12-07:
Also, I confirmed it was good (booted) on a following nightly build.
- Lakka-RPi4-GPICase2.aarch64-devbuild-v5.x-20231206-e66c240.img

Thanks,
ASAI, Shigeaki