libretro/Lakka-LibreELEC #1909: add distroconfig.txt for gpi case 2w to GpiCase.arm and Pi02GPi.arm builds

jdalmanza, 2023-12-20:
To add functionality for the newest retroflag GPI 2W case the values in distroconfig.txt will need to be changed as the screen has a different resolution and different driver.

I propose add the attached file to the /config for the build so that it is included in the root as distroconfig.gpi2w and the user can rename if they intend to use the image in that case

I have tested this on both images (GpiCase.arm and Pi02GPi.arm) on both the 4.3 images and the 5.x nightles currently in development

[distroconfig.gpi2w.txt](https://github.com/libretro/Lakka-LibreELEC/files/13721570/distroconfig.gpi2w.txt)

jdalmanza, 2023-12-20:
an alternative to adding this text file to the two existing device builds would be to create two new deices and name them accordingly (RPiZeroGPiCase2w and RPiZero2GPiCase2w ... see Issue#1910 for recommendation on updated naming for the previous images) but unless the goal is to look like we support more devices, I think we can serve the users just as well by including an additional distroconfig file to these device builds to be more efficient

ToKe79, 2024-04-11:
this should be solved with 37f1a77f9e5aa731a23ef26d0b4f972327831b6e

closed 2024-04-11 by ToKe79