libretro/Lakka-LibreELEC #1911: wifi not working on some models of RPi Zero 2 -- Pi02GPi.arm build

jdalmanza, 2023-12-22:
dmesg | grep brcm  -- working pizereo2 5.x build (provided by user @CeeAytch on discord

[    9.579539] brcmfmac: F1 signature read @0x18000000=0x1541a9a6
[    9.585185] brcmfmac: brcmf_fw_alloc_request: using **brcm/brcmfmac43430-sdio** for chip **BCM43430/1**
[    9.585340] brcmfmac mmc1:0001:1: Direct firmware load for brcm/brcmfmac43430-sdio.raspberrypi,model-zero-2-w.bin failed with error -2
[    9.597587] usbcore: registered new interface driver brcmfmac
[    9.686058] Bluetooth: hci0: BCM43430A1 'brcm/BCM43430A1.raspberrypi,model-zero-2-w.hcd' Patch
[    9.967580] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM43430/1 wl0: Jul 19 2021 03:24:18 version 7.45.98 (TOB) (56df937 CY) FWID 01-8e14b897
[   51.258879] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save disabled

non-working pizero2 5.x

[   10.601841] brcmfmac: F1 signature read @0x18000000=0x1542a9a6
[   10.608477] brcmfmac: brcmf_fw_alloc_request: using **brcm/brcmfmac43430b0-sdio** for chip **BCM43430/2**
[   10.608708] usbcore: registered new interface driver brcmfmac
[   10.609478] brcmfmac mmc1:0001:1: Direct firmware load for brcm/brcmfmac43430b0-sdio.raspberrypi,model-zero-2-w.bin failed with error -2
[   10.609579] brcmfmac mmc1:0001:1: Direct firmware load for brcm/brcmfmac43430b0-sdio.bin failed with error -2
[   10.672144] Bluetooth: hci0: BCM43430B0 'brcm/BCM43430B0.raspberrypi,model-zero-2-w.hcd' Patch
[   11.638756] brcmfmac: brcmf_sdio_htclk: HT Avail timeout (1000000): clkctl 0x50

example from same device running the 4.3 build

[   10.403782] brcmfmac: F1 signature read @0x18000000=0x1542a9a6
[   10.414036] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43436-sdio for chip BCM43430/2
[   10.414301] usbcore: registered new interface driver brcmfmac
[   10.616190] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43436-sdio for chip BCM43430/2
[   10.616335] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43436-sdio for chip BCM43430/2it can use 
[   10.696026] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM43430/2 wl0: Oct  9 2020 14:44:32 version 9.88.4.65 (test) (f149b32@shgit)  (r679549) FWID 01-f40f3270
[   10.930310] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save disabled

4.3 : "brcm/brcmfmac43436-sdio" is used  for chip BCM43430/2.
5.x : "brcm/brcmfmac43436-sdio" is used  for chip BCM43430/1
5.x : "brcm/brcmfmac43430b0-sdio"  is used  for chip BCM43430/2 for some models of the rpizero2. 

so it looks like the 6.1 kernel is seeing different versions of the chip and using different firmware files accordingly

brcm/brcmfmac43430b0-sdio is not present in the current 5.x tree and therefore we get the following error:

brcmfmac: brcmf_sdio_htclk: HT Avail timeout (1000000): clkctl 0x50

ShigeakiAsai, 2023-12-22:
### Zero2w wifi chip is used one of the two is used.(BCM43430/1 or BCM43430/2)

<BR>

I think this problem is concern with following 2 reason.
- Raspberry Pi Linux update.
- Lakka "brcmfmac_sdio-firmware-rpi" package,

<BR>

**- About Raspberry Pi Linux update.**
Raspberry Pi Linux updated the Zero2w wifi driver.
"brcmfmac: use separate firmware for 43430 revision 2" Commits on Aug 22, 2021
(https://github.com/raspberrypi/linux/commit/f8d6523891cf07468f0bfe21aba8b562a6f75780)
It seemed to change Zero2w BCM43430/2 firmrare file.
(from brcmfmac43436-sdio to  brcmfmac43430b0-sdio)

**- About Lakka "brcmfmac_sdio-firmware-rpi" package,**
"brcmfmac_sdio-firmware-rpi" package includes brcmfmac43430-sdio and brcmfmac43436-sdio firmrares.
But, brcmfmac43430b0-sdio is not included.
**It needs to update.**

yasai@X670E-AORUS-MASTER:~/brcmfmac_sdio-firmware-rpi-a25c7c3d04db0363409dfd17f265bab66f0eae5a/firmware/brcm$ ls -l brcmfmac4343*
lrwxrwxrwx 1 yasai yasai     31 Nov 20  2022 brcmfmac43430-sdio.bin -> ../cypress/cyfmac43430-sdio.bin
lrwxrwxrwx 1 yasai yasai     36 Nov 20  2022 brcmfmac43430-sdio.clm_blob -> ../cypress/cyfmac43430-sdio.clm_blob
lrwxrwxrwx 1 yasai yasai     31 Nov 20  2022 brcmfmac43430-sdio.raspberrypi,3-model-b.bin -> ../cypress/cyfmac43430-sdio.bin
lrwxrwxrwx 1 yasai yasai     36 Nov 20  2022 brcmfmac43430-sdio.raspberrypi,3-model-b.clm_blob -> ../cypress/cyfmac43430-sdio.clm_blob
lrwxrwxrwx 1 yasai yasai     22 Nov 20  2022 brcmfmac43430-sdio.raspberrypi,3-model-b.txt -> brcmfmac43430-sdio.txt
lrwxrwxrwx 1 yasai yasai     31 Nov 20  2022 brcmfmac43430-sdio.raspberrypi,model-zero-w.bin -> ../cypress/cyfmac43430-sdio.bin
lrwxrwxrwx 1 yasai yasai     36 Nov 20  2022 brcmfmac43430-sdio.raspberrypi,model-zero-w.clm_blob -> ../cypress/cyfmac43430-sdio.clm_blob
lrwxrwxrwx 1 yasai yasai     22 Nov 20  2022 brcmfmac43430-sdio.raspberrypi,model-zero-w.txt -> brcmfmac43430-sdio.txt
-rw-r--r-- 1 yasai yasai   1121 Nov 20  2022 brcmfmac43430-sdio.txt
-rw-r--r-- 1 yasai yasai 416101 Nov 20  2022 brcmfmac43436-sdio.bin
-rw-r--r-- 1 yasai yasai  11209 Nov 20  2022 brcmfmac43436-sdio.clm_blob
lrwxrwxrwx 1 yasai yasai     22 Nov 20  2022 brcmfmac43436-sdio.raspberrypi,model-zero-2-w.bin -> brcmfmac43436-sdio.bin
lrwxrwxrwx 1 yasai yasai     27 Nov 20  2022 brcmfmac43436-sdio.raspberrypi,model-zero-2-w.clm_blob -> brcmfmac43436-sdio.clm_blob
lrwxrwxrwx 1 yasai yasai     22 Nov 20  2022 brcmfmac43436-sdio.raspberrypi,model-zero-2-w.txt -> brcmfmac43436-sdio.txt
-rw-r--r-- 1 yasai yasai   1613 Nov 20  2022 brcmfmac43436-sdio.txt
-rw-r--r-- 1 yasai yasai 436733 Nov 20  2022 brcmfmac43436s-sdio.bin
lrwxrwxrwx 1 yasai yasai     23 Nov 20  2022 brcmfmac43436s-sdio.raspberrypi,model-zero-2-w.bin -> brcmfmac43436s-sdio.bin
lrwxrwxrwx 1 yasai yasai     23 Nov 20  2022 brcmfmac43436s-sdio.raspberrypi,model-zero-2-w.txt -> brcmfmac43436s-sdio.txt
-rw-r--r-- 1 yasai yasai   1092 Nov 20  2022 brcmfmac43436s-sdio.txt

Thanks
ASAI, Shigeaki

ShigeakiAsai, 2023-12-22:
Following may work around before "brcmfmac_sdio-firmware-rpi" package update.

packages/linux-firmware/brcmfmac_sdio-firmware-rpi/package.mk

`makeinstall_target() {`
**`  if [ "${DEVICE}" = "Pi02GPi" -o "${DEVICE}" = "GPICase2W-Zero2-64" ]; then`**
**`    ln -sv brcmfmac43436-sdio.bin      firmware/brcm/brcmfmac43430b0-sdio.raspberrypi,model-zero-2-w.bin`**
**`    ln -sv brcmfmac43436-sdio.clm_blob firmware/brcm/brcmfmac43430b0-sdio.raspberrypi,model-zero-2-w.clm_blob`**
**`    ln -sv brcmfmac43436-sdio.txt      firmware/brcm/brcmfmac43430b0-sdio.raspberrypi,model-zero-2-w.txt`**
**`  fi`**
`  DESTDIR=${INSTALL}/$(get_kernel_overlay_dir) ./install`
`}`

Create symbolic link.

Thanks
ASAI, Shigeaki

ShigeakiAsai, 2023-12-29:
@Ntemis 

Please update "brcmfmac_sdio-firmware-rpi" package.
It was added after "Commits on Nov 19, 2023"(bfad24fe1260fa5217327cbd61477aa396ccf9e2) or later.
"Commits on Nov 19, 2023" is bfad24fe1260fa5217327cbd61477aa396ccf9e2 (Added version)
"Commits on Nov 22, 2023" is 26ff205b45dc109b498a70aaf182804ad9dbfea5 (Latest version)
https://github.com/LibreELEC/brcmfmac_sdio-firmware-rpi/commits/master/

RPi Zero 2W wifi (BCM43430/2) will work by this update.

Thanks
ASAI, Shigeaki

jdalmanza, 2024-01-15:
confirmed wifi successfully loaded on the 14-JAN build 

[log-2024-01-15-20.32.48.zip](https://github.com/libretro/Lakka-LibreELEC/files/13942865/log-2024-01-15-20.32.48.zip)

# ... output of dmesg
# Lakka release: Pi02GPi.arm-devel-20240114170117-f787194
# 
...
[   11.023620] brcmfmac: F1 signature read @0x18000000=0x1542a9a6
[   11.030841] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430b0-sdio for chip BCM43430/2
[   11.031323] usbcore: registered new interface driver brcmfmac

closed 2024-01-15 by jdalmanza