libretro/Lakka-LibreELEC #1912: replace "Microsoft X-Box 360 pad"  controller autoconifg file in RPi4-GPiCase2 build with the one from GPiCase build

jdalmanza, 2023-12-23:
The button mappings on the Retroflag GPi case are wrong in the RPi4-GPiCase2 build because the autoconfig file is the default for the xbox 360 controller. 

We should replace the default autoconfig file with the one that was customized for the gpi case present in the GPiCase.arm and Pi02GPi.arm builds:

Microsoft X-Box 360 pad.cfg from GPiCase build:
=========
```
input_driver = "udev"
input_device = "Microsoft X-Box 360 pad"
input_device_display_name = "Retroflag Gpi Case"
input_vendor_id = "1118"
input_product_id = "654"

input_up_btn = "h0up"
input_down_btn = "h0down"
input_left_btn = "h0left"
input_right_btn = "h0right"

input_b_btn = "1"
input_y_btn = "3"
input_a_btn = "0"
input_x_btn = "2"
input_l_btn = "4"
input_r_btn = "5"

input_select_btn = "6"
input_start_btn = "7"
input_menu_toggle_btn = "8"

input_b_btn_label = "B"
input_y_btn_label = "Y"
input_select_btn_label = "Select"
input_start_btn_label = "Start"
input_up_btn_label = "D-Pad Up"
input_down_btn_label = "D-Pad Down"
input_left_btn_label = "D-Pad Left"
input_right_btn_label = "D-Pad Right"
input_a_btn_label = "A"
input_x_btn_label = "X"
input_l_btn_label = "L"
input_r_btn_label = "R"
```

Microsoft X-Box 360 pad.cfg from RPi4GPiCase2 build:
========
```
input_driver = "udev"
input_device = "Microsoft X-Box 360 pad"
input_device_display_name = "Xbox 360 Controller"
input_vendor_id = "1118"
input_product_id = "654"

input_b_btn = "0"  <--incorrect
input_y_btn = "2"  <-- incorrect
input_select_btn = "6"
input_start_btn = "7"
input_up_btn = "h0up"
input_down_btn = "h0down"
input_left_btn = "h0left"
input_right_btn = "h0right"
input_a_btn = "1" <-- incorrect 
input_x_btn = "3"  <-- incorrect
input_l_btn = "4"
input_r_btn = "5"
input_l2_axis = "+2"
input_r2_axis = "+5"
input_l3_btn = "9"
input_r3_btn = "10"
input_l_x_plus_axis = "+0"
input_l_x_minus_axis = "-0"
input_l_y_plus_axis = "+1"
input_l_y_minus_axis = "-1"
input_r_x_plus_axis = "+3"
input_r_x_minus_axis = "-3"
input_r_y_plus_axis = "+4"
input_r_y_minus_axis = "-4"
input_menu_toggle_btn = "8"

input_b_btn_label = "A"
input_y_btn_label = "X"
input_select_btn_label = "Back"
input_start_btn_label = "Start"
input_up_btn_label = "D-Pad Up"
input_down_btn_label = "D-Pad Down"
input_left_btn_label = "D-Pad Left"
input_right_btn_label = "D-Pad Right"
input_a_btn_label = "B"
input_x_btn_label = "Y"
input_l_btn_label = "Left Bumper"
input_r_btn_label = "Right Bumper"
input_l2_axis_label = "Left Trigger"
input_r2_axis_label = "Right Trigger"
input_l3_btn_label = "Left Thumb"
input_r3_btn_label = "Right Thumb"
input_l_x_plus_axis_label = "Left Analog Right"
input_l_x_minus_axis_label = "Left Analog Left"
input_l_y_plus_axis_label = "Left Analog Down"
input_l_y_minus_axis_label = "Left Analog Up"
input_r_x_plus_axis_label = "Right Analog Right"
input_r_x_minus_axis_label = "Right Analog Left"
input_r_y_plus_axis_label = "Right Analog Down"
input_r_y_minus_axis_label = "Right Analog Up"
input_menu_toggle_btn_label = "Guide"

```
Also in the customized version included in the GPiCase build removed all of the buttons that are not present on the case fwiw

ShigeakiAsai, 2023-12-25:
Please confirm it on 25 Dec nightly build I hope.

jdalmanza, 2023-12-26:
Buttons are mapped appropriately as of 25-DEC build.

closed 2024-02-04 by Ntemis