libretro/Lakka-LibreELEC #1921: RPi3 Flycast Inverted on Vertical

almusx, 2024-01-15:
- Which version of Lakka are you using? 
   RPi3.aarch64-devel-20231214220209-1661c01 

- What system hardware are you using? 
   Raspberry Pi 3B

- What did you do? 
   Launching a dreamcast using flycast

- What did you expect to happen? 
   game runs normaly

- What happened instead? 
   game runs with the screen Upside Down (the screen is inverted in the vertical not rotate 180º). I tried rotating the screen in retroarch options, i can make the screen upside but its became inverted in horizontal (text right to left).

[retroarch__2024_01_14__22_49_39.log](https://github.com/libretro/Lakka-LibreELEC/files/13934003/retroarch__2024_01_14__22_49_39.log)

Ntemis, 2024-03-17:
can you retest? i don't have this issue

gouchi, 2024-07-15:
@almusx Can you make a test with [latest nightly](https://nightly.builds.lakka.tv/latest/RPi3.aarch64/) if you reproduce your issue ?

Thank you.

gouchi, 2024-08-07:
@almusx I could make a test on RPi3 using [latest nightly](https://nightly.builds.lakka.tv/latest/RPi3.aarch64/) and I could not also reproduce the issue.

Please confirm it is working also for you, so that we can close this issue.

If you still have the issue, please provide some [log](https://lakka.tv/doc/Troubleshooting-Lakka/).

Thank you.

almusx, 2024-11-11:
Sorry for late reply but i cant test it anymore, my raspberry was not working. Thank you.

closed 2024-11-11 by almusx