libretro/Lakka-LibreELEC #1922: switch - access denied when trying to mount NAS share for roms etc.

jdalmanza, 2024-01-15:
when following the steps to mount a NAS share for roms as described in the lakka documentation:
https://www.lakka.tv/doc/Serving-ROMs-from-a-NAS/

When you reach the step to enable the storage-{path}.mount service you get an Access denied error:

Lakka:~ # systemctl enable storage-nas.mount
Failed to enable unit: Access denied

This issue is present on 4.3 and still an issue in the 5.x nightly builds (tested on 15-JAN nightly build)

Ntemis, 2024-01-15:
Thank you for the report!

jdalmanza, 2024-01-24:
@Ntemis , i tried this on the 17-JAN build and got he same results
logset attached, please let me know if you need anything else
[log-2024-01-24-00.51.52.zip](https://github.com/libretro/Lakka-LibreELEC/files/14031317/log-2024-01-24-00.51.52.zip)

synthic, 2024-02-05:
Searching for this issue yields results relating to selinux, but I don't think that applies here. Still, you can try to run `systemctl daemon-reexec` before the enable command and see if that helps.

jdalmanza, 2024-02-05:
akka (official): devel-20240123161432-37de45a (Switch.aarch64)

lakka-switch:~ # ls /storage/.config/system.d/
README                   multi-user.target.wants  openvpn.service.sample
cifs.mount.sample        nfs.mount.sample         storage-nas.mount
lakka-switch:~ # systemctl daemon-reexec
lakka-switch:~ # systemctl enable storage-nas.mount
Failed to enable unit: Access denied

lakka-switch:~ # cat /storage/.config/system.d/storage-nas.mount 
[Unit]
Description=cifs mount script
Requires=network-online.service
After=network-online.service
Before=retroarch.service

[Mount]
What=//xxx.xxx.xxx.xxx/retroarch
Where=/storage/nas
Options=username=xxx,password=xxx,rw
Type=cifs

[Install]
WantedBy=multi-user.target
lakka-switch:~ # 

This exact file works on all my other lakka devices 4.3 and 5.x the switch build is the only one that fails

Ntemis, 2024-02-06:
ow! missed the switch part. Reopening and assigning to the proper person

GavinDarkglider, 2024-03-17:
> lakka (official): devel-20240123161432-37de45a (Switch.aarch64)
> 
> lakka-switch:~ # ls /storage/.config/system.d/ README multi-user.target.wants openvpn.service.sample cifs.mount.sample nfs.mount.sample storage-nas.mount lakka-switch:~ # systemctl daemon-reexec lakka-switch:~ # systemctl enable storage-nas.mount Failed to enable unit: Access denied
> 
> lakka-switch:~ # cat /storage/.config/system.d/storage-nas.mount [Unit] Description=cifs mount script Requires=network-online.service After=network-online.service Before=retroarch.service
> 
> [Mount] What=//xxx.xxx.xxx.xxx/retroarch Where=/storage/nas Options=username=xxx,password=xxx,rw Type=cifs
> 
> [Install] WantedBy=multi-user.target lakka-switch:~ #
> 
> @synthic This exact file works on all my other lakka devices 4.3 and 5.x the switch build is the only one that fails

This works for me, just need to set protocol version in options line..... testing against samba version packaged in libreelec/lakka, the needed protocol version is 2.1, though that is depending on your server.

`Options=username=xxx,password=xxx,rw,vers=2.1

GavinDarkglider, 2024-03-17:
Ok, figured out what is going on here....... Not really sure how to fix it properly though..... I might be able to hack something together, but this comes down to how systemd enables services(Creates Symlinks in a bunch of folders, to the service/mount file). Unfortunatly, fat32/exfat do not support symbolic links at all, hince the access denied when trying to enable it.

This issue has probably existed since the first version of lakka on the switch, and no one noticed it, because no one else was trying to add systemd unit files to storage and enable them. lol.

That being said, I am going to test some stuff, and see if I can work around the issue for you, probably wont be the cleanest fix ever, but the other option is moving storage to a proper linux partition on the SDcard which kind of ruins the magic of what lakka does for the average switch user.

jdalmanza, 2024-03-17:
actually that makes a lot of sense about the symlinks.
So, let's say that the switch version cannot mount a nas share like a service as the other builds do, how hard would it be to offer an easy way to easily mount a cifs/samba share via the gui. i figure folks that know what they are doing can probably mount manually via the command line or even maybe edit the fstab if that is allowed. 
I'm willing to close this out with that explanation as-is and maybe turn it into a feature request for a gui enabled nas mount function?

GavinDarkglider, 2024-03-17:
> actually that makes a lot of sense about the symlinks. So, let's say that the switch version cannot mount a nas share like a service as the other builds do, how hard would it be to offer an easy way [...]

The reality is, this should be fixed, to keep uniformity with all builds. While I agree with you this should be added to the gui, which is also possible, and then have it start the services for systemd, if the user enables them at boot time......

that is kind of a dirty hack though. Currently, you should be able to use the mount file you created....... `systemctl start storage-nas.mount` you just cant set it to start on boot.

GavinDarkglider, 2024-03-17:
The not so pretty way to work around this would be to call a script in this file: https://github.com/libretro/Lakka-LibreELEC/blob/Lakka-v5.x/packages/lakka/retroarch_base/retroarch/scripts/retroarch-config

then you can put the systemctl command in a script on your sdcard, and still start it before retroarch starts. That being said, it would still need a new image, so let me play with some more messed up, but also more uniform options, that might fix the issue, as that would be ideal.

whmzsu, 2025-01-09:
Is there any options to set with mount  to support Unicode charsets like CJK with the default fat32 partition  or other new partitions ??

shea-sollars, 2025-01-13:
If you run `journalctl storage-nas.mount` (or whatever the name of your `.mount` file is), you'll see it complaining about the file being executable when it shouldn't be. Unfortunately, you cannot remove the executable flag on a file stored in a FAT32 partition.

A possible workaround is to unsquash the `SYSTEM` file from Linux (avoid using WSL on Windows, as it wrecks the file permissions for the entire filesystem), add the `.mount` file to the appropriate folder, and then squash it back up. This will successfully automount the share, but it mounts as `root` instead of `lakka`, meaning you won't be able to save files. Does anyone know how to get it to mount with the proper user?

Here are the steps I follow:

1. Verify that `squashfs-tools` is installed:
    ```
    sudo apt update
    sudo apt install squashfs-tools
    mksquashfs --version
    ```
2. Extract the Lakka-Switch.aarch64-5.0.tar file:
    ```
    cd Lakka-Switch.aarch64-5.0
    ```
3. Unsquash the SYSTEM file:
    ```
    sudo unsquashfs -d custom/unsquashed lakka/SYSTEM
    ```
    _(Requires `sudo` to preserve root ownership of the files.)_
4. Edit the `.mount` file: (adjusting filename)
    ```
    sudo nano custom/unsquashed/usr/lib/systemd/system/storage-DSM-Media.mount
    ```
    Paste the following content adjusting directory/`What`/`Where`, `username`, and `password`:
    ```
    [Unit]
    Description=CIFS mount for NAS
    Requires=network-online.service
    After=network-online.service
    Before=retroarch.service
    
    [Mount]
    What=//192.168.1.101/Media
    Where=/storage/DSM/Media
    Options=username=....,password=....,rw,vers=2.0,uid=1000,gid=1000,forceuid,forcegid
    Type=cifs
    
    [Install]
    WantedBy=multi-user.target
    ```
5. Create the symbolic link for the `.mount` file:
    ```
    sudo ln -s ../storage-DSM-Media.mount custom/unsquashed/usr/lib/systemd/system/multi-user.target/storage-DSM-Media.mount
    ```
6. Repack the `SYSTEM` file:
    ```
    sudo mksquashfs custom/unsquashed lakka/SYSTEM -comp lzo -b 524288
    ```
7. Generate the MD5 checksum:
    ```
    sudo md5sum lakka/SYSTEM > lakka/SYSTEM.md5
    ```
8. Replace the existing `SYSTEM` and `SYSTEM.md5` files on your Switch. I use the USB tool in Hekate for this step.

This process will successfully mount the SMB folder at boot. You'll be able to browse the files. However, **LibreELEC ignores the `uid=1000,gid=1000,forceuid,forcegid` options**, so the network folder gets mounted as `root` (`uid=0, gid=0`). It needs to be mounted as `lakka`.

If you try to save a game or a state, it will get stuck at "Saving 0%," and you won't be able to restart, reboot, or shut down. You'll have to force power down the Switch and inject the payload to start Hekate again.

I spent an entire weekend figuring this out, so I'm taking a break for now. If someone else can figure out how to overcome the `root` obstacle, or if this issue gets fixed in LibreELEC, it would be great. Honestly, though, I’m not sure it can be resolved on a FAT32 partition.

My next idea is to add an `automount.sh` script somewhere in the filesystem, but I’ve already spent too much time on this and lost my motivation.

GavinDarkglider, 2025-01-13:
> If you run `journalctl storage-nas.mount` (or whatever the name of your `.mount` file is), you'll see it complaining about the file being executable when it shouldn't be. Unfortunately, you cannot remove the executable flag on a file stored in a FAT32 partition.
> 
> A possible workaround is to unsquash the `SYSTEM` file from Linux (avoid using WSL on Windows, as it wrecks the file permissions for the entire filesystem), add the `.mount` file to the appropriate folder, and then squash it back up. This will successfully automount the share, but it mounts as `root` instead of `lakka`, meaning you won't be able to save files. Does anyone know how to get it to mount with the proper user?
> 
> Here are the steps I follow:
> 
> 1. Verify that `squashfs-tools` is installed:
>    ```
>    sudo apt update
>    sudo apt install squashfs-tools
>    mksquashfs --version
>    ```
> 2. Extract the Lakka-Switch.aarch64-5.0.tar file:
>    ```
>    cd Lakka-Switch.aarch64-5.0
>    ```
> 3. Unsquash the SYSTEM file:
>    ```
>    mkdir custom
>    sudo unsquashfs -d custom/unsquashed lakka/SYSTEM
>    ```
>    
>    
>        
>          
>        
>    
>          
>        
>    
>        
>      
>    _(Requires `sudo` to preserve root ownership of the files.)_
> 4. Edit the `.mount` file: (adjusting filename)
>    ```
>    sudo nano custom/unsquashed/usr/lib/systemd/system/storage-DSM-Media.mount
>    ```
>    
>    
>        
>          
>        
>    
>          
>        
>    
>        
>      
>    Paste the following content adjusting directory/`What`/`Where`, `username`, and `password`:
>    ```
>    [Unit]
>    Description=CIFS mount for NAS
>    Requires=network-online.service
>    After=network-online.service
>    Before=retroarch.service
>    
>    [Mount]
>    What=//192.168.1.101/Media
>    Where=/storage/DSM/Media
>    Options=username=....,password=....,rw,vers=2.0,uid=1000,gid=1000,forceuid,forcegid
>    Type=cifs
>    
>    [Install]
>    WantedBy=multi-user.target
>    ```
> 5. Create the symbolic link for the `.mount` file:
>    ```
>    sudo ln -s ../storage-DSM-Media.mount custom/unsquashed/usr/lib/systemd/system/multi-user.target/storage-DSM-Media.mount
>    ```
> 6. Repack the `SYSTEM` file:
>    ```
>    sudo mksquashfs custom/unsquashed lakka/SYSTEM -comp lzo -b 524288
>    ```
> 7. Generate the MD5 checksum:
>    ```
>    sudo md5sum lakka/SYSTEM > lakka/SYSTEM.md5
>    ```
> 8. Replace the existing `SYSTEM` and `SYSTEM.md5` files on your Switch. I use the USB tool in Hekate for this step.
> 
> This process will successfully mount the SMB folder at boot. You'll be able to browse the files. However, **LibreELEC ignores the `uid=1000,gid=1000,forceuid,forcegid` options**, so the network folder gets mounted as `root` (`uid=0, gid=0`). It needs to be mounted as `lakka`.
> 
> If you try to save a game or a state, it will get stuck at "Saving 0%," and you won't be able to restart, reboot, or shut down. You'll have to force power down the Switch and inject the payload to start Hekate again.
> 
> I spent an entire weekend figuring this out, so I'm taking a break for now. If someone else can figure out how to overcome the `root` obstacle, or if this issue gets fixed in LibreELEC, it would be great. Honestly, though, I’m not sure it can be resolved on a FAT32 partition.
> 
> My next idea is to add an `automount.sh` script somewhere in the filesystem, but I’ve already spent too much time on this and lost my motivation.

shea-sollars, 2025-01-14:
I ended up going with a `storage-DSM-Media.service` file to solve the `root`/`lakka` user issue. The service will successfully automount the NAS on startup under the correct `lakka` user, so that RetroArch can still save games without freezing up.
```
[Unit]
Description=Mount SMB Share for DSM Media with Correct Ownership
Requires=network-online.service
After=network-online.service
Before=retroarch.service

[Service]
Type=oneshot
ExecStart=/bin/mount -t cifs -o username=....,password=....,rw,uid=1000,gid=1000,vers=2.0,soft,timeo=500 //192.168.1.101/Media /storage/DSM/Media
ExecStop=/bin/umount /storage/DSM/Media
RemainAfterExit=true

[Install]
WantedBy=multi-user.target
```

Although this seems to work great at home, I am not sure how it is going to behave when I take the Switch on the road to a different network. I would need to use a VPN in that case anyways. In earlier tests, the whole RetroArch interface would freeze up if WiFi is lost, and cause me to have to force it to power down, then go through the whole inject payload routine. Which is not ideal on the go. Knowing that my whole setup purely relies on my local NAS, I would like to at least be able to boot back into Hekate to use the normal Switch games in Atmosphere if the mounted folder becomes inaccessible, without relying on a payload mechanism. So I am brainstorming some sort of watchdog script to make the network folder service more reliable, maybe even auto toggle VPN when needed.

Another issue is that RetroArch likes to revert save/state directories to default if they become inaccessible before saving the config. They probably to the for some sort of nuance with directory permissions. I might make a bug report on that if I can confirm the cause.

I'm still coming up to speed on how the whole project is put together:

For the `rewritefs` solution, is that something that we need to wait on LibreELEC to implement? If so, I think a possible temporary solution (but still not great) would be to implement a set of simple scripts that checks the storage overlay for a file with some user defined variables that could be used to mount a network folder, and if the file does not exist it should just exit the script. I could probably do that myself to contribute if you don't think it's worth your time, and again, only if the project is relying on LibreELEC to implement `rewritefs`.

GavinDarkglider, 2025-01-14:
> I ended up going with a `storage-DSM-Media.service` file to solve the `root`/`lakka` user issue. The service will successfully automount the NAS on startup under the correct `lakka` user, so that RetroArch can still save games without freezing up.
> 
> ```
> [Unit]
> Description=Mount SMB Share for DSM Media with Correct Ownership
> Requires=network-online.service
> After=network-online.service
> Before=retroarch.service
> 
> [Service]
> Type=oneshot
> ExecStart=/bin/mount -t cifs -o username=....,password=....,rw,uid=1000,gid=1000,vers=2.0,soft,timeo=500 //192.168.1.101/Media /storage/DSM/Media
> ExecStop=/bin/umount /storage/DSM/Media
> RemainAfterExit=true
> 
> [Install]
> WantedBy=multi-user.target
> ```
> 
> Although this seems to work great at home, I am not sure how it is going to behave when I take the Switch on the road to a different network. I would need to use a VPN in that case anyways. In earlier tests, the whole RetroArch interface would freeze up if WiFi is lost, and cause me to have to force it to power down, then go through the whole inject payload routine. Which is not ideal on the go. Knowing that my whole setup purely relies on my local NAS, I would like to at least be able to boot back into Hekate to use the normal Switch games in Atmosphere if the mounted folder becomes inaccessible, without relying on a payload mechanism. So I am brainstorming some sort of watchdog script to make the network folder service more reliable, maybe even auto toggle VPN when needed.
> 
> Another issue is that RetroArch likes to revert save/state directories to default if they become inaccessible before saving the config. It's probably done as some sort of nuance with directory permissions. I might make a bug report on that if I can confirm the cause.
> 
> I'm still coming up to speed on how the whole project is put together:
> 
> For the `rewritefs` solution, is that something that we need to wait on LibreELEC to implement? If so, I think a possible temporary solution (but still not great) would be to implement a set of simple scripts that checks the storage overlay for a file with some user defined variables that could be used to mount a network folder, and if the file does not exist it should just exit the script. I could probably do that myself to contribute if you don't think it's worth your time, and again, only if the project is relying on LibreELEC to implement `rewritefs`.
> 
> Edit: I'll figure out how to build Lakka when I find time, and I'm sure it will all become more clear to me then.

Rewritefs was a hack added to original switch port to make Bluetooth paths work on fat32. This isn't in libreelec, nor is it likely to ever be.

GavinDarkglider, 2025-02-07:
Proper solution: https://github.com/GregorR/upfs should solve all storage dir permissions/symlink issues. Just needs to be implemented.

GavinDarkglider, 2025-02-23:
Pretty sure I just solved this issue, will clean up code, and post pr.

@jdalmanza please test with this build, clean or installed with updater: https://nightly.builds.lakka.tv/members/gavin/Lakka-6.0/Lakka-Switch.aarch64-6.0-devel-20250223015131-3ec2137.tar

Assuming it causes no issues, and fixes the problem, and the fix persists across reboots, I am happy if you are happy

jdalmanza, 2025-02-23:
just tested the build, fresh install. A couple of things:

- Wifi will not join network, it jsut goes to lunch after i enter my passcode
- while troubleshooting, noticed that reboot and shutdown just reset retroarch, does not do full reboot, had to hard boot (hold power for 15 sec) and reboot to hekate
- running hekate 6.2.1 currently if that halps

it might be worth mentioning on other devices i have noticed that with the 6.x builds the network stack loads out of order from the 5.x branch. the cifs share mount fails to run because it's called before the ipv4 stack is initialized. Basically the ipv6 stack loads, satisfies the conditions for the mount service to run, but fails because the network is "offline" ... because ipv4 hasn't initialized yet. As soon as RA boots, you can ssh in and run restart the service and it mounts just fine. As a workaround, we have disabled ipv6 in the cmdline.txt. Since that file is not in the switch build, is there a way to disable ipv6? Or is that not going to be an issue in this build?

GavinDarkglider, 2025-02-23:
> just tested the build, fresh install. A couple of things:
>
> * Wifi will not join network, it jsut goes to lunch after i enter my passcode
> * while troubleshooting, noticed that reboot and shutdown [...]

WiFi and shutdown issues are fix related. I thought I had fixed that, but for some reason since there are no permissions set yet, ownership hasn't been set to anything other than root, and I can't seem to get it to do that on boot. Possible race condition? I will look again tomorrow and get back to you.

GavinDarkglider, 2025-02-25:
So, think I have a viable workaround,until I can work through half the tree and fix permissions all over the place, which is only an issue with switch builds currently. Still needs extensive testing on my part to make sure nothing else breaks along the way.

GavinDarkglider, 2025-02-27:
Think I found a way to just make it work..... Need to test some stuff over the weekend, and get back, it will be a hack of fuse layers, but it should automatically fix ownership across the tree, and ensure user that runs RetroArch/Kodi is the owner of everything in /storage.....

GavinDarkglider, 2025-03-02:
Fixed internally. Leave open until changes are merged. Waiting on tester to make sure everything else still works properly.

GavinDarkglider, 2025-03-02:
@shea-sollars @jdalmanza please test this build:

https://nightly.builds.lakka.tv/members/gavin/Lakka-6.0/Lakka-Switch.aarch64-6.0-devel-20250302081852-4cd0eb4.tar

Add enable_upfs=1 or enable_upfs=2 to lakka ini to enable changes....
Difference between 1 and 2:

1 stores all permissions in ext4 partition image /lakka/storage_p.img in case storage permissions get screwed up(erasing/moving files outside of lakka), erase this, and it will be regenerated on boot. Doing this will disable all user mounts/systems services, so you will need to manually re enable them.

2. Stores permissions in lakka storage folder in .upfs files in every directory. Much harder to clean up reliably, in the case of changing files outside of lakka. I will add a clean up feature if a specific file doesn't exist in storage, just haven't done it yet.

I need both options tested, but generally would suggest using option 1.