libretro/Lakka-LibreELEC #1925: [Feature Request] reboot to bootloader or optional payload on switch

JimZiii, 2024-01-26:
Please add the option to boot into hekate or atmosphere. Currently the reboot function is booting back into lakka through hekate, so it should be easy to implement.

Right now you either have to reboot and hold vol- or shutdown and push a payload again, having the option to boot back into hekate or atmosphere makes it possible to boot into lakka from atmospheres tesla menu and then boot back into atmosphere from lakka, all from the dock.

I googled this a bit and it seems like this used to be possible by changing the "reboot_action" line from payload to bootloader in a file called uenv.txt, a file that no longer exists.

GavinDarkglider, 2024-03-17:
> Please add the option to boot into hekate or atmosphere. Currently the reboot function is booting back into lakka through hekate, so it should be easy to implement.
>
> Right now you either have to [...]

All uenv.txt options have been moved into lakka.ini in bootloader/ini folder. This is standard with all linux distros on the switch now. As for reboot to specific payload, not really a lakka specific issue, as that is code that needs to be implemented in retroarch. Kernel implementation is there, so one in therory could add a script, or something that sets the proper values for their switch to reboot to whatever they want...... and actually, you might be able to do that via hekate ini file as well.

Ntemis, 2024-03-17:
ok to close?

GavinDarkglider, 2024-03-17:
> ok to close?

Can it be closed, since it is a retroarch side of things issue? sure...... But since it is also lakka specific, and on my long list of things I dont have time to work on........ it might be worth leaving, incase someone else gets to it before i do...... which I doubt will happen, as most look at my code in this tree with confusion. lol.

closed 2024-03-18 by Ntemis