libretro/Lakka-LibreELEC #1945: Gamecon module is not enabled in the latest Lakka Development builds

BDJoe360, 2024-03-17:
Since Nightly builds (Generic.x86_64) has not been updated for a long time, only Development builds (Generic.x86_64) can be used for the time being.
However, I found that this version of LAKKA does not have a Gamecon module, and I hope to add it to it.

         Thank you!

BDJoe360, 2024-03-17:
Also, the new Development builds (Lakka-Generic.x86_64-devbuild-v5.x-20240317-95029ef.img.gz) seems to have a problem with Chinese Simplified, and I still get garbled after I choose Chinese font, is it wrong for me to set it up?

gouchi, 2024-03-17:
Related commits [1](https://github.com/libretro/Lakka-LibreELEC/pull/292) [2](https://github.com/libretro/Lakka-LibreELEC/pull/146). 

> Also, the new Development builds (Lakka-Generic.x86_64-devbuild-v5.x-20240317-95029ef.img.gz) seems to have a 
> problem with Chinese Simplified, and I still get garbled after I choose Chinese font, is it wrong for me to set it up?

You should open another issue for this one. Thank you.

gouchi, 2024-04-02:
@BDJoe360 Can you make a [test with latest](https://nightly.builds.lakka.tv/devbuild_latest/Generic.x86_64/) and confirm it is working ?

Thank you.

BDJoe360, 2024-04-07:
OK, I've tested the latest "Lakka-Generic.x86_64-devbuild-v5.x-20240407" and I'm working with the Gamecon controller!

BDJoe360, 2024-04-07:
But there is still a small bug:
[image]
The Device Index does not show the correct model of the controller (the original version shows the PSX Controler).

If you use the Xinput handle series with USB interface, it can be displayed normally
[image]
It would be better if it could be solved!
Thank you.

ToKe79, 2024-04-07:
not sure why the device does not show up with proper name. but at least it works now.
next build (20240408) should fix the issue with chinese symbols (we fixed the path for the fonts in configuration of fontconfig).

closed 2024-04-23 by ToKe79