libretro/Lakka-LibreELEC #1953: Lakka-v5.x : x11.x86_64 : retroarch.service is failed and restarted - repeated.

ShigeakiAsai, 2024-04-04:
### 'retroarch' screen is not shown at the latest Lakka-v5.x-x11.x86_64.
  During this time, retroarch.service is failed and restarted - repeated.
  Please confirm attached [journalctl log](https://github.com/libretro/Lakka-LibreELEC/files/14863821/journalctl-yasaiPC.log).

At this time, it can show 'retroarch' screen by 2 method.
(a) Push [Ctrl] + [Alt] + [F2] via keyboard.
(b) Connect ssh terminal and start 'retroarch' as program.

<BR>

- Which version of Lakka are you using?
Lakka-x11.x86_64-devbuild-v5.x-20240403-d197bf3.img.gz

- What system hardware are you using?
CPU : Intel Core i3-12100T
MB : ASUS H670-PLUS D4 (UEFI:3202)
Mem : DDR4-3200 32GB
VGA : AMD RADEON RX6400
SSD : NVMe SSD 500GB (WD)

- What did you do?
 Collect 'retroarch' log files.
 \* [This](https://github.com/libretro/Lakka-LibreELEC/files/14863852/AsService.retroarch__2024_04_04__17_46_32.log) is retroarch service log. - Fail
 \* [This](https://github.com/libretro/Lakka-LibreELEC/files/14863893/AsProgram.retroarch__2024_04_04__17_53_54.log) is retroarch log as program. - Success

- What did you expect to happen?
Show retroarch screen without retroarch.service failed.

- What happened instead?
Black screen is showed with cursor.

<BR>

P.S.
I have local build environment for Lakka v5.x.
Please let me know if it needs something.

Thanks
ASAI, Shigeaki

gouchi, 2024-04-07:
Hi,

I reproduce this issue using latest nightly `Lakka-x11.x86_64-devbuild-v5.x-20240407-f14fe89.img.gz`

@ToKe79 found that [this commit](https://github.com/libretro/Lakka-LibreELEC/commit/3504404a4c284edce999e542072dd40adedfe477) is missing.

ToKe79, 2024-04-07:
Hello @ShigeakiAsai,
to fix locally, you can add following line to `/storage/.cache/services/retroarch.conf`:
```
DISPLAY=:0.0
```

should by fixed by https://github.com/libretro/Lakka-LibreELEC/commit/4a1181ae64a09fa2150000bf0297b04a1504c06e

please, retest with next dev build. thank you!

ShigeakiAsai, 2024-04-07:
I tested on my local environment about following 2 things.
Both working. Thank you very much.

- Add 'DISPLAY=:0.0' line into /storage/.cache/services/retroarch.conf file on devbuild-v5.x-20240403.
Show retroarch screen and there is no fail retrorach.service,
- Build with your commit.
Show retroarch screen and there is no fail retrorach.service,

I will close this issue after retest on next dev build.

Thanks
ASAI, Shigeaki

ShigeakiAsai, 2024-04-08:
> I will close this issue after retest on next dev build.

Test result is 'OK'.
- retroarch screen is shown.
- There is no fail retrorach.service in [journalctl log](https://github.com/libretro/Lakka-LibreELEC/files/14906209/AfterFix-journalctl.log).

Image : `Lakka-x11.x86_64-devbuild-v5.x-20240408-139b8c1.img.gz`

I close this issue. Thanks!!

P.S.
A following log was recoreded in log file.
`Apr 08 22:33:01 Lakka retroarch[896]: sh: xdg-screensaver: not found`
Is this problem?

closed 2024-04-08 by ShigeakiAsai

ShigeakiAsai, 2024-04-08:
Just Information.

I also confirmed wayland image on same PC environment.
retroarch is not shown, only [1] is shown on left top corner.
I will create new issue for this problem.

Thanks,
ASAI, Shigeaki