libretro/Lakka-LibreELEC #1955: Lakka-v5.x : wayland.x86_64 : retroarch screen is not shown.

ShigeakiAsai, 2024-04-08:
### 'retroarch' screen is not shown at the latest Lakka-v5.x-wayland.x86_64.

  During this time, retroarch.service is NOT failed.
  [[wayland-NG]retroarch__2024_04_08__23_07_03.log](https://github.com/libretro/Lakka-LibreELEC/files/14906623/wayland-NG.retroarch__2024_04_08__23_07_03.log)

  But, Kernel bug is logged in journalctl log.
  [[wayland-NG]journalctl.log](https://github.com/libretro/Lakka-LibreELEC/files/14906639/wayland-NG.journalctl.log)

<BR>

- There is no way around it.

- Which version of Lakka are you using?
`Lakka-wayland.x86_64-devbuild-v5.x-20240408-139b8c1.img.gz`

- What system hardware are you using?
CPU : Intel Core i3-12100T
MB : ASUS H670-PLUS D4 (UEFI:3202)
Mem : DDR4-3200 32GB
VGA : AMD RADEON RX6400
SSD : NVMe SSD 500GB (WD)

- What did you do?
 Collect above log files.

- What did you expect to happen?
Show retroarch screen without retroarch.service failed.

- What happened instead?
Black screen and left top show [1] is shown.

<BR>

P.S.
I have local build environment for Lakka v5.x.
Please let me know if it needs something.

Thanks
ASAI, Shigeaki

ShigeakiAsai, 2024-04-08:
About this issue, I was searching the reason and trying to fix.

It would be fixed by updating nvidia driver package.
- packages/graphics/nvidia/package.mk
current version : PKG_VERSION="520.56.06", This was committed Dec 1, 2022
lates version : PKG_VERSION="550.67", LibreELEC.tv committed Mar 20, 2024

In my local environment, it builds using updated nvidia and libglvnd both packagege.
And this build image shows retroarch.

My English so wrong, if you can't understand I mean, please let me know.
I will more confirm on the other videocard tomorrow.

ToKe79, 2024-04-08:
You can send pull request with your fix and we will review it. 
Thanks!

ShigeakiAsai, 2024-04-09:
## Report
>I will more confirm on the other videocard tomorrow.

This report is test result before updating nvidia/libglvnd package.

<BR>

### Which version of Lakka are you using?
Lakka-wayland.x86_64-devbuild-v5.x-20240408-139b8c1.img.gz

- Intel GPU (CPU integrated) : NG
  \- retroarch is not shown and left top show [1] is shown.
  \- retroarch.service is success.
  \- retroarch log is [here](https://github.com/libretro/Lakka-LibreELEC/files/14912174/intel-NG.retroarch__2024_04_09__09_25_22.log)
  \- journalctl log is [here](https://github.com/libretro/Lakka-LibreELEC/files/14912179/intel-NG.journalctl.log)

- NVIDIA GPU (GT 1030) : NG
  \- retroarch is not shown and black screen is shown.
  \- retroarch.service is fail - restart repeated.
  \- retroarch log is [here](https://github.com/libretro/Lakka-LibreELEC/files/14912205/nvidia-NG.retroarch__2024_04_09__09_36_16.log)
  \- journalctl log is [here](https://github.com/libretro/Lakka-LibreELEC/files/14912209/nvidia-NG.journalctl.log)

ShigeakiAsai, 2024-04-09:
I am tring fix this problem.
https://github.com/ShigeakiAsai/Lakka-LibreELEC/tree/Fix-wayland-x86_64

Intel GPU and AMD GPU are fixed.
But NVIDIA GPU (GT 1030) shows another error.
[image]

I will investigate this error

@gouchi @ToKe79 
Have you any information for this error?

ShigeakiAsai, 2024-04-09:
This caused by NVIDIA Proprietary drivers.
https://www.reddit.com/r/Fedora/comments/10hnrd1/fedora_37_nvidia_proprietary_drivers_acpi_errors/
and more found.

I will NVIDIA driver roll back by reference to [LibreELEC.tv](https://github.com/LibreELEC/LibreELEC.tv).

ToKe79, 2024-04-09:
Yes, do not bother with Nvidia, x11 build is meant for Nvidia for now.

ShigeakiAsai, 2024-04-09:
@ToKe79 
> Yes, do not bother with Nvidia, x11 build is meant for Nvidia for now.

I have a question.
It means this NVIDIA's problem allows on wayland?
It is so, I will send pull request.

Sorry and thanks

ShigeakiAsai, 2024-04-09:
I confirmed several versions nvidia package that is used by LibreELEC.tv.
Error was happened every version on NVIDIA GPU. 
- PKG_VERSION="550.67"
- PKG_VERSION="550.54.14"
- PKG_VERSION="545.29.06"
- PKG_VERSION="545.29.02"
- PKG_VERSION="525.116.03"

I will send pull request using latest nvidia package version.
If it is not OK, it does not merged please.

ShigeakiAsai, 2024-04-09:
Now, Ntemis updates kernel, firmware, driver.
This updates are contained in them.
I would like to wait for this.

ShigeakiAsai, 2024-04-10:
Ntemis done following packages commit.
[nvidia: update to 550.67](https://github.com/libretro/Lakka-LibreELEC/commit/2be35de58dedec7b612ddde7813dd385085edff4)
[libglvnd: update to 1.7.0](https://github.com/libretro/Lakka-LibreELEC/commit/f78b38b0fd07c4ef2fcc3bc5ee276b9917681465)
and more.

By these commit, retroarch on wayland was shown on AMD/Intel GPU I confirmed.
`Lakka-wayland.x86_64-devbuild-v5.x-20240409-c6eb4be.img.gz`

Therefore, I close this issue.

P.S.
retroarch is not shown on NVIDIA GPU, but this is an another NVIDIA driver's problem I think.

Thank you very much
ASAI, Shigeaki

closed 2024-04-10 by ShigeakiAsai