libretro/Lakka-LibreELEC #1956: autoconfig file for retroflag GPi case builds

jdalmanza, 2024-04-08:
Replace the autoconfig file for /joypads/udev/Microsoft X-Box 360 pad.cfg with the following for the 4 gpi case builds

GPICase.arm
Pi02GPi.arm
RPi4-GPICase2.aarch64
RPiZero2-GPiCASE2W.aarch64

NOTE: since the RPi4-GPICase2.aarch64 build does support an external dock with USB ports for controllers it may cause an issue for those that want to use generic xbox 360 controllers  vendor/prod ID# 1118/654 ... apparently this is a very common set of IDs for cloned controllers.

**RetroFlag Gpi Case.cfg**
`input_driver = "udev"
input_device = "Microsoft X-Box 360 pad"
input_device_display_name = "Retroflag Gpi Case"
input_vendor_id = "1118"
input_product_id = "654"

input_up_btn = "h0up"
input_down_btn = "h0down"
input_left_btn = "h0left"
input_right_btn = "h0right"

input_b_btn = "1"
input_y_btn = "3"
input_a_btn = "0"
input_x_btn = "2"
input_l_btn = "4"
input_r_btn = "5"

input_select_btn = "6"
input_start_btn = "7"
input_menu_toggle_btn = "8"

input_b_btn_label = "B"
input_y_btn_label = "Y"
input_select_btn_label = "Select"
input_start_btn_label = "Start"
input_up_btn_label = "D-Pad Up"
input_down_btn_label = "D-Pad Down"
input_left_btn_label = "D-Pad Left"
input_right_btn_label = "D-Pad Right"
input_a_btn_label = "A"
input_x_btn_label = "X"
input_l_btn_label = "L"
input_r_btn_label = "R"
`
<file attached>
[RetroFlag Gpi Case.cfg.zip](https://github.com/libretro/Lakka-LibreELEC/files/14911050/RetroFlag.Gpi.Case.cfg.zip)

jdalmanza, 2024-04-08:
[RetroFlag Gpi Case.cfg.zip](https://github.com/libretro/Lakka-LibreELEC/files/14911136/RetroFlag.Gpi.Case.cfg.zip)

ShigeakiAsai, 2024-04-10:
@ToKe79 
Can I try to fix this issue?

My fix plan is attached patch file puts into GPICase device patch directory.
[GPiCase_pad_config_replace.zip](https://github.com/libretro/Lakka-LibreELEC/files/14933417/GPiCase_pad_config_replace.zip)

Patch file do following 2 things.
- 'Microsoft X-Box 360 pad.cfg' file is removed via patch apply
- 'RetroFlag Gpi Case.cfg' file is added via patch apply

It also remove current button mapping change patch file.

Thanks
ASAI,Shigeaki

ToKe79, 2024-04-10:
this issue is a follow up to a discord discussion, where an issue was raised, that some controllers auto-map incorrectly. the issue was tracked down to xpadneo driver and kernel patch. the kernel patch was responsible for preventing hid-generic to claim some bluetooth gamepads and these were then handled by xpadneo, which then changed VID/PID of the gamepad reported to userspace. after removing the offending patch and xpadneo driver, the issue got fixed.

my proposal, before tracking down the cause of the issue, was to replace the joypad autoconfig files in the image with custom ones. this however after considerng the consequences, is not one of my brightest ideas. these configs might then work for the affected gamepads, but might be incompatible with the real gamepads (e.g. gamepad A is reported as B, we modify the config file for B, user connects gamepad A, retroarch identifies it as B and provides our config for B, user is happy, as buttons match; user connects real gamepad B, retroarch identifies it as gamepad B and provides our config for B, which has incorrect mapping for the real gamepad B, user is unhappy, as buttons do not match).

@jdalmanza the configs currently provided in the images for the above consoles work as expected?

ToKe79, 2024-04-10:
as reported by @bagelbyheart on discord, build without xpadneo and the patch works correctly with xbox one s wireless as well as 8bitdo sn30+ pro gamepads. I will push my changes later.

ToKe79, 2024-04-10:
@jdalmanza on gpicase we [patch](https://github.com/libretro/Lakka-LibreELEC/blob/Lakka-v5.x/projects/RPi/devices/GPICase/patches/retroarch_joypad_autoconfig/retroarch_joypad_autoconfig-01-fix-ABXY-buttons.patch) the config file (we do not change the display name) to exact buttons as you have provided - swapping A<->B and x<->Y ("nintendo" layout). strange that out of box behavior is not as expected.

ToKe79, 2024-04-11:
I prefer in this case the patch route - we are targeting a file that is picked by RetroArch and modifying it for the need of the GPiCase devices. in case something will happen to the file we are patching, we will find out at build-time and fix the issue, rather than having e.g. conflicting files and incorrect one would get picked by RetroArch. I will enhance the patch so it will also show correct labels for the buttons and correct display name.

commit 03a87492 (2024-04-11): GPICase: update patch for internal controls

- update patch
- remove same patch for Pi02GPi, RPi4-GPICase2 and RPiZero2-GPiCASE2W
- provide symlinks for Pi02GPi, RPi4-GPICase2 and RPiZero2-GPiCASE2W for
  easier update of the patch in the future

Closes #1956
closed 2024-04-11 by ToKe79