libretro/Lakka-LibreELEC #1962: Installer leaves system unbootable (N100 Beeline Mini S12)

rtcvb32, 2024-04-13:
Image: Lakka-Generic.x86_64-devbuild-v5.x-20240412-48beb90.img
 System: Beeline 12th gen MiniPC w/ N100; 8Gb Ram, 256Gb Flash storage.

 Had issues installing 4.3, so i tried the nightly build. This bug is for the nightly build. After setting to install there's a brief blip of something that doesn't specify an error happened. Trying to boot leaves me going to the BIOS or USB rather than the installed Lakka OS.

 4.3 would boot but not get past the resize and running state, while the nightly works. Eventually looking at the boot folders, the EFI/BOOT is missing the boot files. 

 Eventually i resolved it by copying the bootia32.efi, bootx64.efi and ldlinux.e64 from the 4.3 image to the installed nightly build drive;. Then the system booted without issue.

[20240413014624.log](https://github.com/libretro/Lakka-LibreELEC/files/14968261/20240413014624.log)

ToKe79, 2024-04-13:
seems like when merging upstream into Lakka the grub package with our changes was overwritten - please try using this development build:
https://nightly.builds.lakka.tv/members/vudiq/devel/Generic.x86_64/

closed 2024-04-13 by ToKe79

ToKe79, 2024-04-13:
fixed images for Lakka 5.0 are now available: https://github.com/libretro/Lakka-LibreELEC/releases/tag/v5.0

commit 3ed5f0a3 (2024-04-13): grub: revert changes from merge

for Lakka grub must copy both 32 and 64 bit EFI loaders, as Lakka does
use syslinux only for legacy (non-EFI) boot. in upstream syslinux is
used for legacy and 64-bit EFI bootloader, grub is used only for 32 bit
EFI bootloader. this makes bootloader consistent for all EFI platforms.

fixes #1962