libretro/Lakka-LibreELEC #1963: [RPi 5] Partition size issue on a 2TB NVMe

arusabu, 2024-04-14:
I'm using a Raspberry Pi 5, a PineberryPi HatDrive! Bottom and a 2TB NVMe, which is my main and only one drive (I want to have os and rom on the same disk).

On the first boot, Lakka doesn't resize correctly partitions :
[image]
(screen taken under an older version of Lakka but the exact same problem occurs with the current version of Lakka, "Lakka (official): 5.0 (RPi5.aarch64)")

The "LAKKA_DISK" partition can't be expanded to 1.82Tio with gparted by using resize2fs in Gparted ("New size results in too many block group descriptors.").

For this 2nd partition, do you use ex4 with 1KiB block size ?
If so, that's the/my issue.

As a workaround, once the NVMe is flashed and before using it, I use gparted to resize the 2nd partition to 2Gio and to create a 3rd one in ExFAT by using the remaining space.
[image]

ToKe79, 2024-04-14:
the partition a creation time is 32 MiB, so `mke2fs` determines automatically small block size, most probably 1 KiB. i think we can go with 4 KiB blocks, so we can add `-b 4096` to `mke2fs`.

closed 2024-04-14 by ToKe79

arusabu, 2024-04-17:
Unless I'm doing something wrong, this issue is still presents on the devel-20240415170722-afaeeac (RPi5.aarch64)

After the 1st boot, I'm getting a new error :
[image]
[image]

And once everything is done and I can boot on Lakka : 
[image]
Storage size appears to be 26 MB.

And here's what I've on GPARTED : 
[image]
Used : 31.36Mio
Unused : 652.00Kio
Unallocated : 1.82Tio
Size : 1.82Tio

ToKe79, 2024-04-17:
Unfortunately I don't have a spare 2TB NVMe drive to debug the issue. You have to use manual partition setup.

ToKe79, 2024-04-17:
@arusabu this image should fix the issue: https://nightly.builds.lakka.tv/members/lakkabuildbot/fix_resize/RPi5.aarch64/

arusabu, 2024-04-18:
@ToKe79 I confirm it does fix this issue.

There is no file system corruption and resizing is correctly performed on initial boot.
[image]

Thank you very much for your support.

TechnologyClassroom, 2024-06-29:
I was going to report this issue and workaround, but it looks like it is fixed. I am seeing this issue on regular 128GB MicroSD cards with the RPi5 Lakka 5.0 img.

This fix alone should probably be reason for a 5.1 release as most people do not know their way around gparted.

commit afaeeacb (2024-04-14): image: create storage partition with 4KiB blocks

by default mke2fs assumes block size by the size of the partition. when
the partition is being resized on an 2TiB disk, the resize fails.
forcing 4 KiB block will allow resize of the parittion on such large
drives.

closes #1963
commit 44a99326 (2024-04-14): image: create storage partition with 4KiB blocks

by default mke2fs assumes block size by the size of the partition. when
the partition is being resized on an 2TiB disk, the resize fails.
forcing 4 KiB block will allow resize of the parittion on such large
drives.

closes #1963
commit 46bb6fa5 (2024-04-17): busybox/resizefs: make fs online during resize2fs

see [1] - online resize is recommended
https://bugs.launchpad.net/ubuntu/+source/e2fsprogs/+bug/1796788/comments/2

fixes #1963