libretro/Lakka-LibreELEC #1964: Rollback options no longer available

port19x, 2024-04-16:
Version: 5.0
Hardware Raspberry Pi 4 - 2GB

I just updated to 5.0.
I assumed that old versions would continue to be accessible under `Online Updates > Update Lakka`.
Instead, `Update Lakka` causes a short black screen, followed by a UI restart.

This bothers me as, unrelated to this, the new mgba core no longer has the color correction option.
Taking away the escape hatch from the update process is a weird choice, if this isn't a bug outright.

ToKe79, 2024-04-16:
it is a bug, we will investigate.

ToKe79, 2024-04-16:
Hello,
the bug was identified. Unfortunately, the only way you can downgrade now is connecting via `ssh`, downloading .tar file using wget into `/storage/.update` (or via copying the .tar file over network share into Update folder) and rebooting.

closed 2024-04-16 by ToKe79

port19x, 2024-05-18:
yikes