libretro/Lakka-LibreELEC #1965: No Lakka 5.0 Rpi4 images

andre-rosa, 2024-04-17:
Link is broken: https://le-builds.lakka.tv/RPi4.aarch64/Lakka-RPi4.aarch64-5.0.img.gz

Could you please also double-check the RPi5 images as well?

Thank you.

andre-rosa, 2024-04-17:
I believe that because of that the online updater does not work too.

ToKe79, 2024-04-17:
new images were uploaded. there was a nasty bug in the original images: everybody who updated to Lakka 5.0 gets a crash when Lakka online updater is opened in RetroArch.

For those who updated to 5.0 already, here are the options:
1) you stick to 5.0 and wait for next version and use the lakka-update.sh script to download the next release (ssh access needed), or
2) you manually put the fixed image to the update folder (either via scp, ssh & wget or samba)

closed 2024-04-17 by ToKe79

andre-rosa, 2024-04-17:
Thank you, @ToKe79!