libretro/Lakka-LibreELEC #1966: Lakka-v5.xhdmi not display

sentball, 2024-04-17:
This issue tracker is NOT for help or support. Please use one of the following channels for help and support:
- the Lakka forum (https://forums.libretro.com/c/libretro/lakka-tv-general)
- via IRC @ #lakkatv on freenode.net
- #lakkatv on libretro discord https://discord.gg/retroarch

Before you report a bug make sure you have tried the latest nightly version of Lakka at https://nightly.builds.lakka.tv/latest/. Your bug might be already fixed.

If you are sure that it's a bug in Lakka, please answer the following questions:
- Which version of Lakka are you using?
- What system hardware are you using?
- What did you do?
- What did you expect to happen?
- What happened instead?

It is important to provide logs for debugging as part of your report. Instructions for generating logs can be found in the Troubleshooting Lakka documentation at https://www.lakka.tv/doc/Troubleshooting-Lakka/

Please delete this text before posting.

sentball, 2024-04-17:
createlog: 
[log-2023-04-03-21.03.42.zip](https://github.com/libretro/Lakka-LibreELEC/files/15008215/log-2023-04-03-21.03.42.zip)

sentball, 2024-04-22:
who can make some help. appreciate with that!

flydt, 2025-12-12:
> who can make some help. appreciate with that!

S905L3B have no HDMI driver, you cannot get output, that's why don't buy S905L3B BOX in china, it cannot run LibreELEC, CoreELEC or anything which need output.