libretro/Lakka-LibreELEC #1968: mGba Core Missing Interframe Blending

BranwonX, 2024-04-20:
This issue tracker is NOT for help or support. Please use one of the following channels for help and support:
- the Lakka forum (https://forums.libretro.com/c/libretro/lakka-tv-general)
- via IRC @ #lakkatv on freenode.net
- #lakkatv on libretro discord https://discord.gg/retroarch

Before you report a bug make sure you have tried the latest nightly version of Lakka at https://nightly.builds.lakka.tv/latest/. Your bug might be already fixed.

If you are sure that it's a bug in Lakka, please answer the following questions:
- Which version of Lakka are you using?
-Tried Lakka 5.0 stable, 5.0 Dev and tried 5.0 Nightly 20240420
- What system hardware are you using?
- Raspberry Pi 4 8GB
- What did you do?
- Loaded GBA Game
- What did you expect to happen?
- Have the option for Interframe Blending In Video Settings of mGba Core Like Lakka 4.3 did for games like Golden Sun.
- What happened instead?
- This Setting was removed completely.

It is important to provide logs for debugging as part of your report. Instructions for generating logs can be found in the Troubleshooting Lakka documentation at https://www.lakka.tv/doc/Troubleshooting-Lakka/

Please delete this text before posting.

ToKe79, 2024-04-21:
Lakka does not change any core settings, are you sure the setting is present when mgba is used with RetroArch on Windows/Linux/Mac/...?

BranwonX, 2024-04-23:
I'm not exactly sure what happened between 4.3 and 5.0 that changed but if you load any GBA game using mGba core, go to quick menu then core options then video and you will see that (Interframe Blending) is missing. Not sure if this is an issue on other devices but it is on a raspberry pi 4 Lakka 5.0, also this setting is still availabe on windows with the latest retroarch, just not on lakka 5.0 on Rpi4.

BranwonX, 2024-04-23:
Everytime from a fresh install of Lakka 5.0 to a SD Card for Raspberry Pi 4 i get this and have no choice but to let the repair continue,

Filesystem corruption has been detected!
To prevent an automatic repair attempt continuing,
press any key or power off your system within the next 120 seconds

Another issue is when you load/exit any game from any core or shut down the system, this message appears briefly,

[     3.420725] systemd[1]: Cannot open /etc/machine-id: No such file or directory
Lakka (official): 5.0
[     201.328696] v3d fec00000.v3d: MMU error from client CLE (4) at 0x7021000, pte invalid

ToKe79, 2024-04-23:
Regarding the filesystem corruption: this looks like the issue with resize2fs, which was already fixed.

To verify if this is the issue, login via ssh and check the size of the `/storage` partition (e.g. `df -h | grep storage`. If it is less than / around 32 MB, then the partition was not properly resized.

You can use [latest build](https://nightly.builds.lakka.tv/latest/) for fresh install/flash, or fix the current install:

1) log in to your RPi4 via ssh while Lakka is running and
2) run the command `resize2fs /dev/mmcblk0p2`

Then check again for partition size - it should be now resized to the remaining capacity of your SD card (i.e. total capacity minus 2 GB boot partition).

Regarding mgba: we use the upstream repo of mGBA to build the libretro core/library for mgba, so we are checking if we should revert back to libretro fork repo (which was used in Lakka 4.x builds).

ToKe79, 2024-04-24:
Hi, you can test devel image from https://nightly.builds.lakka.tv/devbuild_latest/RPi4.aarch64/.
There are now two mgba cores - mgba (libretro fork build) and mgba-upstream (mgba-emu libretro build). the mgba core should have the missing options available. for the test I recommend separate SD card. also the issue with resizing should be gone.

BranwonX, 2024-04-24:
Thanks, i will test this out later today.

BranwonX, 2024-04-24:
So it appears to be a problem with the mGBA upstream build, 2 settings are missing within the core video settings (Color Correction) and (Interframe Blending). I tired the libretro fork build of mGBA with the new lakka 5.0 devbuild you commented above and the settings were back.

Ntemis, 2024-04-24:
please retest next nightly

BranwonX, 2024-04-24:
ok so i just tested the new nightly and it also has those settings back with the libretro fork build of mGBA (mGBA upstream now crashes everytime you load a game), i also get this message upon system shut down,

Lakka (official): devel-20240424154413-67d2ab5
[      31.098509] (sd-unmount)[1083]: Failed to unmount /run/shutdown/mounts/c150520f382a4d47: Device or resource busy
[      31.902557] systemd-shutdown[1]: Failed to finalize file systems, loop devices, ignoring.

If i reboot i get this,

Lakka (official): devel-20240424154413-67d2ab5
[      21.204504] watchdog: watchdog0: watchdog did not stop!
[      23.676242] (sd-unmount)[1069]: Failed to unmount /run/shutdown/mounts/fab941de5ff4a995: Device or resource busy
[      23.679435] watchdog: watchdog0: watchdog did not stop!
[      23.680266] systemd-shutdown[1]: Failed to finalize file systems, loop devices, ignoring.

 sometimes it is followed by a ton of text that just keeps going.

Ntemis, 2024-04-25:
you just tested the @ToKe79 changes but those were reverted. My proper fix will be in the next nightly, sorry for the inconvenience

BranwonX, 2024-04-25:
Ok no worries, I will test the next nighlty when it is available.

ToKe79, 2024-04-25:
hmm... the crashes of mgba (upstream) could be realted to that, that both cores probably use the same settings/config files and some options are not available in the upstream core and hence the crash. lets see with next nightly.

ToKe79, 2024-04-25:
@BranwonX you can test changes proposed by @Ntemis and enhanced by myself:
https://nightly.builds.lakka.tv/members/vudiq/mgba_upstream_changes/
the image still has two mgba cores - mgba (downstream/libretro fork) and mgba_upstream (libretro core built from upstream).
thank you!

BranwonX, 2024-04-25:
I turned on color correction and interframe blending within mgba libretro fork, saved the settings and went to load a game with mgba_upstream and had no issues. Both cores now work regardless of core settings changes to either one.

Ntemis, 2024-04-25:
is better if we keep the upstream one @ToKe79

BranwonX, 2024-04-25:
upstream works just as good as downstream/libretro fork, it is only missing the 2 settings,  certain GBA games have an issue with ghosting effects like Golden Sun in the overworld where the screen pixels shake badly, Interframe Blending fixes this issue.

ToKe79, 2024-04-26:
that means the issue must be reported in the upstream here: https://github.com/mgba-emu/mgba/issues

mention that the options are missing when building with `-DBUILD_LIBRETRO=ON` and also that the forked core has the missing options, which indicates that the libretro (`src/platform/libretro`) part must be fixed. for example `src/platform/libretro/libretro_core_options.h` does not include this whole part from the fork:

https://github.com/libretro/mgba/blob/b2564482c86378581a7a43ef4e254b2a75167bc7/src/platform/libretro/libretro_core_options.h#L175-L209

closed 2024-04-26 by ToKe79

BranwonX, 2024-04-26:
I have submitted the issue to the upstream link you provided, Thanks.

BranwonX, 2024-04-26:
Just thought you should know that this was thier response,

(These options were intentionally not backported from downstream due to their implementation in the core instead of the frontend where they belong. If they were in the frontend they could be implemented in a way that was fast, such as shaders. However, Retroarch decided to implement them in a way that would be slow, especially on weaker systems.)

I personally think the libretro fork of mGBA should be used at the moment until Retroarch gets around to implementing these options, this is just a suggestion, and again the upstream core works just as well but will not fix ghosting issues in certain games.

ToKe79, 2024-04-28:
Thank you for trying. We will bundle `mgba_fork` core, the upstream core will be called `mgba`. This change should be already available in the [current 5.x nightly builds](https://nightly.builds.lakka.tv/2024-04-28_5.x/). In devel branch these are stil `mgba_upstream` and `mgba` but will be changed to same naming as in 5.x.

synthic, 2024-04-29:
Based on endrift's response, aren't shaders the proper solution here? Using `glsl-shaders/motionblur/mix_frames.glslp` handles interframe blending properly (as tested in F-Zero Maximum Velocity). There are shaders available for GBA color correction as well.

BranwonX, 2024-04-29:
I will check out the avilable shaders to see if this is the solution, thanks.

BranwonX, 2024-04-29:
So i justed tested the shader settings with lakka 5.0 stable, all 5.0 nightlies, and dev builds, the mix_frames.glslp you mentoined does indeed fix the ghosting issues on the upstream core of mGBA. I guess i didn't quite understand the comment they made. @ToKe79 and @Ntemis, feel free to disregard this entire thread thanks to @synthic, i did not realize that Interframe blending and GBA Color Correction were already re-introduced into the shaders section.

synthic, 2024-04-30:
@BranwonX I'm glad I could help!

@ToKe79 I think just keeping the upstream build makes the most sense since RetroArch already has shaders to make up for these options and as endrift said it will likely be faster to handle it in the frontend.

commit 67d2ab56 (2024-04-23): have both mgba cores (upstream and libretro fork)

fixes #1968