libretro/Lakka-LibreELEC #1970: Lakka-AMLGX.aarch64-5.0-lepotato - cvbs dont work

efirshik, 2024-04-28:
cvbs out (composite video) dont work.
`find /sys/devices -name "edid"
/sys/devices/platform/soc/d0100000.vpu/drm/card0/card0-HDMI-A-1/edid`
find only HDMI out

ToKe79, 2024-04-28:
does it work with LibreELEC image?

ToKe79, 2024-04-28:
Also - I'm not sure if the dtb provided by Libre Computer for Lakka 4.2 is still compatible with Lakka 5.0.

efirshik, 2024-04-29:
> Also - I'm not sure if the dtb provided by Libre Computer for Lakka 4.2 is still compatible with Lakka 5.0.
thx for answer. i test without replace dtb, and write here after test.

efirshik, 2024-04-29:
without replace dtb:
`/sys/devices/platform/soc/d0100000.vpu/drm/card0/card0-HDMI-A-1/edid
/sys/devices/platform/soc/d0100000.vpu/drm/card0/card0-Composite-1/edid`

but need add option on boot line in file /flash/extlinux/extlinux.conf:
`console=tty0 video=Composite-1:720x480@60ie`
without this cvbs not work.

this solution find on libreelec forum https://forum.libreelec.tv/thread/26430-composite-video-output-on-aml-s905x-cc-le-potato/

dsx724, 2024-04-30:
@efirshik Lakka previously would only work on CVBS if we did not disable it in the DT. Thanks for testing.

closed 2024-04-30 by efirshik