libretro/Lakka-LibreELEC #1974: Lakka Touch Screen Support?

BranwonX, 2024-05-13:
I am using a raspberry pi 4b 8GB model with the official Raspberry Pi 7 Inch Touch Screen and lakka 4.3, the touch screen does not work, how would i go about enabling touch screen? I already went into lakka settings and turned ON enable touch support and rebooted but still did not work. I could not find info on this anywhere.

ToKe79, 2024-05-13:
`CONFIG_DRM_PANEL_RASPBERRYPI_TOUCHSCREEN` is not enabled in kernel configuration.
I will prepare an image with this option enabled for test on your side.

BranwonX, 2024-05-13:
Thanks, and by any chance would you be able to enable this on Lakka 4.3 specifically, if not that is ok i'll still take it in 5.0

ToKe79, 2024-05-14:
Image with enabled `CONFIG_DRM_PANEL_RASPBERRYPI_TOUCHSCREEN` is here:
https://nightly.builds.lakka.tv/members/vudiq/rpi4-touchscreen/

Are there specific reasons why you want to use v4.3?

BranwonX, 2024-05-14:
I will test this now, there is no particular reason in why i prefer 4.3 at this time, it works really well for as long as i have used it. I just usually like to wait for a couple of stable updates in a newer release, so that most, if any, bugs would already be fixed before making it my primary OS.

BranwonX, 2024-05-14:
the touch screen still did not work, menu cannot be controlled and in game overlays do not respond either.

ToKe79, 2024-05-14:
Unfortunately this was the only option that made sense to me, but maybe other options need to enabled as well. If you can get hold of a kernel config from a distro, where the touchscreen works and upload it, I could take a look at the differences. On Ubuntu the config is usually stored in `/boot`, but on Raspberry it might be different location - it should reside next to the kernel and initrd files.

gouchi, 2024-05-14:
It seems that we have to use also `dtoverlay=vc4-kms-dsi-7inch` ?

[Source](https://www.raspberrypi.com/documentation/accessories/display.html#touch-display-device-tree-option-reference)

BranwonX, 2024-05-14:
@gouchi i will test this later today and report back.

BranwonX, 2024-05-14:
@ToKe79 i will see if i can find the config within Raspbian OS.

ToKe79, 2024-05-14:
I guess kernel is configured properly and really (hopefully) only the overlay is missing.

BranwonX, 2024-05-15:
@gouchi dtoverlay=vc4-kms-dsi-7inch or dtoverlay=vc4-kms-dsi-7inch,sizex,sizey still did not allow touch to work, @ToKe79 also not to sure if this matters or not but, the official raspberry screen uses only GPIO pins 3-6 and a ribbon cable to connect from the screen board to the raspberry display ribbon connector.

BranwonX, 2024-05-15:
i added the 3 lines to config.txt and used ssh to apply dtoverlay edt-ft5406 and i kept getting this error,  * Failed to apply overlay '0_edt-ft5406' (kernel)

gouchi, 2024-07-09:
@BranwonX I am confused is it working or not ? Because of #1973.

Thank you.

BranwonX, 2024-07-09:
The touch does not work on the official 7 inch raspberry pi touchscreen with any version of LAKKA, The #1973 issue was about 2 rows of pixels cut off screen but turned out to be a faulty screen.