libretro/Lakka-LibreELEC #1976: Lakka 5.0 fails at resizing the file system

port19x, 2024-05-18:
Sup.

I just received the Raspberry Pi 5 today and noticed that the filesystem does not resize as indicated by the messages during first boot.
I have a 256GB microSD card.
I previously used lakka 5.0 upgraded from 4.3 on a Pi 4.

[image]

Partitions appear to be okay.

During second boot, it complains about a corrupted filesystem, I suspect this is the same error.

port19x, 2024-05-18:
Workaround: [manual resize2fs](https://unix.stackexchange.com/questions/270595/how-do-i-expand-a-file-system-to-fill-a-partition)

<details>
<summary>screenshot</summary>
[image]
</details>

port19x, 2024-05-18:
Could be 46bb6fa

closed 2024-05-18 by ToKe79

ToKe79, 2024-05-18:
the issue presents itself only on SDcards / thumbdrives with size >= 64 GB, so it slipped through during testing. it is already resolved in daily builds.