libretro/Lakka-LibreELEC #1977: More LAKKA 5.0 issues

port19x, 2024-05-18:
I'm not gonna make 1 entry per issue, since I have little hope they get resolved.

## Weird diagonal triangle artifact on transitioins

When I open or close a rom, the top left of the TV is a light blue triangle for a splitsecond

## Stronger color banding in XMB

I don't recall there being any color banding on the prior install of 5.0 on my Pi 4, perhaps related to the previous issue with sth of the graphics pipeline being borked.

## No sound over hdmi

Do I need to elaborate?
Both on default alsathread and alsa

Did anyone actually test the release candidate for Lakka 5.0 on a Raspberry Pi 5?
It's astounding how many issues it has

ToKe79, 2024-05-19:
Hello,

re triangle: this is very subtle and does not occur on every transition, not a breaking issue
re banding: very hard to asses, could you provide some images for comparison?
re sound: here it works out of the box, not sure what happened

Lakka is developed and maitained by limited number of volunteers in their free time (besides their day job and family). Therefore we mostly rely on feedback of the community using daily builds. None of the devs owns all of the target devices Lakka is built for and some devices are not available to any of the devs.

Ntemis, 2024-05-19:
> I'm not gonna make 1 entry per issue, since I have little hope they get resolved.

Did you even count our standing open issues before doing these statements?
We might be the ONLY linux distribution in the world at the time with just 18 unresolved ones.

> ## No sound over hdmi 
> Do I need to elaborate? Both on default alsathread and alsa
Because you cant get audio working on your setup doesn't mean its not working.

> Did anyone actually test the release candidate for Lakka 5.0 on a Raspberry Pi 5? It's astounding how many issues it has
Many many people tested prior to the release and we had zero known issues at the time of it

**P.S.** 
I was about to close this up just by your opening behavior been coming out as aggressive and insulting to the work we do here but since @ToKe79 tried to answer it i let it pass this time.
If you continue coming here with this attitude this is your strike one so i would try to add some politeness next time around. 
Being self-assertive again and you are on your own.

port19x, 2024-05-20:
I'd say I was kind and productive in my first two issues for 5.0, but at the point of opening this I got quite annoyed.

Either way, I guess we have different standards for OSS and quality.
I have now migrated to a competing distro.
Don't think we'll get on good terms

closed 2024-05-20 by port19x

ToKe79, 2024-05-20:
Yes, we have very different standards and ideas, what OSS is. For example I (personally) don't think there is competition in OSS, but there is coexistance. Many other distros are inspired by Lakka and Lakka gets inspired by other distros as well.

But that does not mean we need to be hostile at each other. Next time (but I guess there will be no next time here, but maybe next time when opening issues for other projects) try to be more friendly, try to read other (also closed) issues, check forums - maybe other people had similar issues as you and those issues might have been already solved (in daily builds) or there are workarounds. And maybe a litte bit constructive. If you bring only hate and anger to the table, nobody will be willing to help you.

Ntemis, 2024-05-20:
If and i say "if" there is a next time then logs or gtfo