libretro/Lakka-LibreELEC #1979: Wifi networks not available with Broadcom BCM4360

artandmath, 2024-05-24:
Hi - Wifi networks are not showing on a Mac mini. Have had to do workarounds in some linux on minis to load Broadcom divers and other linux variants they work out of the box. Unable to resolve the issue with Lakka-LibreELEC. Looks like Lakka can see the wireless device with lspci -vnn | grep 14e4: but further to that something else seems to be going on as wifi networks aren't seen. 
 
- Which version of Lakka are you using?
Lakka (official): 5.0 (Generic.x86_64)

- What system hardware are you using?
Macmini7,1 (2.6 GHz Core i5)

- What did you do?
Fresh install, navigate to wifi networks in the graphical UI.

- What did you expect to happen?
Connect to wifi

- What happened instead?
No wifi networks available

Terminal outputs:

Lakka:/ # lspci -vnn | grep 14e4:
02:00.0 Network controller [0280]: Broadcom Inc. and subsidiaries BCM4360 802.11ac Wireless Network Adapter [14e4:43a0] (rev 03)
03:00.0 Ethernet controller [0200]: Broadcom Inc. and subsidiaries NetXtreme BCM57766 Gigabit Ethernet PCIe [14e4:1686] (rev 01)
	Subsystem: Broadcom Inc. and subsidiaries NetXtreme BCM57766 Gigabit Ethernet PCIe [14e4:1686]
03:00.1 SD Host controller [0805]: Broadcom Inc. and subsidiaries BCM57765/57785 SDXC/MMC Card Reader [14e4:16bc] (rev 01) (prog-if 01)
	Subsystem: Broadcom Inc. and subsidiaries BCM57765/57785 SDXC/MMC Card Reader [14e4:0000]

Lakka:/ # ip address
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host 
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP8000> mtu 1500 qdisc mq qlen 1000
    link/ether 78:7b:8a:c3:bd:38 brd ff:ff:ff:ff:ff:ff
    inet 192.168.20.119/24 brd 192.168.20.255 scope global eth0
       valid_lft forever preferred_lft forever
3: wlan0: <NO-CARRIER,BROADCAST,MULTICAST,UP,LOWER_UP8000> mtu 1500 qdisc fq_codel qlen 1000
    link/ether 1c:36:bb:00:6b:82 brd ff:ff:ff:ff:ff:ff
    inet6 fe80::1e36:bbff:fe00:6b82/64 scope link 
       valid_lft forever preferred_lft forever

Lakka:/ # connmanctl enable wifi
wifi is already enabled
Lakka:/ # connmanctl agent on
Error: Not supported in non-interactive mode
Lakka:/ # connmanctl
connmanctl> agent on
connmanctl> Agent registered
Lakka:/ # connmanctl scan wifi
Scan completed for wifi
Lakka:/ # connmanctl services
*AO Wired                ethernet_787b8ac3bd38_cable
Lakka:/ #

ToKe79, 2024-05-24:
Hi, can you check `dmesg` and look for any missing firmware?

artandmath, 2024-05-24:
This is a partial output containing the wl & Broadcom entires

[   10.506106] Bluetooth: Core ver 2.22
[   10.506119] input: HDA Intel PCH Headphone as /devices/pci0000:00/0000:00:1b.0/sound/card1/input10
[   10.506122] NET: Registered PF_BLUETOOTH protocol family
[   10.506123] Bluetooth: HCI device and connection manager initialized
[   10.506127] Bluetooth: HCI socket layer initialized
[   10.506130] Bluetooth: L2CAP socket layer initialized
[   10.506134] Bluetooth: SCO socket layer initialized
[   10.565207] usbcore: registered new interface driver btusb
[   10.566916] cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
[   10.567084] cfg80211: Loaded X.509 cert 'wens: 61c038651aabdcf94bd0ac7ff06c7248db18c600'
[   10.679078] Bluetooth: hci0: BCM: chip id 73 build 1127
[   10.680047] Bluetooth: hci0: BCM: product 05ac:8289
[   10.681043] Bluetooth: hci0: BCM: features 0x07
[   10.691132] usb 1-3.1: USB disconnect, device number 4
[   10.697077] Bluetooth: hci0: BCM20702B0 Generic USB Class 1 @ 20 MHz
[   10.797576] wl: loading out-of-tree module taints kernel.
[   10.797585] wl: module license 'MIXED/Proprietary' taints kernel.
[   10.797588] Disabling lock debugging due to kernel taint
[   10.853798] wlan0: Broadcom BCM43a0 802.11 Hybrid Wireless Controller 6.30.223.271 (r587334)
[   10.853804] 
[   10.962175] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   10.962182] Bluetooth: BNEP filters: protocol multicast
[   10.962187] Bluetooth: BNEP socket layer initialized
[   10.968194] Bluetooth: MGMT ver 1.22
[   11.004815] usb 1-3.2: USB disconnect, device number 5
[   14.892634] tg3 0000:03:00.0 eth0: Link is up at 1000 Mbps, full duplex
[   14.892646] tg3 0000:03:00.0 eth0: Flow control is on for TX and on for RX
[   14.892651] tg3 0000:03:00.0 eth0: EEE is disabled
[   14.892693] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   14.895668] 8021q: 802.1Q VLAN Support v1.8

webpat, 2024-08-23:
Same Issue with Broadcom BCM4315

webpat, 2024-10-24:
here are my dmsg entries regarding the Broadcom wireless card:

> [    0.601233] pci 0000:0c:00.0: [14e4:4315] type 00 class 0x028000`
> [   11.397380] wlan0: Broadcom BCM4315 802.11 Hybrid Wireless Controller 6.30.223.271 (r587334)

and the ifconfig entry for wlan:

> lan0     Link encap:Ethernet  HWaddr 00:22:69:B1:74:4B
>           inet6 addr: fe80::222:69ff:feb1:744b/64 Scope:Link
>           UP BROADCAST MULTICAST  MTU:1500  Metric:1
>           RX packets:0 errors:0 dropped:0 overruns:0 frame:0
>           TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
>           collisions:0 txqueuelen:1000
>           RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
>           Interrupt:17 Base address:0xc000

webpat, 2024-10-24:
[image]
my WLAN card is a DW1397, the driver should be a Broadcom BCM4312