libretro/Lakka-LibreELEC #1982: Raspberri Pi 5 - Nintendo 64 fails to start on all available cores

Foolson, 2024-06-06:
Tested both nightly and delel versions.

When starting a N64 game with any of the cores available the core and/or Lakka crashes ad Lakka restarts.

- Which version of Lakka are you using?
  Lakka-RPi5.aarch64-devel-20240516-a9f41a3.img.gz
- What system hardware are you using?
  Raspberry Pi 5 8GB
- What did you do?
  Start a N64 game.
- What did you expect to happen?
  The game to start.
- What happened instead?
  Lakka and/or the core crashed and Lakka restarted

```
[INFO] RetroArch 1.18.0 (Git 06fa532)
[INFO] === Build =======================================
[INFO] Capabilities: ASIMD 
[INFO] Version: 1.18.0
[INFO] Git: 06fa532
[INFO] Built: May 14 2024
[INFO] =================================================
[INFO] [Input]: Found input driver: "udev".
[INFO] [Environ]: SET_PIXEL_FORMAT: RGB565.
[INFO] [Core]: Version of libretro API: 1, Compiled against API: 1
[INFO] [Core]: Geometry: 320x240, Aspect: 1.333, FPS: 60.00, Sample rate: 48000.00 Hz.
[INFO] [Audio]: Set audio input rate to: 48000.00 Hz.
[INFO] [Video]: Set video size to: 3840x2160.
[INFO] [Video]: Starting threaded video driver..
[INFO] [Vulkan]: Vulkan dynamic library loaded.
[INFO] [Vulkan]: Found vulkan context: "khr_display".
[INFO] [Vulkan]: Detecting screen resolution: 0x0.
[INFO] [Vulkan]: Found GPU at index 0: "V3D 7.1.7".
[INFO] [Vulkan]: Using GPU index 0.
[INFO] [Vulkan]: Found GPU at index 0: "V3D 7.1.7".
[INFO] [Vulkan]: Using GPU index 0.
[INFO] [Vulkan]: Using semaphores for WSI acquire.
[INFO] [Vulkan]: Using GPU: "V3D 7.1.7".
[INFO] [Vulkan]: Queue family 0 supports 1 sub-queues.
[INFO] [Vulkan]: Got 3 swapchain images.
[INFO] [Vulkan]: Using resolution 3840x2160.
[INFO] [Vulkan]: Using RGB565 format.
[INFO] [Vulkan]: Loading stock shader.
[INFO] [Video]: Graphics driver did not initialize an input driver. Attempting to pick a suitable driver.
[DEBUG] [udev] Adding devices of type 0 -> "ID_INPUT_KEY"
[DEBUG] [udev] Adding device (t0) "/sys/devices/platform/pwr_button/input/input3"
[DEBUG] [udev] Adding device (t0) "/sys/devices/platform/pwr_button/input/input3/event3"
[DEBUG] [udev] Adding device (t0) "/sys/devices/virtual/input/input2"
[DEBUG] [udev] Adding device (t0) "/sys/devices/virtual/input/input2/event2"
[DEBUG] [udev] Adding devices of type 1 -> "ID_INPUT_MOUSE"
[DEBUG] [udev] Adding devices of type 2 -> "ID_INPUT_TOUCHPAD"
[WARN] [udev]: Full-screen pointer won't be available.
[INFO] [udev]: Keyboard #0: "pwr_button" (/dev/input/event3).
[INFO] [udev]: Keyboard #1: "cec-mini-kb" (/dev/input/event2).
[INFO] [udev]: Pad #0 (/dev/input/event4) supports force feedback.
[INFO] [udev]: Pad #0 (/dev/input/event4) supports 16 force feedback effects.
[INFO] [Autoconf]: Xbox 360 Controller configured in port 1.
[INFO] [Joypad]: Found joypad driver: "udev".
[INFO] [Video]: Found display server: "null".
[WARN] [udev]: Mouse grab/ungrab feature unavailable.
[INFO] [ALSA] Using ALSA version 1.2.11
[DEBUG] [ALSA]: Requesting device "default" for PLAYBACK stream
[INFO] [ALSA]: Using FLOAT_LE sample format for PLAYBACK device "default"
[INFO] [ALSA]: Period: 4 periods per buffer (768 frames, 6144 bytes)
[INFO] [ALSA]: Buffer size: 3072 frames (24576 bytes)
[INFO] [ALSA]: Can pause: no.
[INFO] [ALSA]: Initialized PLAYBACK device "default"
[INFO] [Audio]: Started synchronous audio driver.
[DEBUG] [ALSA] [playback thread 0x7fffaa95f100]: Beginning playback worker thread
[... 395 lines cut ...]
[DEBUG]     Cartridge (required)
[DEBUG] Subsystems: 2
[ERROR] [MIDI]: snd_seq_open failed with error -2.
[ERROR] [MIDI]: Initialization failed.
[INFO] [Playlist]: Loading history file: "/storage/.config/retroarch/content_history.lpl".
[INFO] [Playlist]: Loading history file: "/storage/.config/retroarch/content_music_history.lpl".
[INFO] [Playlist]: Loading history file: "/storage/.config/retroarch/content_video_history.lpl".
[INFO] [Playlist]: Loading history file: "/storage/.config/retroarch/content_image_history.lpl".
[INFO] [Playlist]: Loading favorites file: "/storage/.config/retroarch/content_favorites.lpl".
[libretro DEBUG] mupen64plus: [EmuThread] M64CMD_EXECUTE
[INFO] [Environ]: SYSTEM_DIRECTORY: "/tmp/system".
[libretro INFO] mupen64plus: Game controller 0 (Standard controller) has a Memory pak plugged in
[libretro INFO] mupen64plus: Game controller 1 (Standard controller) has nothing plugged in
[libretro INFO] mupen64plus: Game controller 2 (Standard controller) has nothing plugged in
[libretro INFO] mupen64plus: Game controller 3 (Standard controller) has nothing plugged in
[libretro INFO] mupen64plus: Using CIC type X105
[INFO] [Environ]: SYSTEM_DIRECTORY: "/tmp/system".
[INFO] [Environ]: SYSTEM_DIRECTORY: "/tmp/system".
[INFO] [Environ]: SYSTEM_DIRECTORY: "/tmp/system".
[INFO] [Environ]: SYSTEM_DIRECTORY: "/tmp/system".
[INFO] [Environ]: SYSTEM_DIRECTORY: "/tmp/system".
[libretro INFO] mupen64plus: Initializing 4 RDRAM modules for a total of 8 MB
[libretro INFO] mupen64plus: Starting R4300 emulator: Dynamic Recompiler
[libretro INFO] mupen64plus: Init new dynarec
```

Ntemis, 2024-06-06:
You  will have to  lower your  resolution to 1080p.
You are  trying to  run the  roms @4K

Foolson, 2024-06-06:
> You will have to lower your resolution to 1080p. You are trying to run the roms @4k

Lowered the resolution in Lakka to 1080p. Still not starting N64 games.

```
[INFO] RetroArch 1.18.0 (Git 06fa532)
[INFO] === Build =======================================
[INFO] Capabilities: ASIMD 
[INFO] Version: 1.18.0
[INFO] Git: 06fa532
[INFO] Built: May 14 2024
[INFO] =================================================
[INFO] [Input]: Found input driver: "udev".
[INFO] [Environ]: SET_PIXEL_FORMAT: RGB565.
[INFO] [Core]: Version of libretro API: 1, Compiled against API: 1
[INFO] [Core]: Geometry: 320x240, Aspect: 1.333, FPS: 60.00, Sample rate: 48000.00 Hz.
[INFO] [Audio]: Set audio input rate to: 48000.00 Hz.
[INFO] [Video]: Set video size to: 1920x1080.
[INFO] [Video]: Starting threaded video driver..
[INFO] [Vulkan]: Vulkan dynamic library loaded.
[INFO] [Vulkan]: Found vulkan context: "khr_display".
[INFO] [Vulkan]: Detecting screen resolution: 0x0.
[INFO] [Vulkan]: Found GPU at index 0: "V3D 7.1.7".
[INFO] [Vulkan]: Using GPU index 0.
[INFO] [Vulkan]: Found GPU at index 0: "V3D 7.1.7".
[INFO] [Vulkan]: Using GPU index 0.
[INFO] [Vulkan]: Using semaphores for WSI acquire.
[INFO] [Vulkan]: Using GPU: "V3D 7.1.7".
[INFO] [Vulkan]: Queue family 0 supports 1 sub-queues.
[INFO] [Vulkan]: Got 3 swapchain images.
[INFO] [Vulkan]: Using resolution 1920x1080.
[INFO] [Vulkan]: Using RGB565 format.
[INFO] [Vulkan]: Loading stock shader.
[INFO] [Video]: Graphics driver did not initialize an input driver. Attempting to pick a suitable driver.
[DEBUG] [udev] Adding devices of type 0 -> "ID_INPUT_KEY"
[DEBUG] [udev] Adding device (t0) "/sys/devices/platform/pwr_button/input/input3"
[DEBUG] [udev] Adding device (t0) "/sys/devices/platform/pwr_button/input/input3/event3"
[DEBUG] [udev] Adding device (t0) "/sys/devices/virtual/input/input2"
[DEBUG] [udev] Adding device (t0) "/sys/devices/virtual/input/input2/event2"
[DEBUG] [udev] Adding devices of type 1 -> "ID_INPUT_MOUSE"
[DEBUG] [udev] Adding devices of type 2 -> "ID_INPUT_TOUCHPAD"
[WARN] [udev]: Full-screen pointer won't be available.
[INFO] [udev]: Keyboard #0: "pwr_button" (/dev/input/event3).
[INFO] [udev]: Keyboard #1: "cec-mini-kb" (/dev/input/event2).
[INFO] [udev]: Pad #0 (/dev/input/event4) supports force feedback.
[INFO] [udev]: Pad #0 (/dev/input/event4) supports 16 force feedback effects.
[INFO] [Autoconf]: Xbox 360 Controller configured in port 1.
[INFO] [Joypad]: Found joypad driver: "udev".
[INFO] [Video]: Found display server: "null".
[WARN] [udev]: Mouse grab/ungrab feature unavailable.
[INFO] [ALSA] Using ALSA version 1.2.11
[DEBUG] [ALSA]: Requesting device "default" for PLAYBACK stream
[INFO] [ALSA]: Using FLOAT_LE sample format for PLAYBACK device "default"
[INFO] [ALSA]: Period: 4 periods per buffer (768 frames, 6144 bytes)
[INFO] [ALSA]: Buffer size: 3072 frames (24576 bytes)
[INFO] [ALSA]: Can pause: no.
[INFO] [ALSA]: Initialized PLAYBACK device "default"
[INFO] [Audio]: Started synchronous audio driver.
[DEBUG] [ALSA] [playback thread 0x7fffa1d9f100]: Beginning playback worker thread
[INFO] [ALSA] Using ALSA version 1.2.11
[INFO] [Microphone]: Initialized microphone driver.
[INFO] [Display]: Found display driver: "vulkan".
[ERROR] [MIDI]: snd_seq_open failed with error -2.
[ERROR] [MIDI]: Initialization failed.
[INFO] [Playlist]: Loading history file: "/storage/.config/retroarch/content_history.lpl".
[INFO] [Playlist]: Loading history file: "/storage/.config/retroarch/content_music_history.lpl".
[INFO] [Playlist]: Loading history file: "/storage/.config/retroarch/content_video_history.lpl".
[INFO] [Playlist]: Loading history file: "/storage/.config/retroarch/content_image_history.lpl".
[INFO] [Playlist]: Loading favorites file: "/storage/.config/retroarch/content_favorites.lpl".
[INFO] [Vulkan]: QueuePresent failed, destroying swapchain.
[... 384 lines cut ...]
[DEBUG]     Cartridge (required)
[DEBUG] Subsystems: 2
[ERROR] [MIDI]: snd_seq_open failed with error -2.
[ERROR] [MIDI]: Initialization failed.
[INFO] [Playlist]: Loading history file: "/storage/.config/retroarch/content_history.lpl".
[INFO] [Playlist]: Loading history file: "/storage/.config/retroarch/content_music_history.lpl".
[INFO] [Playlist]: Loading history file: "/storage/.config/retroarch/content_video_history.lpl".
[INFO] [Playlist]: Loading history file: "/storage/.config/retroarch/content_image_history.lpl".
[INFO] [Playlist]: Loading favorites file: "/storage/.config/retroarch/content_favorites.lpl".
[libretro DEBUG] mupen64plus: [EmuThread] M64CMD_EXECUTE
[INFO] [Environ]: SYSTEM_DIRECTORY: "/tmp/system".
[libretro INFO] mupen64plus: Game controller 0 (Standard controller) has a Memory pak plugged in
[libretro INFO] mupen64plus: Game controller 1 (Standard controller) has nothing plugged in
[libretro INFO] mupen64plus: Game controller 2 (Standard controller) has nothing plugged in
[libretro INFO] mupen64plus: Game controller 3 (Standard controller) has nothing plugged in
[libretro INFO] mupen64plus: Using CIC type X105
[INFO] [Environ]: SYSTEM_DIRECTORY: "/tmp/system".
[INFO] [Environ]: SYSTEM_DIRECTORY: "/tmp/system".
[INFO] [Environ]: SYSTEM_DIRECTORY: "/tmp/system".
[INFO] [Environ]: SYSTEM_DIRECTORY: "/tmp/system".
[INFO] [Environ]: SYSTEM_DIRECTORY: "/tmp/system".
[libretro INFO] mupen64plus: Initializing 4 RDRAM modules for a total of 8 MB
[libretro INFO] mupen64plus: Starting R4300 emulator: Dynamic Recompiler
[libretro INFO] mupen64plus: Init new dynarec
```

synthic, 2024-06-06:
I have a feeling this has to do with the RPi 5 kernel using a 16K pagesize which broke all sorts of software. Patches have already been added for Flycast and DOSBox Pure to address this so these cores will probably require fixes as well. Something as simple as adding `-DPAGESIZE=16384` to the compiler flags should do the trick if that is in fact the issue.

Foolson, 2024-06-07:
I'll give it a try and report back!

gouchi, 2024-06-08:
Related [PR](https://github.com/libretro/Lakka-LibreELEC/pull/1941) for Flycast.

ToKe79, 2024-06-11:
Have you tried latest nightly build for 5.x? Because there were updates to the libretro cores since May 16th in Lakka. On RPi5 we patch the flycast source to "have" 16kB PAGE_SIZE [here](https://github.com/libretro/Lakka-LibreELEC/blob/Lakka-v5.x/projects/RPi/devices/RPi5/patches/flycast/0001-define-pagesize.patch).

Foolson, 2024-06-12:
> The issue with Flycast was that the dynamic recompiler wasn't compatible with the new 16K pagesize. Since your log ends at "Init new dynarec" that is likely exactly what is happening here as well. If you disable the option for it in the config file, there will be a slight performance hit, but it should fix the crash.
> 
> Navigate to `\\lakka\Configfiles\retroarch\config` and modify the cpucore line in the config file for the emulator you are trying to use:
> 
> Mupen64Plus-Next `mupen64plus-cpucore = "cached_interpreter"`
> 
> ParaLLEl N64 `parallel-n64-cpucore = "cached_interpreter"`

Tested, no difference. But have to check the logs, will do after work.

> Have you tried latest nightly build for 5.x? Because there were updates to the libretro cores since May 16th in Lakka. On RPi5 we patch the flycast source to "have" 16kB PAGE_SIZE [...]

Yes, same issue with latest nightly. Will do some more tests with logging and provide the logs after work.

Foolson, 2024-06-12:
Managed to get ParaLLEl N64 to work with cached_interpreter but not Mupen64Plus-Next.

```
[INFO] RetroArch 1.19.0 (Git 00b8a5f)
[INFO] === Build =======================================
[INFO] Capabilities: ASIMD 
[INFO] Version: 1.19.0
[INFO] Git: 00b8a5f
[INFO] Built: May 31 2024
[INFO] =================================================
[INFO] [Input]: Found input driver: "udev".
[ERROR] Couldn't find any cloud sync driver named ""
[INFO] Available cloud sync drivers are:
[INFO] 	webdav
[INFO] 	null
[WARN] Going to default to first cloud sync driver...
[INFO] [Environ]: SET_PIXEL_FORMAT: RGB565.
[INFO] [Core]: Version of libretro API: 1, Compiled against API: 1
[INFO] [Core]: Geometry: 320x240, Aspect: 1.333, FPS: 60.00, Sample rate: 48000.00 Hz.
[INFO] [Audio]: Set audio input rate to: 48000.00 Hz.
[INFO] [Video]: Set video size to: fullscreen.
[INFO] [Video]: Starting threaded video driver..
[WARN] [DRM]: Couldn't get device resources.
[WARN] [DRM]: Couldn't get device resources.
[INFO] [DRM]: Found 2 connectors.
[INFO] [DRM]: Connector 0 connected: yes
[INFO] [DRM]: Connector 0 has 47 modes.
[INFO] [DRM]: Connector 0 assigned to monitor index: #1.
[INFO] [DRM]: Connector 1 connected: no
[INFO] [DRM]: Connector 1 has 0 modes.
[INFO] [DRM]: Mode 0: (3840x2160) 3840 x 2160, 60.000000 Hz
[INFO] [DRM]: Mode 1: (4096x2160) 4096 x 2160, 60.000000 Hz
[INFO] [DRM]: Mode 2: (4096x2160) 4096 x 2160, 59.940098 Hz
[INFO] [DRM]: Mode 3: (4096x2160) 4096 x 2160, 50.000000 Hz
[INFO] [DRM]: Mode 4: (4096x2160) 4096 x 2160, 30.000000 Hz
[INFO] [DRM]: Mode 5: (4096x2160) 4096 x 2160, 29.970001 Hz
[INFO] [DRM]: Mode 6: (4096x2160) 4096 x 2160, 25.000000 Hz
[INFO] [DRM]: Mode 7: (4096x2160) 4096 x 2160, 24.000000 Hz
[INFO] [DRM]: Mode 8: (4096x2160) 4096 x 2160, 23.976000 Hz
[INFO] [DRM]: Mode 9: (3840x2160) 3840 x 2160, 59.940098 Hz
[INFO] [DRM]: Mode 10: (3840x2160) 3840 x 2160, 50.000000 Hz
[INFO] [DRM]: Mode 11: (3840x2160) 3840 x 2160, 30.000000 Hz
[INFO] [DRM]: Mode 12: (3840x2160) 3840 x 2160, 29.970001 Hz
[INFO] [DRM]: Mode 13: (3840x2160) 3840 x 2160, 25.000000 Hz
[INFO] [DRM]: Mode 14: (3840x2160) 3840 x 2160, 24.000000 Hz
[INFO] [DRM]: Mode 15: (3840x2160) 3840 x 2160, 23.976000 Hz
[INFO] [DRM]: Mode 16: (2560x1440) 2560 x 1440, 119.997597 Hz
[INFO] [DRM]: Mode 17: (1920x1080) 1920 x 1080, 120.000000 Hz
[INFO] [DRM]: Mode 18: (1920x1080) 1920 x 1080, 119.880005 Hz
[INFO] [DRM]: Mode 19: (1920x1080) 1920 x 1080, 100.000000 Hz
[INFO] [DRM]: Mode 20: (1920x1080) 1920 x 1080, 60.000000 Hz
[INFO] [DRM]: Mode 21: (1920x1080) 1920 x 1080, 60.000000 Hz
[INFO] [DRM]: Mode 22: (1920x1080) 1920 x 1080, 59.940201 Hz
[INFO] [DRM]: Mode 23: (1920x1080i) 1920 x 1080, 60.000000 Hz
[INFO] [DRM]: Mode 24: (1920x1080i) 1920 x 1080, 59.940201 Hz
[INFO] [DRM]: Mode 25: (1920x1080) 1920 x 1080, 50.000000 Hz
[INFO] [DRM]: Mode 26: (1920x1080i) 1920 x 1080, 50.000000 Hz
[INFO] [DRM]: Mode 27: (1920x1080) 1920 x 1080, 30.000000 Hz
[INFO] [DRM]: Mode 28: (1920x1080) 1920 x 1080, 29.970100 Hz
[INFO] [DRM]: Mode 29: (1920x1080) 1920 x 1080, 25.000000 Hz
[INFO] [DRM]: Mode 30: (1920x1080) 1920 x 1080, 24.000000 Hz
[INFO] [DRM]: Mode 31: (1920x1080) 1920 x 1080, 23.976080 Hz
[INFO] [DRM]: Mode 32: (1280x1024) 1280 x 1024, 60.019741 Hz
[INFO] [DRM]: Mode 33: (1360x768) 1360 x 768, 60.015163 Hz
[INFO] [DRM]: Mode 34: (1152x864) 1152 x 864, 59.966923 Hz
[INFO] [DRM]: Mode 35: (1280x720) 1280 x 720, 60.000000 Hz
[INFO] [DRM]: Mode 36: (1280x720) 1280 x 720, 59.940201 Hz
[INFO] [DRM]: Mode 37: (1280x720) 1280 x 720, 50.000000 Hz
[INFO] [DRM]: Mode 38: (1024x768) 1024 x 768, 60.003841 Hz
[INFO] [DRM]: Mode 39: (800x600) 800 x 600, 60.316540 Hz
[... 615 lines cut ...]
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] [GLSL]: Linking GLSL program.
[INFO] [GL]: Using 1 textures.
[INFO] [GL]: Loaded 1 program(s).
[INFO] [GL]: Initializing HW render (1024x1024).
[INFO] [GL]: Max texture size: 4096 px, renderbuffer size: 4096 px.
[INFO] [GL]: Supports FBO (render-to-texture).
[INFO] [Video]: Graphics driver did not initialize an input driver. Attempting to pick a suitable driver.
[DEBUG] [udev] Adding devices of type 0 -> "ID_INPUT_KEY"
[DEBUG] [udev] Adding device (t0) "/sys/devices/platform/pwr_button/input/input3"
[DEBUG] [udev] Adding device (t0) "/sys/devices/platform/pwr_button/input/input3/event4"
[DEBUG] [udev] Adding device (t0) "/sys/devices/virtual/input/input2"
[DEBUG] [udev] Adding device (t0) "/sys/devices/virtual/input/input2/event2"
[DEBUG] [udev] Adding devices of type 1 -> "ID_INPUT_MOUSE"
[DEBUG] [udev] Adding devices of type 2 -> "ID_INPUT_TOUCHPAD"
[WARN] [udev]: Full-screen pointer won't be available.
[INFO] [udev]: Keyboard #0: "pwr_button" (/dev/input/event4).
[INFO] [udev]: Keyboard #1: "cec-mini-kb" (/dev/input/event2).
[INFO] [udev]: Pad #0 (/dev/input/event3) supports force feedback.
[INFO] [udev]: Pad #0 (/dev/input/event3) supports 16 force feedback effects.
[INFO] [Joypad]: Found joypad driver: "udev".
[INFO] [Video]: Found display server: "kms".
[WARN] [udev]: Mouse grab/ungrab feature unavailable.
[libretro DEBUG] mupen64plus: context_reset()
```

synthic, 2024-06-17:
Looking at the package file for `mupen64plus-next` it contains a long switch statement for all the different devices and their compiler options. However, there is none for `RPi5` so it's likely not being compiled with the correct options at all. We probably need to add a definition with the correct page size, mesa version, etc.

ToKe79, 2024-06-17:
@synthic thank you for the PR. I modified the make options (replaced `rpi-` with `unix-`, as the Makefile was adding incorrect `-mcpu/-mtune` options - for RPi). Also I don't think it is necessary to add `-mcpu/-mtune` options to `CFLAGS`, as these should be already added by the build system / gcc is optimized for the target architecture.
However I looked at the core source and I did not find any place where the define PAGESIZE would be used anywhere, so not sure if it changes anything.
Here is the resulting binary for you guys to test out on RPi5: https://nightly.builds.lakka.tv/members/vudiq/rpi5-mupen64plus_next/

synthic, 2024-06-18:
@ToKe79 Thanks for reviewing it! Unfortunately, I realized today it wasn't building properly so I revised it a bit and pushed a new commit. The incorrect `-mcpu` option was caused because the `CFLAGS` don't seem to be passing through to the Makefile at all (I don't think it even has a proper variable to accept them), so I changed them to `CPUFLAGS` and put them in the switch statement to simplify it a bit, which seems to work properly.

From what I can tell, in the resulting command that the Lakka build system generates there is an `-mtune=cortex-a76` but not a matching `-mcpu` which I believe do different things. Also, including `-mcpu` fixes the issue you mentioned with building, while still including the other general RPi options from the mupen64plus-next Makefile.

Regarding the PAGESIZE, I am not sure if it will fix the issue as I don't have the hardware available to test. But anything starting with `-D` is a gcc macro so it should do something hopefully.

ToKe79, 2024-06-18:
Hi @synthic, thanks for the PR update.
Regarding the `-D` switch, as I wrote, nowhere in the source the macro is used, but as mentioned it looks like mupen is determining the page size at runtime using the sysconf() call, so it might not be necessary. I observed the build commands and the cpu flags are included (using the previous version of your PR with my change of the `platform=...` string:

```
[...]/build.Lakka-RPi5.aarch64/toolchain/bin/aarch64-libreelec-linux-gnu-gcc -march=armv8.2-a+crc+crypto -mabi=lp64 -Wno-psabi -mtune=cortex-a76 -mno-outline-atomics -Wall -pipe  -O2 -fomit-frame-pointer -DNDEBUG -DPAGESIZE=16384 -mcpu=cortex-a76 -mtune=cortex-a76 -std=gnu11 -D_CRT_SECURE_NO_WARNINGS -Wno-discarded-qualifiers -O3 -DNDEBUG -fsigned-char -ffast-math -fno-strict-aliasing -fomit-frame-pointer -fvisibility=hidden -fcommon -DGIT_VERSION=\"" 5d2ac21"\" -DOS_LINUX -I./mupen64plus-core/subprojects/md5 -D__STDC_CONSTANT_MACROS -D__STDC_LIMIT_MACROS -D__LIBRETRO__ -DUSE_FILE32API -DM64P_PLUGIN_API -DM64P_CORE_PROTOTYPES -D_ENDUSER_RELEASE -DSINC_LOWER_QUALITY -DTXFILTER_LIB -D__VEC4_OPT -DMUPENPLUSAPI -I./custom -I./custom/mupen64plus-core -I./custom/android/include -I./custom/GLideN64 -I./GLideN64/src -I./GLideN64/src/osal -I./mupen64plus-core/src -I./mupen64plus-core/src/api -I./custom/mupen64plus-core/plugin/audio_libretro -I./libretro-common/include -I./libretro -I./GLideN64/src/inc -I./custom/dependencies/libpng -I./mupen64plus-core/subprojects/minizip -I./xxHash -I./custom/dependencies/libzlib  -fPIC  -DEGL -DHAVE_OPENGLES -DHAVE_OPENGLES2 -DGLES2 -DNEW_DYNAREC=4 -DDYNAREC -I./mupen64plus-core/src/asm_defines/ -c mupen64plus-core/src/device/r4300/new_dynarec/arm64/linkage_arm64.S -o mupen64plus-core/src/device/r4300/new_dynarec/arm64/linkage_arm64.o
```

synthic, 2024-06-18:
@ToKe79 Oh, good! Thanks for clarifying and sorry I didn't see your earlier note. I just wanted to be doubly sure that I was matching the expected behavior. Feel free to edit as necessary. I am not sure that the page size is the real culprit either, but this case just seems awfully similar to what has been happening with other applications since the Raspberry Pi 5 has been released. The fact that there was no `RPi5` options in our package file at all is also suspect though, so that might just be all it was. Hopefully one of the two fixes will have solved the issue.

jonian, 2024-06-18:
Regarding the 16K page size in mupen64plus it has been reported upstream mupen64plus/mupen64plus-core#1047.

synthic, 2024-06-21:
So I got my hands on a Raspberry Pi 5 and was able to do some testing. I tried the binary provided by @ToKe79 and mupen64plus was still crashing. Then, I decided to compile my own binary using the package file as it is in my latest PR revision and it started right up! Performance is great on the new model and everything seems to work properly. :)

synthic, 2024-06-30:
@Foolson mupen64plus should be working on the latest nightly, if you want to try it out and close the issue.

spork960, 2024-07-09:
@synthic I have a Raspberry Pi 5 8 GB model and am using the latest 5.x Lakka nightly (Lakka-RPi5.aarch64-5.x-20240622-63ed9e5.img.gz). The Mupen64Plus-Next core crashes when it tries to launch a game. The ParaLLEl N64 core is working correctly.

[retroarch.log](https://github.com/user-attachments/files/16136097/retroarch.log)

closed 2024-07-10 by Ntemis

synthic, 2024-07-10:
@spork960 It seems like you aren't using the default config because the core is trying to use the cached interpreter instead of the dynamic recompiler, although that should work as well. So try deleting your config file and lowering the resolution to 1080p.

Foolson, 2024-07-10:
Sorry for the late reply, have been busy.

Flashed the latest nightly and changed the resolution to 1080p.

Both cores still crash.

```
[INFO] RetroArch 1.19.1 (Git 0792144)
[INFO] === Build =======================================
[INFO] Capabilities: ASIMD 
[INFO] Version: 1.19.1
[INFO] Git: 0792144
[INFO] Built: Jun 13 2024
[INFO] =================================================
[INFO] [Input]: Found input driver: "udev".
[ERROR] Couldn't find any cloud sync driver named ""
[INFO] Available cloud sync drivers are:
[INFO] 	webdav
[INFO] 	null
[WARN] Going to default to first cloud sync driver...
[INFO] [Environ]: SET_PIXEL_FORMAT: RGB565.
[INFO] [Core]: Version of libretro API: 1, Compiled against API: 1
[INFO] [Core]: Geometry: 320x240, Aspect: 1.333, FPS: 60.00, Sample rate: 48000.00 Hz.
[INFO] [Audio]: Set audio input rate to: 48000.00 Hz.
[INFO] [Video]: Set video size to: 1920x1080.
[INFO] [Video]: Starting threaded video driver..
[INFO] [Vulkan]: Vulkan dynamic library loaded.
[INFO] [Vulkan]: Found vulkan context: "khr_display".
[INFO] [Vulkan]: Detecting screen resolution: 0x0.
[INFO] [Vulkan]: Found GPU at index 0: "V3D 7.1.7".
[INFO] [Vulkan]: Using GPU index 0.
[INFO] [Vulkan]: Found GPU at index 0: "V3D 7.1.7".
[INFO] [Vulkan]: Using GPU index 0.
[INFO] [Vulkan]: Using semaphores for WSI acquire.
[INFO] [Vulkan]: Using GPU: "V3D 7.1.7".
[INFO] [Vulkan]: Queue family 0 supports 1 sub-queues.
[INFO] [Vulkan]: Got 3 swapchain images.
[INFO] [Vulkan]: Using resolution 1920x1080.
[INFO] [Vulkan]: Using RGB565 format.
[INFO] [Vulkan]: Loading stock shader.
[INFO] [Video]: Graphics driver did not initialize an input driver. Attempting to pick a suitable driver.
[DEBUG] [udev] Adding devices of type 0 -> "ID_INPUT_KEY"
[DEBUG] [udev] Adding device (t0) "/sys/devices/platform/pwr_button/input/input3"
[DEBUG] [udev] Adding device (t0) "/sys/devices/platform/pwr_button/input/input3/event3"
[DEBUG] [udev] Adding device (t0) "/sys/devices/virtual/input/input2"
[DEBUG] [udev] Adding device (t0) "/sys/devices/virtual/input/input2/event2"
[DEBUG] [udev] Adding devices of type 1 -> "ID_INPUT_MOUSE"
[DEBUG] [udev] Adding devices of type 2 -> "ID_INPUT_TOUCHPAD"
[WARN] [udev]: Full-screen pointer won't be available.
[INFO] [udev]: Keyboard #0: "pwr_button" (/dev/input/event3).
[INFO] [udev]: Keyboard #1: "cec-mini-kb" (/dev/input/event2).
[INFO] [udev]: Pad #0 (/dev/input/event4) supports force feedback.
[INFO] [udev]: Pad #0 (/dev/input/event4) supports 16 force feedback effects.
[INFO] [Autoconf]: Xbox 360 Controller configured in port 1.
[INFO] [Joypad]: Found joypad driver: "udev".
[INFO] [Video]: Found display server: "null".
[WARN] [udev]: Mouse grab/ungrab feature unavailable.
[INFO] [ALSA] Using ALSA version 1.2.11
[DEBUG] [ALSA]: Requesting device "default" for PLAYBACK stream
[INFO] [ALSA]: Using FLOAT_LE sample format for PLAYBACK device "default"
[INFO] [ALSA]: Period: 4 periods per buffer (768 frames, 6144 bytes)
[INFO] [ALSA]: Buffer size: 3072 frames (24576 bytes)
[INFO] [ALSA]: Can pause: no.
[INFO] [ALSA]: Initialized PLAYBACK device "default"
[INFO] [Audio]: Started synchronous audio driver.
[DEBUG] [ALSA] [playback thread 0x7fffa520f100]: Beginning playback worker thread
[INFO] [ALSA] Using ALSA version 1.2.11
[INFO] [Microphone]: Initialized microphone driver.
[INFO] [Display]: Found display driver: "vulkan".
[ERROR] [MIDI]: snd_seq_open failed with error -2.
[... 763 lines cut ...]
[DEBUG] [Environ]: GET_VARIABLE: parallel-n64-screensize = "640x480"
[DEBUG] [Environ]: GET_VARIABLE: parallel-n64-angrylion-vioverlay = "Filtered"
[DEBUG] [Environ]: GET_VARIABLE: parallel-n64-angrylion-sync = "Low"
[DEBUG] [Environ]: GET_VARIABLE: parallel-n64-angrylion-multithread = "all threads"
[DEBUG] [Environ]: GET_VARIABLE: parallel-n64-angrylion-overscan = "disabled"
[DEBUG] [Environ]: GET_VARIABLE: parallel-n64-filtering = "automatic"
[DEBUG] [Environ]: GET_VARIABLE: parallel-n64-dithering = "enabled"
[libretro DEBUG] set dithering mode...
[DEBUG] [Environ]: GET_VARIABLE: parallel-n64-polyoffset-factor = "-3.0"
[DEBUG] [Environ]: GET_VARIABLE: parallel-n64-polyoffset-units = "-3.0"
[DEBUG] [Environ]: GET_VARIABLE: parallel-n64-astick-deadzone = "15"
[DEBUG] [Environ]: GET_VARIABLE: parallel-n64-astick-sensitivity = "100"
[DEBUG] [Environ]: GET_VARIABLE: parallel-n64-gfxplugin-accuracy = "veryhigh"
[DEBUG] [Environ]: GET_VARIABLE: parallel-n64-virefresh = "auto"
[DEBUG] [Environ]: GET_VARIABLE: parallel-n64-bufferswap = "disabled"
[DEBUG] [Environ]: GET_VARIABLE: parallel-n64-framerate = "original"
[DEBUG] [Environ]: GET_VARIABLE: parallel-n64-alt-map = "disabled"
[DEBUG] [Environ]: GET_VARIABLE: parallel-n64-pak1 = "none"
[DEBUG] [Environ]: GET_VARIABLE: parallel-n64-pak2 = "none"
[DEBUG] [Environ]: GET_VARIABLE: parallel-n64-pak3 = "none"
[DEBUG] [Environ]: GET_VARIABLE: parallel-n64-pak4 = "none"
[libretro DEBUG] set filtering mode...
[libretro INFO] mupen64plus: Starting R4300 emulator: Dynamic Recompiler
[libretro INFO] mupen64plus: Init new dynarec
```

jonian, 2024-07-10:
@Foolson dynarec can't be fixed in lakka or the libretro core, it has to be fixed upstream (https://github.com/mupen64plus/mupen64plus-core/issues/1047). If you try with cached interpreter it will work.

Foolson, 2024-07-10:
Does not work with cached interpreter either I'm afraid.

```
[INFO] RetroArch 1.19.1 (Git 0792144)
[INFO] === Build =======================================
[INFO] Capabilities: ASIMD 
[INFO] Version: 1.19.1
[INFO] Git: 0792144
[INFO] Built: Jun 13 2024
[INFO] =================================================
[INFO] [Input]: Found input driver: "udev".
[ERROR] Couldn't find any cloud sync driver named ""
[INFO] Available cloud sync drivers are:
[INFO] 	webdav
[INFO] 	null
[WARN] Going to default to first cloud sync driver...
[INFO] [Environ]: SET_PIXEL_FORMAT: RGB565.
[INFO] [Core]: Version of libretro API: 1, Compiled against API: 1
[INFO] [Core]: Geometry: 320x240, Aspect: 1.333, FPS: 60.00, Sample rate: 48000.00 Hz.
[INFO] [Audio]: Set audio input rate to: 48000.00 Hz.
[INFO] [Video]: Set video size to: 1920x1080.
[INFO] [Video]: Starting threaded video driver..
[WARN] [DRM]: Couldn't get device resources.
[WARN] [DRM]: Couldn't get device resources.
[INFO] [DRM]: Found 2 connectors.
[INFO] [DRM]: Connector 0 connected: yes
[INFO] [DRM]: Connector 0 has 47 modes.
[INFO] [DRM]: Connector 0 assigned to monitor index: #1.
[INFO] [DRM]: Connector 1 connected: no
[INFO] [DRM]: Connector 1 has 0 modes.
[INFO] [DRM]: Mode 0: (3840x2160) 3840 x 2160, 60.000000 Hz
[INFO] [DRM]: Mode 1: (4096x2160) 4096 x 2160, 60.000000 Hz
[INFO] [DRM]: Mode 2: (4096x2160) 4096 x 2160, 59.940098 Hz
[INFO] [DRM]: Mode 3: (4096x2160) 4096 x 2160, 50.000000 Hz
[INFO] [DRM]: Mode 4: (4096x2160) 4096 x 2160, 30.000000 Hz
[INFO] [DRM]: Mode 5: (4096x2160) 4096 x 2160, 29.970001 Hz
[INFO] [DRM]: Mode 6: (4096x2160) 4096 x 2160, 25.000000 Hz
[INFO] [DRM]: Mode 7: (4096x2160) 4096 x 2160, 24.000000 Hz
[INFO] [DRM]: Mode 8: (4096x2160) 4096 x 2160, 23.976000 Hz
[INFO] [DRM]: Mode 9: (3840x2160) 3840 x 2160, 59.940098 Hz
[INFO] [DRM]: Mode 10: (3840x2160) 3840 x 2160, 50.000000 Hz
[INFO] [DRM]: Mode 11: (3840x2160) 3840 x 2160, 30.000000 Hz
[INFO] [DRM]: Mode 12: (3840x2160) 3840 x 2160, 29.970001 Hz
[INFO] [DRM]: Mode 13: (3840x2160) 3840 x 2160, 25.000000 Hz
[INFO] [DRM]: Mode 14: (3840x2160) 3840 x 2160, 24.000000 Hz
[INFO] [DRM]: Mode 15: (3840x2160) 3840 x 2160, 23.976000 Hz
[INFO] [DRM]: Mode 16: (2560x1440) 2560 x 1440, 119.997597 Hz
[INFO] [DRM]: Mode 17: (1920x1080) 1920 x 1080, 120.000000 Hz
[INFO] [DRM]: Mode 18: (1920x1080) 1920 x 1080, 119.880005 Hz
[INFO] [DRM]: Mode 19: (1920x1080) 1920 x 1080, 100.000000 Hz
[INFO] [DRM]: Mode 20: (1920x1080) 1920 x 1080, 60.000000 Hz
[INFO] [DRM]: Mode 21: (1920x1080) 1920 x 1080, 60.000000 Hz
[INFO] [DRM]: Mode 22: (1920x1080) 1920 x 1080, 59.940201 Hz
[INFO] [DRM]: Mode 23: (1920x1080i) 1920 x 1080, 60.000000 Hz
[INFO] [DRM]: Mode 24: (1920x1080i) 1920 x 1080, 59.940201 Hz
[INFO] [DRM]: Mode 25: (1920x1080) 1920 x 1080, 50.000000 Hz
[INFO] [DRM]: Mode 26: (1920x1080i) 1920 x 1080, 50.000000 Hz
[INFO] [DRM]: Mode 27: (1920x1080) 1920 x 1080, 30.000000 Hz
[INFO] [DRM]: Mode 28: (1920x1080) 1920 x 1080, 29.970100 Hz
[INFO] [DRM]: Mode 29: (1920x1080) 1920 x 1080, 25.000000 Hz
[INFO] [DRM]: Mode 30: (1920x1080) 1920 x 1080, 24.000000 Hz
[INFO] [DRM]: Mode 31: (1920x1080) 1920 x 1080, 23.976080 Hz
[INFO] [DRM]: Mode 32: (1280x1024) 1280 x 1024, 60.019741 Hz
[INFO] [DRM]: Mode 33: (1360x768) 1360 x 768, 60.015163 Hz
[INFO] [DRM]: Mode 34: (1152x864) 1152 x 864, 59.966923 Hz
[INFO] [DRM]: Mode 35: (1280x720) 1280 x 720, 60.000000 Hz
[INFO] [DRM]: Mode 36: (1280x720) 1280 x 720, 59.940201 Hz
[INFO] [DRM]: Mode 37: (1280x720) 1280 x 720, 50.000000 Hz
[INFO] [DRM]: Mode 38: (1024x768) 1024 x 768, 60.003841 Hz
[INFO] [DRM]: Mode 39: (800x600) 800 x 600, 60.316540 Hz
[... 615 lines cut ...]
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] [GLSL]: Linking GLSL program.
[INFO] [GL]: Using 1 textures.
[INFO] [GL]: Loaded 1 program(s).
[INFO] [GL]: Initializing HW render (1024x1024).
[INFO] [GL]: Max texture size: 4096 px, renderbuffer size: 4096 px.
[INFO] [GL]: Supports FBO (render-to-texture).
[INFO] [Video]: Graphics driver did not initialize an input driver. Attempting to pick a suitable driver.
[DEBUG] [udev] Adding devices of type 0 -> "ID_INPUT_KEY"
[DEBUG] [udev] Adding device (t0) "/sys/devices/platform/pwr_button/input/input3"
[DEBUG] [udev] Adding device (t0) "/sys/devices/platform/pwr_button/input/input3/event3"
[DEBUG] [udev] Adding device (t0) "/sys/devices/virtual/input/input2"
[DEBUG] [udev] Adding device (t0) "/sys/devices/virtual/input/input2/event2"
[DEBUG] [udev] Adding devices of type 1 -> "ID_INPUT_MOUSE"
[DEBUG] [udev] Adding devices of type 2 -> "ID_INPUT_TOUCHPAD"
[WARN] [udev]: Full-screen pointer won't be available.
[INFO] [udev]: Keyboard #0: "pwr_button" (/dev/input/event3).
[INFO] [udev]: Keyboard #1: "cec-mini-kb" (/dev/input/event2).
[INFO] [udev]: Pad #0 (/dev/input/event4) supports force feedback.
[INFO] [udev]: Pad #0 (/dev/input/event4) supports 16 force feedback effects.
[INFO] [Joypad]: Found joypad driver: "udev".
[INFO] [Video]: Found display server: "kms".
[WARN] [udev]: Mouse grab/ungrab feature unavailable.
[libretro DEBUG] mupen64plus: context_reset()
```

pawa75, 2024-07-21:
Parallel N64 works for me on pi5, but only if I choose Angrylion as RDP plugin.