libretro/Lakka-LibreELEC #1983: GPi Case 2 : RetroArch tries to display on the internal monitor when docked

imaspeer, 2024-06-09:
Using a Raspberry Pi Compute Module 4 in a RetroFlag GPi Case 2 and the official RPi4-GPICase2 5.0 image in the default configuration :

When the system is connected to its dock, with the dock's HDMI output connected to a monitor, the internal monitor turns off (correctly) and the external monitor only displays the flower screen, in a 640x480 box in the top left.

It is expected RetroArch shows up on the external monitor instead ; or at the very least if it doesn't switch automatically the internal monitor should not be turned off, as this leaves the user without a way to interact with the system.

Setting `video_monitor_index="2"` in `retroarch.cfg` fixes it, but that breaks the internal monitor (only shows the flower) when not docked.

This happens regardless of if the system is started while already connected to the dock or if it is hotplugged. (On the plus side, the latter case is technically an improvement over the old switching scripts that didn't support hotplugging at all... But it'd be better if it actually worked :D )

Logs don't seem to have anything interesting, still including them just in case:
[retroarch -v with the bug](https://github.com/user-attachments/files/15753609/log_ko.log)
[retroarch -v with retroarch.cfg workaround applied](https://github.com/user-attachments/files/15753610/log_with_workaround.log)

imaspeer, 2024-06-09:
I investigated a bit more and there doesn't seem to be a way to detect whether an external monitor is plugged in or not. In particular `/sys/class/drm/card0/card0-HDMI-A-1/status` is always `connected` even while not on the dock, the edid is also always the same regardless of what monitor is connected or if there is one at all.

The dock itself, however, is detectable as a USB hub with id 1a40:0101. One such hub is always present and a second one is added when docked.

I've come up with a workaround that changes the config and restarts RetroArch when the hub is added or removed. Definitely not a great way to do it, but I couldn't find anything better.

`.config/set-monitor.sh`
```sh
#!/bin/sh
if [[ $(lsusb -d 1a40:0101 |wc -l) -gt 1 ]]; then
	echo 'video_monitor_index = "2"' >/tmp/retroarch-monitor.cfg
else
	echo 'video_monitor_index = "0"' >/tmp/retroarch-monitor.cfg
fi
```

`.config/systemd.d/retroarch.service.d/99-set-monitor.conf`
```
[Service]
ExecStartPre=-/storage/.config/set-monitor.sh
ExecStart=
ExecStart=/usr/bin/retroarch --appendconfig=/tmp/retroarch-monitor.cfg
```

`.config/udev/rules.d/99-gpicase2.dock.rules`
```
ACTION=="add", SUBSYSTEM=="usb", ATTRS{idVendor}=="1a40", ATTRS{idProduct}=="0101", RUN+="/usr/bin/systemctl restart retroarch"
ACTION=="remove", SUBSYSTEM=="usb", ATTRS{idVendor}=="1a40", ATTRS{idProduct}=="0101", RUN+="/usr/bin/systemctl restart retroarch"
```

ShigeakiAsai, 2025-09-05:
I tried your workaround. It's great!
It's your workaround.
https://github.com/user-attachments/assets/ba1981c8-4581-4653-8aca-71fec4b02853

Reference: without workaround
https://github.com/user-attachments/assets/9a761b16-f74a-4c54-8956-0a6c1949268d

image : Lakka-v6.x nightly 2025-05-09
Lakka-RPi4-GPiCase2.aarch64-6.x-20250509-efa398f.img.gz

No modifying your source files but I changed location directory.
755 /storage/.config/set-monitor.sh
644 /storage/.config/system.d/retroarch.service.d/99-set-monitor.conf (It changes the directory 'systemd.d' -> 'system.d')
644 /storage/.config/udev.rules.d/99-gpicase2.dock.rules (It changes the directory 'udev/rules.d' -> 'udev.rules.d' )

I will try to merge this into Lakka tree (^_^)y

Note:
Current RPi4-GPiCase2.aarch64 is working on Fake-KMS.
This workaround was valid on the Fake-KMS.
I wasn't able to use this workaround on the Full-KMS.
I don't know why now, I will investigate later.

Thanks!
ASAI, Shigeaki

ShigeakiAsai, 2025-09-05:
I've finished packaging and enabling.
https://github.com/ShigeakiAsai/Lakka-LibreELEC/tree/RPi4-GPiCase2_Dock_HDMI_support

If you want to try it, please use the following release. 
https://github.com/ShigeakiAsai/Lakka-LibreELEC/releases/tag/GPiCASE2-Dock-HDMI.01

I chenged file names.
set-monitor.sh -> gpicase2-set-monitor.sh
99-set-monitor.conf -> 99-gpicase2-set-monitor.conf
99-gpicase2.dock.rules -> 99-gpicase2_dock_hdmi.rules

Thanks!
ASAI, Shigeaki

ShigeakiAsai, 2025-09-05:
More 1 thing.

Offical Retroflag scripts use to refer GPIO18 for dock detection.
https://github.com/RetroFlag/GPiCase2-Script/blob/de498e9cb378af70a1509cf0473036ae7f74b04e/lcdnext.sh#L2

Current Lakka-v6.x can't use GPIO18 because it is set a2(Alt2) mode in distroconfig.txt.
https://github.com/libretro/Lakka-LibreELEC/blob/3f0077a625b80e597fbdf91f49c33d5caaad4229/projects/RPi/devices/RPi4-GPiCase2/config/distroconfig.txt#L16

GPiCASE2 LCD and GPiCASE2 LCD use Mode 6 (RGB bits 666), so GPIO18 is not used.
https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#gpio-pins

It might be good to free GPIO18 on GPiCASE2.