libretro/Lakka-LibreELEC #1993: Lakka bootloops on Nintendo Switch

Polarite, 2024-06-28:
- Which version of Lakka are you using? Tried Latest and Nightly
- What system hardware are you using? Nintendo Switch Modchipped, updated the sdloader(2.75) and hekate to latest version(6.2.0)
- What did you do? Install as explained in the documentation downloaded 7z version, exported files did the hekate thing and tried starting.
- What did you expect to happen? Start up to the lakka
- What happened instead? Load up begins, then bottom bar fully loads then 2 white-ish lines appear on screen and then freezes on the loading screen.

GavinDarkglider, 2024-06-28:
Release has been tested working, get latest hekate, there were some bugs that broke Linux/android booting, that have since been fixed in revision 4 of that release.

Ntemis, 2024-06-28:
ok to close?

Polarite, 2024-06-29:
Sadly issue still persists even after double checking the 6.2.0 update, its V4 still not working...

GavinDarkglider, 2024-06-29:
> Sadly issue still persists even after double checking the 6.2.0 update, its V4 still not working... EDIT: after downgrading hekate to 6.1.1 the issue still persists, which makes me think the issue is not related to hekate..?

I will look into it, but this is the first I am hearing of this since the release.

Polarite, 2024-06-29:
do you need a video to see the white lines appearing? that doesnt happen when booting other stuff

GavinDarkglider, 2024-06-29:
If by other stuff you mean hos, that follows a different bootchain. The lines are caused by ram training on boot. This is a side effect of seamless display. What it does after that is what I am interested in, but without uart adapter, I can’t say what the issue is.

Polarite, 2024-06-29:
it is fat32, someone at the discord was trying to find what the issue is and while I was trying to take a video, stuff just started to work out of blue, not sure what made it work but it works now... *shrug*

I just downgraded to 6.1.1 and upgraded back to 6.2.0 v4 and it wasnt working, booted to atmosphere, was messing around then while taking a video of opening bit it kinda decided to... just work?? sorry I am not certain what the issue is

GavinDarkglider, 2024-06-29:
I am going to close this then.

closed 2024-06-29 by GavinDarkglider