libretro/Lakka-LibreELEC #1995: WPA3 support Rasberry Pi 4B?

rk9qn3j, 2024-07-29:
Hi,

Is WPA3 supported for Raspberry Pi 4B? I'm trying to connect to my WPA3-only WiFi, but aren't able to do so.

I'm currently running Lakka 5.0.

gouchi, 2024-08-03:
As mentioned in the [issue template](https://github.com/libretro/Lakka-LibreELEC/blob/Lakka-v5.x/.github/ISSUE_TEMPLATE.md), please provide some log otherwise it will be difficult to debug.

> It is important to provide logs for debugging as part of your report. Instructions for generating logs can be found in the > Troubleshooting Lakka documentation at https://www.lakka.tv/doc/Troubleshooting-Lakka/

Thank you.

rk9qn3j, 2024-08-04:
Sorry - it was a more general question if WPA3 was supported or not.

I tried to create log files according to the article (start Lakka in verbose mode and output to `log.txt`), but the log doesn't contain any useful information regarding WiFi. Then I tried to troubleshoot directly via the console, but with no success (e.g. look for any useful information with `journalctl` and connect directly with `wpa_supplicant`). It seems that the necessary versions of the packages are there though (connman 1.41 and wpa_supplicant 2.10).

ShigeakiAsai, 2024-08-11:
Just infromation.

https://forums.raspberrypi.com/viewtopic.php?t=373077
https://github.com/raspberrypi/linux/pull/5945
By these information, the RPi kernel included WPA3 function into rpi-6.6.y branch on on Feb 16.

And Lakka version & RPi kernel version are below (at Aug 11, 2024)
Lakka 5.x : 6.1.77 (commit:14c373b069031f01031e113f62d65c7518f0f084) - Feb 29, 2024
Lakka 6.x : 6.6.33 (commit:51c5d2bb01d3a0cff31d00686f35ed8f33039f78) - Jun 13, 2024

Lakka 6.x may work WPA-3.
If it does not work, latest firmware may be needed.

Thanks,
ASAI, Shigeaki

rk9qn3j, 2024-08-12:
Nice find! Can we expect an update to Lakka 5.x that uses this kernel (>= 6.6) OR do we need to wait for Lakka 6.x? In the meantime, can this be fixed manually using the method described in the link above? Isn't the Lakka file system read-only?

ShigeakiAsai, 2024-11-21:
Sorry for late replying.

LibreELEC-13 on RPi5 also have same trouble.
https://forum.libreelec.tv/thread/29155-rpi5-network-disappeared/

We need to wait more time for fix this issue I think.

Thanks,
ASAI, Shigeaki

gouchi, 2025-02-14:
Hi,

It should be fixed with [latest nightly](https://nightly.builds.lakka.tv/).

[Source](https://github.com/LibreELEC/LibreELEC.tv/pull/9487)