libretro/Lakka-LibreELEC #1996: Lakka Refuses to Connect to Xbox/Stadia/Generic Bluetooth Controllers

F1zzyD, 2024-08-02:
**Which version of Lakka are you using?**
Generic.x86_64-5.0

**What system hardware are you using?**
Mini PC

**What did you do?**
Tried to pair bluetooth controllers (both Xbox and Stadia) with Lakka. Lakka says "connected" when pairing, however both controllers are still in searching mode and eventually turn off.

**What did you expect to happen?**
The controllers to pair lol

**What happened instead?**
The controllers did not pair lol

**What have I done**
There are many posts about this, either in bug reports or forum posts yet there seems to be no resolution. Hopefully making a more recent post will spark conversation and a fix. Here's what I've done:
- #1552 
- #1528 
- #1840 
- [This reddit post](https://www.reddit.com/r/Lakka/comments/i4j5l0/connecting_an_xbox_one_s_controller_via_bluetooth/)
- [This forum post](https://forums.libretro.com/t/connectin-xbox-one-controller-white-with-bluetooth/13878/20)

My logs are identical to everything included in these posts. I'd love to use this, but every controller I throw at it refuses to connect, thus defeating the purpose.

gouchi, 2024-08-03:
Logs have been provided for RPi not Generic platform which are not using the same kernel.

Which xbox controller do you have ?

You did try with the Menu or via `bluetoothctl` ?

For the Stadia controller, there is an [issue](https://github.com/libretro/Lakka-LibreELEC/issues/1824) opened.

jeevadotnet, 2025-01-20:
Did you perhaps ever get this working? 
When I connect my Xbox Wireless controller, the interfaces goes haywire and then doesn't allow any input. Neither from my wireless keyboard or Xbox controller. It is not frozen since the background is still weaving.