libretro/Lakka-LibreELEC #1997: raspberry pi 4b screen upside down

amhooman, 2024-08-03:
for some reason whenever i use the raspberry pi 4 screen with lakka it is upside down. Is there any way to fix it?

gouchi, 2024-08-03:
As mentioned in the [issue template](https://github.com/libretro/Lakka-LibreELEC/blob/Lakka-v5.x/.github/ISSUE_TEMPLATE.md), please provide some log otherwise it will be difficult to debug.

> It is important to provide logs for debugging as part of your report. Instructions for generating logs can be found in the > Troubleshooting Lakka documentation at https://www.lakka.tv/doc/Troubleshooting-Lakka/

Thank you.

ToKe79, 2025-02-02:
please check the documentation to the raspberry pi screen, most probably you must add some args to cmdline to rotate the output.

closed 2025-02-02 by ToKe79