libretro/Lakka-LibreELEC #1999: Odroid Xu4 image issues?

bubuntux, 2024-08-07:
If you are sure that it's a bug in Lakka, please answer the following questions:
- Which version of Lakka are you using?
   5.0 and 5.x
- What system hardware are you using?
Odroid Xu4
- What did you do?
Flashed the img into a microSD using etcher and dd
- What did you expect to happen?
Lakka to boot
- What happened instead?
Nothing

Description:
I flashed lakka 5.0 and nightly 5.x versions into a microSD, i see 2 partitions:
    LAKKA_DISK ext4 seems empty (just .please_resize_me)
    LAKKA vfat with booting files 

When i turn on the computer nothing happens, not only there is no output on the screen the computer seems off (no leds on the usb controllers) seems like the microsd is not properly flashed, i tried etecher and dd , also multiple micro sd and the computer seems to work just fine using batocera, also made sure to disable EDID as suggest the wiki and still no luck.

**I started to wondering if there is an issue with the image itself**.

bubuntux, 2024-08-08:
i was able to boot using the x3u img, couple of things don't work (such as the Ethernet port ) but i think is expected since is the wrong device (although similar enough go boot), this reinforce my theory that the xu4 img has an issue

v0d0r, 2025-02-25:
+1. I have also not had any luck with the x4u image. I thought i was going crazy because everybody just says the the boot.ini EDID.  But ive tried various of those settings and tried various screens. No display.

gouchi, 2025-02-28:
Hi,

Same issue with latest 6.x [nightly](https://nightly.builds.lakka.tv/latest/Exynos.arm/) or [devel build](https://nightly.builds.lakka.tv/devbuild_latest/Exynos.arm/) ? 

Thank you.

aaronouthier, 2025-04-16:
I seem to be having this issue with 2 different Odroid XU4. Also tried 2 different MicroSD Cards. Other OS' boot fine in all cases, just not Lakka.

I have a debug cable connected, and am getting the following output:
`U-Boot 2022.10 (Apr 17 2024 - 02:47:11 +0200) for ODROID-XU4

CPU:   Exynos5422 @ 800 MHz
DRAM:  �`

System hangs after that. Got red power LED, but no blue heartbeat LED, as would normally be the case.

aaronouthier, 2025-04-24:
I forgot to mention that I received the same output in the latest nightly. Well, latest as of last week, when I tried all of this.

ShigeakiAsai, 2025-04-29:
Hello, all

I got and received the Odroid-XU4 today. I also meet this issue.

As [aaronouthier](https://github.com/aaronouthier) said, Odroid-XU4 image on XU4 device is stopping boot at u-boot.
[localbuild-Lakka-XU4.log](https://github.com/user-attachments/files/19960499/localbuild-Lakka-XU4.log)

And, as [bubuntux](https://github.com/bubuntux) said, Odroid-XU3 image + Odroid-XU4 device tree blob on XU4 device is able to start linux and retroarch.
[localbuild-Lakka-XU3+XU4sDT(extlinux_conf-custom).log](https://github.com/user-attachments/files/19960527/localbuild-Lakka-XU3%2BXU4sDT.extlinux_conf-custom.log)

This problem is related Odroid-XU4 u-boot I think.
And this problem is also happened LibreELEC.

I will look into the cause a little bit.

Thanks
ASAI, Shigeaki

v0d0r, 2025-04-29:
I think an important question would be are there anybody that has managed to boot it successfully that can confirm for us it works and/or point us in the right direction or has this been non working for some time?  Maybe getting a release number that actually worked then we can test different releases until we see when it breaks.

ShigeakiAsai, 2025-04-30:
LibreELEC will fix this issue.
Please wait a moment please.

ShigeakiAsai, 2025-05-01:
If you want to confirm the fix, please use my local build version.
https://github.com/ShigeakiAsai/Lakka-LibreELEC/releases/tag/OdroidXU4-FixBootFail-Lakkav6

When I confirmed to fix on LibreELEC 12 version, I will create new PR on Lakka-v6.x with LibreELEC's fix.

Thanks
ASAI, Shigeaki

v0d0r, 2025-05-01:
@ShigeakiAsai I can confirm it boots first time no issues.  It starts with the logo then expands the disk, does a reboot and finally ends up on the first screen as per attachment.  I did not go beyond this but in terms of the boot issue everything looks great now.
Thanks so much for your efforts and good job. 

[image]

ShigeakiAsai, 2025-05-01:
@gouchi 
I have a question.

This problem also occurs on Lakka v5.
Should fix it on Lakka v5?

gouchi, 2025-05-01:
> This problem also occurs on Lakka v5.
> Should fix it on Lakka v5?

If the patch is working for Lakka v5, it might a good idea to have it also. 

Thank you.

ShigeakiAsai, 2025-05-02:
@gouchi 
Thank you replying!
I was able to do it I think.

to all
If you want to confirm the fix on Lakka-v5.x version, please use my local build version.
https://github.com/ShigeakiAsai/Lakka-LibreELEC/releases/tag/OdroidXU4-FixBootFail-Lakkav5

Thanks
ASAI, Shigeaki

Ntemis, 2025-05-03:
Issue fixed, closing down

closed 2025-05-03 by Ntemis