libretro/Lakka-LibreELEC #2001: Tanix TX6 H6 - Black screen on Muppen 64

RuBeN16v, 2024-08-15:
Hello, just tried the last stable and last nihgtly versions on lakka for this device and have the same problem
There is no way to play N64 games, Parrallel runs bad on this low end device, and Muppen 64 plus next seems to be working at full speed , but I can only get audio in a black screen.
I can open the retroarch menu, and all is working good except the screen is black.
I have tried to change the shader to null and some things I have read but nothing helps.
So hope you find a solution. Thanks in advance

Ntemis, 2024-08-15:
Have in mind that GPU (Mali-T720 MP2) is really bad performer on this SOC, has lower gflops than an old Rpi3, dont expect miracles.

Ntemis, 2024-08-15:
[image]

RuBeN16v, 2024-08-17:
I know that´s not the best soc, in fact i was impressed when tried lakka and saw that it can play dreamcast/atomiswave/naomi and even some light psp games.
Just curious why N64 has that black screen because I can start for example 1080 snowboarding and start a race and the sound is good like the game is runing smooth
I don´t want to play N64 at 1080p resolution I know that´s a weak soc but I think it must show image even if the games runs slow as in Parallel

orangechickenpasta, 2024-09-20:
There was a similar issue created 2 years ago but it was never correctly fixed. If anyone else is experiencing this with the H6 your only real option is another operating system.

Ntemis, 2024-11-11:
this was fixed by me, reopen if issue still exist

closed 2024-11-11 by Ntemis