libretro/Lakka-LibreELEC #2006: Pcsx-rearmed black screen on Raspberry pi 5

rafahrpl, 2024-11-10:
If you are sure that it's a bug in Lakka, please answer the following questions:
- Which version of Lakka are you using?
Lakka-RPi5.aarch64-5.x-20240723-e2c1b74.img.gz
- What system hardware are you using?Raspberry pi 5
- What did you do?
I installed Lakka and run Pcsx-rearmed
- What did you expect to happen?
Run PS1 games normally.
- What happened instead?
PS1 games like crash bandicoot freeze screen, and after enter in menu the game run a few seconds, and freeze screen again. Games like Gran Turismo 2 freeze image too, after a few seconds game continue and freeze again.

Ntemis, 2024-11-11:
Can you provide any logs?

rafahrpl, 2024-11-11:
How I can provide this logs?

gouchi, 2024-11-11:
@rafahrpl Please check this [documentation](https://www.lakka.tv/doc/Troubleshooting-Lakka/).

You could find it making [this research](https://lmddgtfy.net/?q=lakka%20logs).

Thank you.

synthic, 2024-11-17:
Have you tried the Beetle PSX HW core? It seems to be more capable than PCSX-ReARMed on Raspberry Pi 5.

rafahrpl, 2024-11-23:
Bellow the files logs. 
[log.txt](https://github.com/user-attachments/files/17881973/log.txt)
[log1.txt](https://github.com/user-attachments/files/17881974/log1.txt)

> Have you tried the Beetle PSX HW core? It seems to be more capable than PCSX-ReARMed on Raspberry Pi 5.
Don't appears Beetle PSX HW core or Beetle PSX, only PCSX-ReARMed on Raspberry PI.

rafahrpl, 2024-11-23:
Bellow the files logs.
[log.txt](https://github.com/user-attachments/files/17882002/log.txt)
[log1.txt](https://github.com/user-attachments/files/17882003/log1.txt)

> Have you tried the Beetle PSX HW core? It seems to be more capable than PCSX-ReARMed on Raspberry Pi 5.

Don't appears Beetle PSX HW core or Beetle PSX, only PCSX-ReARMed on Raspberry PI.

gouchi, 2024-11-24:
@rafahrpl Thank you. It seems there is an issue using Vulkan driver so can you try to make a test with another video driver ? Also, PCSX-ReARMed  core is not using  hardware renderers.

rafahrpl, 2024-12-10:
Hi, I'm try again using glcore and the core run without freeze.

closed 2025-02-09 by Ntemis