libretro/Lakka-LibreELEC #2017: Lakka Not Booting on 2GB Raspberry Pi5

paqman3d, 2024-12-19:
As reported on foobar (https://forums.libretro.com/t/raspberry-pi-5-2gb-no-boot/45941/2), I can confirm that the 2GB Pi5 does not boot with Lakka 5.X from 7/23/24. 

I don't have another 4GB to test, but I'm getting the same behavior as foobar -- black screen/no signal, 100% fan.

ShigeakiAsai, 2024-12-22:
I had same issue on my "Lakka-v5.x 2GB Raspberry Pi5".
And I had same issue on my "LibreELEC-v11.0.6 2GB Raspberry Pi5".

I confirmed Lakka & linux source tree.
- Lakka-v5.x RPi linux (6.1.77) does not support "2GB Raspberry Pi5".
- Official RPi linux supports this device on 6.6.21 or later.

So we should use Lakka-v6.x better I think.
Because I was able to start "Lakka-v6.x 2GB Raspberry Pi5" (Commit : 71cb84343a0bb27df0fce69f21db146d24dc390e)

Thanks
ASAI, Shigeaki

ToKe79, 2025-01-29:
fixed in 6.x / devel branch

closed 2025-01-29 by ToKe79