libretro/Lakka-LibreELEC #2025: H616 compilation problem

986046679, 2025-01-08:
Hello, I would like to ask, I see that the official website download includes the image of Allwinner H616 chip, but the Allwinner device in the project does not have the option of H616, I would like to ask if I need to compile the image supported by H616 how to compile the project

Diegrosan, 2025-01-16:
> Hello, I would like to ask, I see that the official website download includes the image of Allwinner H616 chip, but the Allwinner device in the project does not have the option of H616, I would like to ask if I need to compile the image supported by H616 how to compile the project

Allwinner H616 has the hardware requirement to run emuelec but at the moment this is the only compatible one and it may take a while to configure the H616.

https://github.com/EmuELEC/EmuELEC/tree/dev/projects/Allwinner/devices
H2-plus | Allwinner: u-boot: Add missing patches for H2+ | 2 years ago
H3 | u-boot (Allwinner H3): rebase patches for 2022.10-rc3 | 3 years ago
H5 | Merge pull request #5435 from antonlacon/le11-arch-options | 4 years ago
H6 | Allwinner: H6: linux: Enable iommu | 3 years ago
R40 | Allwinner: linux: Remove vmalloc workaround

986046679, 2025-01-17:
I found a branch of H616 under device in the branch of lakka-5x, but I have a question as to whether it works for BSP

Diegrosan, 2025-01-17:
> I found a branch of H616 under device in the branch of lakka-5x, but I have a question as to whether it works for BSP

I then advise you to try compiling lakka-5x
No need to hesitate, just try, see the lakka documentation, forum and some YouTube tutorials that work

986046679, 2025-01-20:
Hello, I would like to ask whether there is any H616 finished development board that can be used for this H616 engineering image, and whether you can give me an equipment model so that I can purchase and test it.

Diegrosan, 2025-01-22:
Sorry, but I don't know any.

ShigeakiAsai, 2025-01-22:
Lakka-v5.x contains "Orange-Pi Zero 2" supports.
http://www.orangepi.org/html/hardWare/computerAndMicrocontrollers/details/Orange-Pi-Zero-2.html
https://github.com/libretro/Lakka-LibreELEC/blob/Lakka-v5.x/scripts/uboot_helper#L165

@Ntemis 
Do you have/know about Lakka-v6,x Allwinner H616 support?
Drop?

Ntemis, 2025-01-22:
Well i never got time to finish it

ShigeakiAsai, 2025-01-23:
@Ntemis
Thank you for your information.

@986046679
There is no fully supported H616's Lakka.
But, Lakka-v5.x may work on Orange Pi  Zero 2.
If you have this product, please try the following nighthly image.
https://nightly.builds.lakka.tv/5.x/2024-07-23/H616.aarch64/
[Lakka-H616.aarch64-5.x-20240723-e2c1b74-orangepi-zero2.img.gz](https://nightly.builds.lakka.tv/5.x/2024-07-23/H616.aarch64/Lakka-H616.aarch64-5.x-20240723-e2c1b74-orangepi-zero2.img.gz)

@Diegrosan
Many thanks!

986046679, 2025-01-23:
Do you know anything about emuelec, or do you have time to help me adapt emuelec? I lack an expert who knows libreelec u-boot. After I replaced the u-boot of H313 ubuntu with emuelec, there were too many problems that I could not solve. If you're interested, can you help me?

ShigeakiAsai, 2025-02-07:
I confirmed my Orange Pi Zero 2 (H616) booting u-boot & linux via UART and NO HDMI display showing.
Linux : 6.13
u-boot : 2024.10

NO HDMI display showing because H616 display engine is not implemented yet to mainline linux.
And H616 HDMI function is not implemented yet too.
Now this implementation is working development by ROCKNIX developers. It will apper soon I think.

@986046679 
Please wait several time.

986046679, 2025-02-08:
@ShigeakiAsai Hey bro, thank you for answering my question. Now I want to try the configuration of H313, but I didn't know that you had already ported the latest u-boot and kernel. I used your lakka H616 project to replace U-boot 2024.10 and Linux v6.11 version, but my Linux compilation failed. So I would like to ask if you could share the source code of the u-boot 2024.10 & Linux 6.13 versions you mentioned. I want to try compiling and modifying them.

[image]

ShigeakiAsai, 2025-02-08:
@986046679 

At first, I'm sorry for my English is strange because I am not native.

I have changed my H616 environment to Lakka devel branch.
By this changing...
Linux : 6.12.10
u-boot : 2025.01  (2024.10 is also able to start I think)

This version also started u-boot and linux without display output on Allwinner H616 - Orange Pi Zero 2.
Attached file is boot log.
[Allwinner-H616-OrangePiZero2-boot-log.txt](https://github.com/user-attachments/files/18718670/Allwinner-H616-OrangePiZero2-boot-log.txt)

You can see my modification on my Lakka repo - Test-Allwinner-H616 branch.
https://github.com/ShigeakiAsai/Lakka-LibreELEC/tree/Test-Allwinner-H616

BTW, this issue is not Lakka's issue now I think.
If you can agree with my suggestion, I hope to close this please.

Thanks
ASAI, Shigeaki

986046679, 2025-02-11:
@ShigeakiAsai 
Hello, the following error occurred when I compiled the Test-Allwinner-H616 provided by you, may I ask how to solve this problem? And I am sorry for raising the problem here, because I did not find issues under the github of Test-Allwinner-H616 to raise the problem

ShigeakiAsai, 2025-02-11:
@986046679 
Ah, Sorry
Enabled 'Issue' function on my github Lakka repo.
Please ask on my repo.

Thanks
ASAI, Shigeaki

closed 2025-02-13 by 986046679